Bodmer/TFT_eSPI #647: TFT_eSPI() disables D1/2 on a NodeMCU

RogerDavidClarke, 2020-05-31:
Hi,

I'm having some issues with Pins D1/D2 on a NodeMCU being "disabled" when I invoke "TFT_eSPI()". Attached to D1/2 is an MPL3115A2. Using Adafruit_ILI9341 both the display and sensor work except the adafruit is inferior to yours and I'm "leveraging" the ring meters from the examples so I don't really want to spend time working on an alternative to drawString(...)

Kind of stumped on this one. There must be something in the eSPI code that changes the state of D1 and/or D2 preventing the ANY 1CS devices being detected. Any help would be appreciated.

With both libraries, the display uses the same pins: 

RST - RST
#define TFT_CS   PIN_D8  // Chip select control pin D8
#define TFT_DC   PIN_D3  // Data Command control pin
#define TFT_RST  PIN_D4  

MOSI ->D7
SCK - D5
MISO ->D6

Default config: User_Setup.h

I'm not using touch. I tired the following changes:
TFT_e_SPI.CPP
//pinMode(TFT_D1, OUTPUT); digitalWrite(TFT_D1, HIGH);
//pinMode(TFT_D2, OUTPUT); digitalWrite(TFT_D2, HIGH);

User_Setup_select.h
#define PIN_D1   5  // GPIO5        User purpose
#define PIN_D2   4  // GPIO4        User purpose

Bodmer, 2020-05-31:
Is the TFT working OK?

RogerDavidClarke, 2020-05-31:
Yes the display is work fine. It's just pins D1 and D2 that are not

Bodmer, 2020-05-31:
OK, I will run some tests on a representative setup with an I2C sensor and see what happens.

Bodmer, 2020-05-31:
I radn this sketch and it finds the I2C device. Can you try it.
```
#include <TFT_eSPI.h> // Hardware-specific library

TFT_eSPI tft = TFT_eSPI();       // Invoke custom library

#include <Wire.h>

void setup()
{
  Wire.begin();
  tft.init();
  tft.fillScreen(TFT_RED);
  Serial.begin(9600);
  while (!Serial);             // Leonardo: wait for serial monitor
  Serial.println("\nI2C Scanner");
}

void loop()
{
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ ) 
  {
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4) 
    {
      Serial.print("Unknow error at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

  tft.fillScreen(random(0x10000));

  delay(5000);           // wait 5 seconds for next scan
}
```

RogerDavidClarke, 2020-05-31:
worked perfectly. Oddly it works using  PlatformIO but not the Arduino IDE even with a fresh TFT lib installed. Not sure what's going on there but if it works in PIO, that's sufficient for me to progress my project. Thanks for the help. Appreciated.

closed 2020-05-31 by RogerDavidClarke

Bodmer, 2020-06-01:
Interesting. Which version of the Arduino IDE did you try?

RogerDavidClarke, 2020-06-01:
Window store (1.8.33.0)