Bodmer/TFT_eSPI #648: Speedup setAddr

tablatronix, 2020-06-01:
Check out performance whisperer speedup for setAddr

I saw about 3-4 fps boost in setWindow with a quick test.

https://github.com/PaulStoffregen/ILI9341_t3/pull/59/commits/a744ba709209a33cfa2f7a27af10327b697c60c9#diff-e72979a4307b888dee6807fff2037855R325

Bodmer, 2020-06-01:
The library already adopts this method for [drawPixel](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L2697-L2709), where the biggest gains can be made.  This then means it is possible to order groups of single pixel writes to minimise the changes in x and y [as done here](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L1616-L1626) so that, for example, circle outlines are drawn faster because pixel positions are orthogonally positioned.

I do not recommend missing out TFT_RAMWR because there may also be other commands sent between screen area updates by the user sketch that prevent further writes, the data sheets for a range of displays is not always clear on this command interaction. If a TFT_RAMRD was last sent then writes will then fail without the TFT_RAMWR, however the existing readAddrWindow in the library already covers this particular case as the last row and column variables are set invalid in that function to force the next window update.

To use the existing column and row variables (addr_col and addr_row) this should work:
```
void TFT_eSPI::setWindow(int32_t x0, int32_t y0, int32_t x1, int32_t y1)
{
  //begin_tft_write(); // Must be called before setWindow

#ifdef CGRAM_OFFSET
  x0+=colstart;
  x1+=colstart;
  y0+=rowstart;
  y1+=rowstart;
#endif

  // No need to send x if it has not changed (speeds things up)
  if (addr_col != (x0<<16 | x1)) {
    DC_C; tft_Write_8(TFT_CASET);
    DC_D; tft_Write_32C(x0, x1);
    addr_col = (x0<<16 | x1);
  }

  // No need to send y if it has not changed (speeds things up)
  if (addr_row != (y0<<16 | y1)) {
    DC_C; tft_Write_8(TFT_PASET);
    DC_D; tft_Write_32C(y0, y1);
    addr_row = (y0<<16 | y1);
  }

  DC_C; tft_Write_8(TFT_RAMWR);
  DC_D;

  //end_tft_write(); // Must be called after setWindow
}
```
However parts of drawPixel will also need to be updated to avoid a particular corner case to use:
```
  // No need to send x if it has not changed (speeds things up)
  if (addr_col != (x<<16 | x)) {
    DC_C; tft_Write_8(TFT_CASET);
    DC_D; tft_Write_32D(x);
    addr_col = (x<<16 | x);
  }

  // No need to send y if it has not changed (speeds things up)
  if (addr_row != (y<<16 | y)) {
    DC_C; tft_Write_8(TFT_PASET);
    DC_D; tft_Write_32D(y);
    addr_row = (y<<16 | y);
  }

```
The bad news is that drawLine() will probably then be slower but that probably has negligible impact in most sketches.

tablatronix, 2020-06-01:
I just happen to be in my code when he posted it, so it was REAL quick test,  I do not even know what TFT_RAMWR is, I did see addr_col but did not investigate drawPixel I figured there was something like that going on.

I can test some stuff out you mentioned above if I have time to check the actual benchmark tests.

readAddrWindow is for reading the screen or local buffers? I have not played with that yet, I do not have a MISO pin to spare..

Bodmer, 2020-06-02:
I have run some simple tests on the ESP32 and the "graphicstest" improvement is small at ~0.7% though there may be some tests (in particular for tiny screen block updates) that show better improvement.
readAddrWindow as you have surmised is the read screen area equivalent of setAddrWindow.

Thanks for raising this and prompting  me to test this.

Bodmer, 2020-06-06:
Updated setWindow()

closed 2020-06-06 by Bodmer