Bodmer/TFT_eSPI #649: 3.5" TFT LCD ILI9486

Oscar230296, 2020-06-02:
hello, i'm trying to use the TFT_eSPI with ILI9486 parallel mode but i have one problem the screen are like this and i dont know how to solve
// See SetupX_Template.h for all options available

#define ESP32_PARALLEL

#define ILI9488_DRIVER

// ESP32 pins used
#define TFT_CS   33  // Chip select control pin
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin

#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    2

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits
#define TFT_D2   26
#define TFT_D3   25
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

[image]

can you help me because, i'm trying to develop a lowcost termal camera and only the screen is wrong.
thanks for your comments

Oscar230296, 2020-06-02:
i'm trying to solve and found other problem that is mirror and the width and length is reduced can you help me please
[image]

Bodmer, 2020-06-02:
The problem may be because the setup file is for the wrong driver (ILI9488).

If you have an ILI9486 then you need to use the following in setup:

#define ILI9486_DRIVER

Bodmer, 2020-06-05:
I suspect the display is not an ILI9486 type.

Try the [MCUFRIEND_kbv library ](https://github.com/prenticedavid/MCUFRIEND_kbv) as that library will detect the driver for you and report the type.

Bodmer, 2020-06-16:
Closing, assumed to be fixed by user.

closed 2020-06-16 by Bodmer