Bodmer/TFT_eSPI #652: STM32F103 ST7789

cgrrty, 2020-06-03:
Hi
    How to use ST7789 with STM32F103?The code how to config?
Thanks

Bodmer, 2020-06-05:
Use setup 32 and edit the User_Setup_Select header to point to that setup:
Comment out [this line](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L22).
Uncomment [this line](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L62) to include the setup.
Edit https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup32_ILI9341_STM32F103.h#L11 to read:
```
#define ST7789_DRIVER
```

Bodmer, 2020-06-05:
That assumed you have a SPI display...

closed 2020-06-06 by Bodmer

cgrrty, 2020-06-11:
Hi
 Now the device is working, but it works error
when i used    _**tft.fillScreen(TFT_PINK);**_
The device no full screen display
I used 240*135 st7789 tft display

[image]

cgrrty, 2020-06-11:
`      //////////////////////////////////////////////////////////////
        // Setup for STM32F103 (e.g. Blue Pill) and ILI9341 display //
        //////////////////////////////////////////////////////////////

// Last update by Bodmer: 14/1/20

// Define STM32 to invoke STM32 optimised driver
#define STM32

// Define the TFT display driver
#define ST7789_DRIVER     // Configure all registers

#define TFT_WIDTH  135
#define TFT_HEIGHT 240

#define CGRAM_OFFSET      // Library will add offsets required

// MOSI and SCK do not need to be defined, connect:
//  - Arduino SCK (Blue Pill pin A5)  to TFT SCK
//  - Arduino MOSI (Blue Pill pin A7) to TFT SDI (may be marked SDA or MOSI)
//  - Optional to read TFT: Arduino MISO (Blue Pill pin A6) to TFT SDO (may be marked MISO or may not exist on TFT)
//  Note: not all TFT's support reads

// reminder: Blue Pill SPI pins are SCK=A5, MISO = A6, MOSI=A7

// Can use Ardiuno pin references, arbitrary allocation, TFT_eSPI controls chip select
#define TFT_CS   A4 // Chip select control pin to TFT CS
#define TFT_DC   A3 // Data Command control pin to TFT DC (may be labelled RS = Register Select)
#define TFT_RST  A2 // Reset pin to TFT RST (or RESET)

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// STM32 support for smooth fonts via program memory (FLASH) arrays
#define SMOOTH_FONT

// #define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// Assuming the processor clock is 72MHz:
#define SPI_FREQUENCY  36000000   // 36MHz SPI clock
//#define SPI_FREQUENCY  18000000   // 18MHz SPI clock

#define SPI_READ_FREQUENCY  12000000 // Reads need a slower SPI clock

#define SPI_TOUCH_FREQUENCY  2500000 // Must be very slow

// This has no effect, transactions are automatically enabled for STM32
// #define SUPPORT_TRANSACTIONS

#include <TFT_eSPI.h>       // Hardware-specific library

TFT_eSPI tft = TFT_eSPI();  // Invoke custom library

void setup(void) {
  tft.init();
  tft.setRotation(3);
  tft.fillScreen(TFT_PINK);
  delay(100);
  }
void loop() {
}`

cgrrty, 2020-06-11:
```
#include <TFT_eSPI.h> // Hardware-specific library

TFT_eSPI tft = TFT_eSPI(); // Invoke custom library

void setup(void) {
tft.init();
tft.setRotation(3);
tft.fillScreen(TFT_PINK);
delay(100);
}
void loop() {
}
```

Bodmer, 2020-06-11:
These display drivers can drive displays with much higher resolution (240x320) and the small 135x240 screen area can be positioned anywhere in this larger area so offsets have to be used. The offsets for rotation 3 are defined [here](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ST7789_Rotation.h#L62-L73).  These offsets are for 240x240 and the TTGI 135x240 display, it appears that you have a new display configuration that requires different offsets.  The required offsets will need to be determined by experimentation.  As I do not have a display like yours I cannot determine what they may be.

cgrrty, 2020-06-15:
```
/* Choose a display rotation you want to use: (0-3) */

#define ST7789_ROTATION 1       

#ifdef USING_135X240

    #if ST7789_ROTATION == 0
        #define ST7789_WIDTH 135
        #define ST7789_HEIGHT 240
        #define X_SHIFT 53
        #define Y_SHIFT 40
    #endif

    #if ST7789_ROTATION == 1
        #define ST7789_WIDTH 240
        #define ST7789_HEIGHT 135
        #define X_SHIFT 40
        #define Y_SHIFT 52
    #endif

    #if ST7789_ROTATION == 2
        #define ST7789_WIDTH 135
        #define ST7789_HEIGHT 240
        #define X_SHIFT 52
        #define Y_SHIFT 40
    #endif

    #if ST7789_ROTATION == 3
        #define ST7789_WIDTH 240
        #define ST7789_HEIGHT 135
        #define X_SHIFT 40
        #define Y_SHIFT 53
    #endif

```
**Use this codes, the display is normal
I don’t see any difference, I’m confused.**

```
/**
 * @brief Initialize ST7789 controller
 * @param none
 * @return none
 */
void ST7789_Init(void)
{
		HAL_Delay(20);
    ST7789_RST_Clr();
    HAL_Delay(20);
    ST7789_RST_Set();
    HAL_Delay(50);
		
    ST7789_WriteCommand(ST7789_COLMOD);		//	Set color mode
    ST7789_WriteSmallData(ST7789_COLOR_MODE_16bit);
  	ST7789_WriteCommand(0xB2);				//	Porch control
	{
		uint8_t data[] = {0x0C, 0x0C, 0x00, 0x33, 0x33};
		ST7789_WriteData(data, sizeof(data));
	}
	ST7789_SetRotation(ST7789_ROTATION);	//	MADCTL (Display Rotation)
	
	/* Internal LCD Voltage generator settings */
    ST7789_WriteCommand(0XB7);				//	Gate Control
    ST7789_WriteSmallData(0x35);			//	Default value
    ST7789_WriteCommand(0xBB);				//	VCOM setting
    ST7789_WriteSmallData(0x19);			//	0.725v (default 0.75v for 0x20)
    ST7789_WriteCommand(0xC0);				//	LCMCTRL	
    ST7789_WriteSmallData (0x2C);			//	Default value
    ST7789_WriteCommand (0xC2);				//	VDV and VRH command Enable
    ST7789_WriteSmallData (0x01);			//	Default value
    ST7789_WriteCommand (0xC3);				//	VRH set
    ST7789_WriteSmallData (0x12);			//	+-4.45v (defalut +-4.1v for 0x0B)
    ST7789_WriteCommand (0xC4);				//	VDV set
    ST7789_WriteSmallData (0x20);			//	Default value
    ST7789_WriteCommand (0xC6);				//	Frame rate control in normal mode
    ST7789_WriteSmallData (0x0F);			//	Default value (60HZ)
    ST7789_WriteCommand (0xD0);				//	Power control
    ST7789_WriteSmallData (0xA4);			//	Default value
    ST7789_WriteSmallData (0xA1);			//	Default value
	/**************** Division line ****************/

	ST7789_WriteCommand(0xE0);
	{
		uint8_t data[] = {0xD0, 0x04, 0x0D, 0x11, 0x13, 0x2B, 0x3F, 0x54, 0x4C, 0x18, 0x0D, 0x0B, 0x1F, 0x23};
		ST7789_WriteData(data, sizeof(data));
	}

    ST7789_WriteCommand(0xE1);
	{
		uint8_t data[] = {0xD0, 0x04, 0x0C, 0x11, 0x13, 0x2C, 0x3F, 0x44, 0x51, 0x2F, 0x1F, 0x1F, 0x20, 0x23};
		ST7789_WriteData(data, sizeof(data));
	}
    ST7789_WriteCommand (ST7789_INVON);		//	Inversion ON
		ST7789_WriteCommand (ST7789_SLPOUT);	//	Out of sleep mode
  	ST7789_WriteCommand (ST7789_NORON);		//	Normal Display on
  	ST7789_WriteCommand (ST7789_DISPON);	//	Main screen turned on	

	HAL_Delay(50);
	ST7789_Fill_Color(BLACK);				//	Fill with Black.
}
```

cgrrty, 2020-06-15:
https://item.taobao.com/item.htm?spm=a1z09.2.0.0.63d22e8dGKvkII&id=586409078629&_u=f1ng8o2lae04

Bodmer, 2020-06-15:
The simple solution is to [edit the file shown here](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ST7789_Rotation.h). I do not debug code written by users unless there is a potential problem in the library.
The display controller can address a 240x320 pixel area, your screen pixels are mapped somewhere in that area. The default row and column offsets are for a display are shown in pink below, within the total screen area shown in grey. You need to determine what your offsets (colstart, rowstart) are and edit the numbers, recompile upload and test. This is the default offset screen area mapping:
[image]

cgrrty, 2020-06-19:
Hi
I want to know the drivers st7789 used stm32, have you tested it?
There are few people who use arduino with stm32 ?

Bodmer, 2020-06-29:
Try:
#define ST7789_2_DRIVER
Instead of:
#define ST7789_DRIVER

In the setup file.

cgrrty, 2020-06-30:
use
`#define ST7789_2_DRIVER`
The phenomenon that cannot be displayed in full screen is more obvious

Bodmer, 2020-06-30:
Which version of TFT_eSPI are you running?
Have you tried reducing the SPI clock rate?

cgrrty, 2020-06-30:
```
// Stop fonts etc being loaded multiple times
#ifndef _TFT_eSPIH_
#define _TFT_eSPIH_

#define TFT_ESPI_VERSION "2.2.13"
```
```
        //////////////////////////////////////////////////////////////
        // Setup for STM32F103 (e.g. Blue Pill) and ILI9341 display //
        //////////////////////////////////////////////////////////////

// Last update by Bodmer: 14/1/20

// Define STM32 to invoke STM32 optimised driver
#define STM32

// Define the TFT display driver
#define ST7789_DRIVER     // Configure all registers
// #define ST7789_2_DRIVER

#define TFT_WIDTH  135
#define TFT_HEIGHT 240

#define CGRAM_OFFSET      // Library will add offsets required
#define MULTI_TFT_SUPPORT

// MOSI and SCK do not need to be defined, connect:
//  - Arduino SCK (Blue Pill pin A5)  to TFT SCK
//  - Arduino MOSI (Blue Pill pin A7) to TFT SDI (may be marked SDA or MOSI)
//  - Optional to read TFT: Arduino MISO (Blue Pill pin A6) to TFT SDO (may be marked MISO or may not exist on TFT)
//  Note: not all TFT's support reads

// reminder: Blue Pill SPI pins are SCK=A5, MISO = A6, MOSI=A7

// Can use Ardiuno pin references, arbitrary allocation, TFT_eSPI controls chip select
#define TFT_CS   A4 // Chip select control pin to TFT CS
#define TFT_DC   A3 // Data Command control pin to TFT DC (may be labelled RS = Register Select)
#define TFT_RST  A2 // Reset pin to TFT RST (or RESET)

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// STM32 support for smooth fonts via program memory (FLASH) arrays
#define SMOOTH_FONT

// #define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// Assuming the processor clock is 72MHz:
#define SPI_FREQUENCY  36000000   // 36MHz SPI clock
//#define SPI_FREQUENCY  18000000   // 18MHz SPI clock

#define SPI_READ_FREQUENCY  12000000 // Reads need a slower SPI clock

#define SPI_TOUCH_FREQUENCY  2500000 // Must be very slow

// This has no effect, transactions are automatically enabled for STM32
// #define SUPPORT_TRANSACTIONS
```
**the SPI clock rate is default , I didn't change it.**

Bodmer, 2020-06-30:
Try 18Mhz I.e. 18000000