Bodmer/TFT_eSPI #655: How to config TFT 4 inch Touch Screen ( ST7796 ) with ESP32 ?

TridentTD, 2020-06-07:
I'm using this TFT 4 inch (ST7796) touch screen with ESP32 .

https://www.aliexpress.com/item/32812882281.html

I configed it by the following ...

```cpp
#define ST7796_DRIVER	

#define SPI_MISO    19
#define SPI_MOSI    23
#define SPI_SCLK    18

#define TFT_MISO    SPI_MISO
#define TFT_MOSI    SPI_MOSI
#define TFT_SCLK    SPI_SCLK
#define TFT_CS      14      // Chip select control pin
#define TFT_DC      27      // Data Command control pin
#define TFT_RST     33
#define TFT_BL      32       // TFT backlight  pin
#define TFT_BACKLIGHT_ON HIGH

#define TOUCH_MISO     SPI_MISO
#define TOUCH_MOSI     SPI_MOSI
#define TOUCH_SCLK     SPI_SCLK
#define TOUCH_CS       12      // TouchScreen Chip Select
```

This TFT 4" now can display anything to the screen fine, 
however it can't response any touch 
( I tried by the TFT_eSPI's `keypad480x320` example. )

I wish to config about the touch spi too.
How to config the touch spi ?

Thank you very much.

Bodmer, 2020-06-07:
Make sure your screen has the touch controller chip fitted and the touch film on the front of the TFT.

Make sure the touch controller I/O pins are wired to the ESP32.

Run the test and diagnostics sketch "Test_Touch_Controller", report the results.

TridentTD, 2020-06-07:
My connection are wired as the following ...

[image]

This wiring is fine for my another TFT 3.5" (ILI9488)  Touch screen 
with TFT_eSPI old version (1.4.9).

When I tried by  TFT_eSPI current version 2.2.9, 
both TFT 4" (ST7796)  & 3.5" (ILI9488)  Touch screen
can display well , except the touch controller.

With  testing by sketch "Test_Touch_Controller",
both (TFT 3.5" & TFT 4") are displayed 4 the boxes at the corner of screen 
as the following clip.

TridentTD, 2020-06-08:
OK, it's fine now, some wiring are not stable.

Thank you very much.

closed 2020-06-28 by Bodmer

thhe3001, 2020-12-27:
Hello @TridentTD 
did you really get the display and the touch working on an esp32 with SPI? I'm trying to get the same configuration working and need some hints.

Display works well. Touch works fine with the xpt2046-library but touch doesn't get any values with the TFT_Espi library.  If I try to use both libraries, the xpt2046-library doesn't work anymore.

Ali-Abdelwdod, 2021-10-02:
> 
> 
> I'm using this TFT 4 inch (ST7796) touch screen with ESP32 .
> 
> https://www.aliexpress.com/item/32812882281.html

but your link has iLi19486 not ST7796 can iLi9486 work with ST7796 library?

SBeneitO, 2023-05-18:
I have the TFT ST7796 4" 480x320 and the screen is always blank. I have read all the forum options but I have not gotten anything. It works well with Adafruit_ST7796S_kbv.h but not with the speed of your library. I don't know what I can do anymore. Thank you