Bodmer/TFT_eSPI #661: About 'Tips', how to understand ?

LusterWong, 2020-06-11:
The library file has been successfully added. 

There are 3 projects in my project that need to use this library, but the LCD controllers are different. 

According to the tips, I don’t know how to operate. What I can do is copy 3 copies of the eSPI library and modify the User_Setup.h of each library. And User_Setup_Select.h, this is a very bad practice, resulting in the library can not be updated and maintained online.

Is there a better way?

Bodmer, 2020-06-11:
You do not need to have 3 copies of the library. The idea is to create your own user setup file for each board configuration and include just one, the one you want to compile.  To swap to another hardware setup this just means editing the User_Setup_Select.h to load the one you want (add// remove // and save).
The Arduino IDE does not allow a library to "see" the #defines in a sketch and there are so many options to support many displays that function parameter based configuration is impractical and inefficient.
The key advantage of this approach is that once the setup is defined all sketches will run without modification.
For project based developments I suggest you use the PlatformIO IDE as you can create projects with the hardware configurations built into the project.

closed 2020-06-11 by Bodmer

LusterWong, 2020-06-12:
> You do not need to have 3 copies of the library. The idea is to create your own user setup file for each board configuration and include just one, the one you want to compile. To swap to another [...]

For some special reasons, currently only eclipse or arduino ide can be used.

If I use arduino ide, how can I organize engineering for different LCD controllers? 
Where are User_Setup.h and User_Setup_Select.h placed? How to include and call the project master file? 
Can you provide an example? 

Thank you very much.

Bodmer, 2020-06-12:
Read the comments in the default setup in the library folder [here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h) and all should be clear.