Bodmer/TFT_eSPI #662: Calling setTextColor(xx, xx) after the edge font display error

LusterWong, 2020-06-11:
The program uses \examples\160 x 128\Arduino_Life、RLE_Font_test.ino without any modification.

It is found that there are many screens with flower screen phenomenon, two of which display the same content and font, but there is a display error on the edge.

[image]

This is the code used in the picture above
 ```
tft.fillScreen(TFT_BLACK);
tft.setTextColor(TFT_BLUE, TFT_BLACK);
tft.drawString("012345", 0, 0, 6);
tft.drawString("6789", 0, 40, 6);
tft.drawString("apm-:.", 0, 80, 6);
delay(WAIT);
```
[image]

This is the code used in the picture above
```
tft.fillScreen(TFT_BROWN);
tft.setTextColor(TFT_BLUE);
tft.drawString("012345", 0, 0, 6);
tft.drawString("6789", 0, 40, 6);
tft.drawString("apm-:.", 0, 80, 6);
delay(WAIT);
```

You can see from the two pictures above that the 5 character on the first line is incorrect, while the other is correct.
The difference is that when calling setTextColor one uses the bgcolor parameter and the other does not. Why does using bgcolor cause wrong characters to be displayed on the edges?

LusterWong, 2020-06-11:
I use the ESP32 platform, the LCD display chip is st7735, the resolution is 128x160
IDE is eclipse.

Bodmer, 2020-06-11:
This is a restriction in the way the RLE fonts are rendered for speed since the fonts were originally developed for the humble UNO where every line of code mattered.  When plotting the RLE fonts direct to the screen (i.e. not via a Sprite) then you must ensure all font characters are positioned completely within the screen area.

Bodmer, 2020-06-11:
I have updated the library to allow partially off screen RLE font rendering.

closed 2020-06-11 by Bodmer