Bodmer/TFT_eSPI #663: Multiple display support

SunboX, 2020-06-12:
Hi,

I read about the reasons why only one display is supported from the library in issues #138

Two years have passed now, is there any plan to support multiple displays? Otherwise we have to implement a facade for each function call to simplify our code.

greetings,
André

Bodmer, 2020-06-12:
Thanks for the question. Unfortunately there are no plans for supporting multiple displays so the chip selects for the displays have to be managed by the sketch and this approach appears relatively simple to implement.

There is a "gotcha" in the latest library that improves performance by minimising the setAddrWindow position commands needed, you will need to tweak the library to avoid this for multiple displays by commenting out the "if" conditions [here ](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L2625)and [here](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L2632).

I understand that it is not ideal but the rarity of multiple display support means I cannot justify the effort to implement independent display classes that track all the different settings (font selected, cursor position etc) for different displays.

SunboX, 2020-06-13:
Hey, thanks for the quick response! I understand your reasons, could I maybe offer a small wish?

In our case, it's easy. We have two identical displays. And there's a huge chance that project is getting bigger and live for a longer time. Because of the second reason, I don't like to break the update-ability by forking your lib and modifiyng the code for our needs.

Is there a chance to add a small feature like this?
```cpp
#define MULTIPLE_DISPLAY_SUPPORT

#if !(defined(MULTIPLE_DISPLAY_SUPPORT))
  // No need to send x if it has not changed (speeds things up)
  if (addr_col != (x0<<16 | x1)) {
    DC_C; tft_Write_8(TFT_CASET);
    DC_D; tft_Write_32C(x0, x1);
    addr_col = (x0<<16 | x1);
  }
#endif
```

greetings,
André

Bodmer, 2020-06-13:
OK, no problem, I will add this tomorrow sometime.

SunboX, 2020-06-13:
That's really great, thanks a lot! Btw, no hurry, we have time enough. ;)

Bodmer, 2020-06-15:
Added.
Put
#define MULTI_TFT_SUPPORT
in your setup file.

closed 2020-06-15 by Bodmer

SunboX, 2020-06-15:
Really great, thanks a lot!

Witchdoc59, 2024-02-14:
Could we create separate folders within the library folder then load the library from the different folders.   Copy the TFT_eSPI files to a folder called TFT_eSPI_ILI9341 then when writing code that will use the ILI9341 you call the TFT_eSPI library from that folder?  #include .\TFT_eSPI_ILI9341\TFT_eSPI.h ??