Bodmer/TFT_eSPI #664: Question: Multiple instances of GFXFonts loaded

jpasqua, 2020-06-12:
I have a fairly complex project and within it there are 6 or 7 modules which are each responsible for displaying a "screen" in my application. They are modular and none of them knows about the others. The c++ files corresponding to each screen are stored in a subdirectory:
````
[sketch directory]
|
+--[src]
     |
     +--[gui]
           |--Screen1.cpp
           |--Screen2.cpp
           |- ...
````
Each of these does a `#include <TFT_eSPI>` since they are all making use of the library. I am finding that if I use a GFXFont (e.g. `FreeSansBold9pt7b`) in multiple files, I get multiple copies of the font. This chews up memory fast. 

Two questions: 
1. Am I right in assuming that this is due to the fact that these modules are not in the main sketch directory and treated than files that are?
2. Is there a proper approach to dealing with this?

TIA

Bodmer, 2020-06-13:
This is not something I have tried, unfortunately this is not something I have ever attempted to I am unable to help.

closed 2020-06-13 by Bodmer

jpasqua, 2020-06-13:
Thanks for the quick reply (as always). I'll report back if I learn anything useful.

jpasqua, 2020-06-14:
As it turns out, it's easy to reproduce this issue with just two files all in the sketch directory. No subdirectories. The key is having a .ino file and a .cpp file. As long as all the files are .ino's in the sketch directory, they are merged and compiled as a single translation unit. Thus there is only one definition of each GFXfont even if all of the .ino's separately include TFT_eSPI.h.

As soon as one file is a .cpp, it is compiled separately and pulls in a second definition each referenced font cause the program size to increase. The following minimal example demonstrates the issue. You don't need to run it, just compile it and see how the size goes up the more references there are to fonts in SecondFile.cpp. If you change the name to SecondFile.ino, the program size will not increase as long as it references the same fonts as in Foo.ino.

````
---- Foo.ino:

#include "TFT_eSPI.h"

extern void foo();

// Use hardware SPI
TFT_eSPI tft = TFT_eSPI();

void setup(void) {
  tft.begin();
  foo();
}

void loop() {
  tft.setFreeFont(&FreeSerifBold9pt7b);
  tft.setFreeFont(&FreeSerifBold12pt7b);
  // tft.setFreeFont(&FreeSerifBold18pt7b);
}

---- SecondFile.cpp:
#include "TFT_eSPI.h"

extern TFT_eSPI tft;

void foo() {
  tft.setFreeFont(&FreeSerifBold9pt7b);
  tft.setFreeFont(&FreeSerifBold12pt7b);
  // tft.setFreeFont(&FreeSerifBold18pt7b);
}
````

jpasqua, 2020-06-14:
Here's what I ended up doing in case it might help others.
* The key is to only reference GFXfonts from a single module in the project.
* Since all .ino files at the top level are merged and compiled together, you can spread references across .ino files
* I didn't want to do that in my project because it resulted in too much clutter and was hard to do with a mix of headers and implementation files.
* Instead, I created a wrapper function for setFreeFont (for both TFT_eSPI objects and sprites) and put them in a single module
  * All other modules call that wrapper function with an index that identifies the font to be used.
  * In this way,  none of them reference the GFXfonts directly.
  * The one module that does reference GFXfonts causes the appropriate ones to be linked in and the overall program size is minimized.