Bodmer/TFT_eSPI #665: ESP32 / Wemos D32 Pro | Guru Meditation Error

Handgasaffe, 2020-06-14:
`Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.`

I do always have this issue while trying to run my xpt2046 touchscreen. 

Does anybody know something  about it?

Bodmer, 2020-06-15:
Run the "Test_Touch_Controller" sketch on a bare board (nothing connected to it). Does it run and report values to the serial monitor? If it crashes then it appears that bad GPIO pins have been allocated in the setup file, perhaps they conflict with the hardware design, check the schematic..

Connect you display with touch controller and run the sketch again. Do the numbers in the monitor window change when the screen is touched?

Handgasaffe, 2020-06-16:
No nothing changed.

I disconnected everything from the board and run the program again. The same error as mentioned appeared.

Bodmer, 2020-06-16:
How odd, there must be a pin function conflict. Please post the setup file you have used so that I can see the configuration you are using.

Handgasaffe, 2020-06-16:
So I am a beginner, I hope this is what you wanted to have:)

`[env:lolin_d32_pro]
platform = espressif32
board = lolin_d32_pro
framework = arduino
upload_port = /dev/tty.usbserial-1410
lib_deps = 
    ;Adafruit GFX Library
    ;Adafruit ST7735 and ST7789 Library
    ;Wire
    ;Adafruit ILI9341
    ;XPT2046 Touchscreen
    TFT_eSPI
    
lib_ignore = 
    SD
build_flags =
  -Os
  -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG
  -DUSER_SETUP_LOADED=1
  -DILI9163_DRIVER=1
  -DTFT_WIDTH=320
  -DTFT_HEIGHT=240
  -DTFT_MISO=6
  -DTFT_MOSI=11
  -DTFT_SCLK=13
  -DTFT_CS=10
  -DTFT_DC=9
  -DTFT_RST=8
  -DLOAD_GLCD=1
  -DTOUCH_CS=12`

Bodmer, 2020-06-16:
You need to set the pins to those that are available on the board! This is the cause of the reset.

[Check the documentation](https://docs.wemos.cc/en/latest/d32/d32_pro.html). For example TFT_SCLK is typically GPIO 18.

Handgasaffe, 2020-06-16:
Changed all pins now, same error appeared:(

`build_flags =
  -Os
  -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG
  -DUSER_SETUP_LOADED=1
  -DILI9163_DRIVER=1
  -DTFT_WIDTH=320
  -DTFT_HEIGHT=240
  -DTFT_MISO=19
  -DTFT_MOSI=23
  -DTFT_SCLK=22
  -DTFT_CS=14
  -DTFT_DC=27
  -DTFT_RST=33
  -DLOAD_GLCD=1
  -DTOUCH_CS=12`

Bodmer, 2020-06-16:
Do you have an ILI9163 display? If so these are typically 128x160 pixels and do not have a touch screen. For the TFT_SCLK line pin 18 would be a better choice.

Have you got any software running on the board?

Unfortunately these potential issues with the settings would not cause the error so you need to check on the PlatformIO forum to see if anyone else has had any issues with that ESP32 board.

I expect you will find it works fine with the Arduino IDE environment in which case the PlatformIO environment settings might be the problem.

Handgasaffe, 2020-06-16:
I so have a ILI9341  2,8“ Display with 320x240 Pixels and a touchscreen with the xpt2046 driver. 

Of Course, i got another TFT Display, the 1.4“, running.

The Display depicts my complete source code. Only Touch function is not working.

Bodmer, 2020-06-17:
What happens with the display you have running if you define a touch pin? viz -DTOUCH_CS=12

Handgasaffe, 2020-06-23:
Now I do always get the same values:

x: 8191     y: 8191     z: 4095 

no matter if I touch or not.

Bodmer, 2020-06-28:
Have you tried using the Arduino IDE?

Handgasaffe, 2020-06-29:
No, only tried the Platform IO workspace. Is it different with the arduino?

Bodmer, 2020-06-29:
Users have reported success with PlatformIO but I only test and run using Arduino environment.

Handgasaffe, 2020-07-05:
Never used it. I will try to fix it in PlatformIO. And of nothing works, I will Uwe the Arduino IDE environment.
Thanks

closed 2020-07-05 by Handgasaffe