Bodmer/TFT_eSPI #666: Possible to rotate text yet?

Gun-neR, 2020-06-15:
I have a TTGO ESP32 T-Cam Plus with LCD and camera. I swapped the camera for one I want to use, but it has a different sensor orientation. I can rotate the whole screen, but the text is always out 90 degrees from the camera image. Can't change the sensor option, short of replacing camera with one with long cable and physically twisting it around... doh... so rotating text is other, logical, option.

I have read how one option is to convert text to bitmap then pass that into SPIFFS and rotate it... but I was lost at the "convert to bitmap" part :P so hoping to use a more simple "text rotate" command :D

Bodmer, 2020-06-15:
You can rotate text by using a Sprite. See "Rotated_Sprite_X" examples.

You can use 1bpp Sprites to save memory. Note you can define a colour as transparent as note in the examples.

closed 2020-06-15 by Bodmer

Gun-neR, 2020-06-15:
OK, I think I get it now... Thanks.