Bodmer/TFT_eSPI #667: ER-TFTM035-6 Color Issue

tpitman, 2020-06-15:
I have an ESP32 connected via SPI to a ER-TFTM035-6 display. It is showing stuff on the screen, but the colors are wrong.

I think the R and B might be swapped.

I am using the UTFT_Demo_480x320 to double check.

When I read the code it seems to indicate that the final screen where it says the demo is done I think it is saying that the background should be TFT_BLUE, the message area in the middle should be TFT_RED and the text should be TFT_WHITE with TFT_RED background.

Well, the full screen background is yellow, the center rounded rect is cyan and the text is black (as far as I can tell it is black.)

This screen has the ILI9488 driver. I didn't change anything in the setup other than putting in the correct SPI and other pins. Since this is SPI and not parallel I am not sure how wiring could come into it, so assume it is a config issue.

Bodmer, 2020-06-15:
Try the Colour_Test diagnostic sketch.

Adding tft.invertDisplay( false ); or tft.invertDisplay( true ); in  the setup() section will correct the problem.

I will see if I can find my ILI9488 serial and parallel displays and test the latest library. It may be that SPI and parallel displays require a different setting.

tpitman, 2020-06-15:
I will be able to give this a try a little later and let you know. Thank you!

tpitman, 2020-06-16:
I was able to use the color test to see that invert would work and was able to make that change in my code. That fixes it. Thank you!

closed 2020-06-16 by tpitman