Bodmer/TFT_eSPI #668: "Erroe compiling for board ESP32 Dev Module"

billd39ET, 2020-06-17:
HI, great looking library, thnx! I'm having trouble at the very fist hurdle . . . none of the example sketches will compile for ESP32 Dev Module (or any board that I select)

Arduino IDE 1.8.12
ESP32 Dev Module
MCU Friend TFT ILI9325 (I haven't got that far yet . . .)
Sketch: TFT_Print+Test.ino

My User_Setup.h
////////////////////////////////////////////////////////////////////////////////////////////////////////////
#define ILI9341_DRIVER

#define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
//#define TFT_RST   4  // Reset pin (could connect to RST pin)
#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

#define TFT_PARALLEL_8_BIT

// The ESP32 and TFT the pins used for testing are:
#define TFT_CS   33  // Chip select control pin (library pulls permanently low
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin, toggles on startup

#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    2  // Read strobe control pin

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits.
#define TFT_D2   26  // Pins can be randomly assigned, this does not affect
#define TFT_D3   25  // TFT screen update performance.
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000 // Actually sets it to 26.67MHz = 80/3
#define SPI_READ_FREQUENCY  20000000

#define SPI_TOUCH_FREQUENCY  2500000
////////////////////////////////////////////////////////////////////////////////////////////////////////////////

The error output below is for TFT_Print_Test.ino, but I get multiple errors, failed to compile no matter what example sketch I use.
////////////////////////////////////////////////////////////////////////////////////////////////////////////////
In file included from C:\Users\billd\OneDrive\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:20:0:

C:\Users\billd\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c: In member function 'bool TFT_eSPI::initDMA()':

C:\Users\billd\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:666:20: error: 'TFT_MOSI' was not declared in this scope

     .mosi_io_num = TFT_MOSI,

                    ^

C:\Users\billd\OneDrive\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:668:20: error: 'TFT_SCLK' was not declared in this scope

     .sclk_io_num = TFT_SCLK,

                    ^

exit status 1
Error compiling for board ESP32 Dev Module.
//////////////////////////////////////////////////////////////////////////////////////////////////////////////

Any help much appreciated.

thnx
billd

Bodmer, 2020-06-17:
You must choose either a serial (SPI) or parallel (8 bit) type display and not try to use both in the setup.

I am guessing you have a parallel mcufriend style UNO board in which case you should start with [this setup](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup14_ILI9341_Parallel.h).

billd39ET, 2020-06-18:
Thnx. 

I found that (and other pre-defined user setups after I posted, this (and a few others) compiled OK but no output to the display. I suspect that it's an issue with my TFT driver - ILI9325 - it's not specifically defined in any of your libraries, I'll keep going trial-and error.

thnx
again

Bodmer, 2020-07-09:
Assumed solved, closing.

closed 2020-07-09 by Bodmer

bobburudinesh, 2021-03-27:
Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Users\bobbu\Documents\Arduino\TouchScreen\TouchScreen.ino: In function 'void loop()':

TouchScreen:110:25: error: 'class TFT_eSPI' has no member named 'getTouch'

   boolean pressed = tft.getTouch(&t_x, &t_y);   

                         

C:\Users\bobbu\Documents\Arduino\TouchScreen\TouchScreen.ino: In function 'void touch_calibrate()':

TouchScreen:237:9: error: 'class TFT_eSPI' has no member named 'setTouch'

     tft.setTouch(calData); 

         

TouchScreen:256:9: error: 'class TFT_eSPI' has no member named 'calibrateTouch'

     tft.calibrateTouch(calData, TFT_MAGENTA, TFT_BLACK, 15);

         

exit status 1

'class TFT_eSPI' has no member named 'getTouch'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Bodmer, 2021-03-27:
I am guessing you have an 8 bit parallel display. The library does not provide touch support for parallel displays, you can use alternative libraries, see [ReadMe](https://github.com/Bodmer/TFT_eSPI#8-bit-parallel-support).

bobburudinesh, 2021-03-28:
No sir, I am using it in SPI mode
1.Basically I need have 4 buttons on ILI9341(2.8 Inch 240x320 display) when each one of them is pressed different action should be taken place.
2.I have an ESP32 Dev Module,2.8 Inch SPI Touch Screen Module TFT Interface 240*320
This is the link of that TFT Module:
https://robu.in/product/2-8-inch-spi-touch-screen-module-tft-interface-240320/?gclid=CjwKCAjwr_uCBhAFEiwAX8YJgQHoEan1sU4JDCywGPlIbrAvoYkc0vm86lwDYHdcTduiIDQwzP9y-RoC_-kQAvD_BwE

Bodmer, 2021-03-28:
Check your setup file within the library has TOUCH_CS defined.

Run the Read_User_Setup diagnostic sketch to verify the compiler is picking up the correct pins and that you have correctly eddited the setup file.

I expect you realise the touch controller pins need to be wired up to the SPI bus.

bobburudinesh, 2021-03-28:
Yes Sir when I compiled for first time the code got compiled and I dumped it on to esp32 and I made connections as follows
VCC: Vin
GND: Gnd
CS: D5
RESET: D4
DC: D2
SDI(MOSI): D23
SCK: D18
SDOK(MISO): D19
T_CLK: D18
T_CS: D21
T_DIN: D19
T_DO: D23
T_IRQ: Not Connected
I am Getting white Screen on my Display When I dump code for Keypad_240x320,On_Off_Button
Nothing is Printing in serial monitor even serial monitor baud(9600) is also correct.

Bodmer, 2021-03-28:
The Read_User_Setup sketch uses 115200 baud, [see here.](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Test%20and%20diagnostics/Read_User_Setup/Read_User_Setup.ino#L31-L32)

bobburudinesh, 2021-03-28:
This is what I got after running Read_User_Setup on Serial monitor

23:26:09.665 -> 
23:26:09.665 -> [code]
23:26:09.665 -> TFT_eSPI ver = 2.3.61
23:26:09.665 -> Processor    = ESP32
23:26:09.665 -> Frequency    = 240MHz
23:26:09.665 -> Transactions = Yes
23:26:09.665 -> Interface    = SPI
23:26:09.665 -> Display driver = 9341
23:26:09.665 -> Display width  = 240
23:26:09.665 -> Display height = 320
23:26:09.665 -> 
23:26:09.665 -> MOSI    = GPIO 23
23:26:09.665 -> MISO    = GPIO 19
23:26:09.665 -> SCK     = GPIO 18
23:26:09.665 -> TFT_CS   = GPIO 15
23:26:09.665 -> TFT_DC   = GPIO 0
23:26:09.665 -> TFT_RST  = GPIO 2
23:26:09.665 -> TOUCH_CS = GPIO 21
23:26:09.665 -> 
23:26:09.665 -> Font GLCD   loaded
23:26:09.665 -> Font 2      loaded
23:26:09.665 -> Font 4      loaded
23:26:09.665 -> Font 6      loaded
23:26:09.665 -> Font 7      loaded
23:26:09.665 -> Font 8      loaded
23:26:09.665 -> Smooth font enabled
23:26:09.665 -> 
23:26:09.665 -> Display SPI frequency = 40.00
23:26:09.711 -> Touch SPI frequency   = 2.50
23:26:09.711 -> [/code]

bobburudinesh, 2021-03-28:
[image]
I uncommented the circled part, TOUCH_CS and I followed connections as they were mentioned in header file User_Setup.h

Bodmer, 2021-03-28:
The default setup is for en ESP8266 so you must also [comment out or delete these lines](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L159-L161).

bobburudinesh, 2021-03-28:
> The default setup is for an ESP8266 so you must also [comment out or delete these lines](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L159-L161).

This part I have Commented out sir

bobburudinesh, 2021-03-28:
> This is what I got after running Read_User_Setup on Serial monitor
> 
> 23:26:09.665 ->
> 23:26:09.665 -> [code]
> 23:26:09.665 -> TFT_eSPI ver = 2.3.61
> 23:26:09.665 -> Processor = ESP32
> 23:26:09.665 -> Frequency = 240MHz
> 23:26:09.665 -> Transactions = Yes
> 23:26:09.665 -> Interface = SPI
> 23:26:09.665 -> Display driver = 9341
> 23:26:09.665 -> Display width = 240
> 23:26:09.665 -> Display height = 320
> 23:26:09.665 ->
> 23:26:09.665 -> MOSI = GPIO 23
> 23:26:09.665 -> MISO = GPIO 19
> 23:26:09.665 -> SCK = GPIO 18
> 23:26:09.665 -> TFT_CS = GPIO 15
> 23:26:09.665 -> TFT_DC = GPIO 0
> 23:26:09.665 -> TFT_RST = GPIO 2
> 23:26:09.665 -> TOUCH_CS = GPIO 21
> 23:26:09.665 ->
> 23:26:09.665 -> Font GLCD loaded
> 23:26:09.665 -> Font 2 loaded
> 23:26:09.665 -> Font 4 loaded
> 23:26:09.665 -> Font 6 loaded
> 23:26:09.665 -> Font 7 loaded
> 23:26:09.665 -> Font 8 loaded
> 23:26:09.665 -> Smooth font enabled
> 23:26:09.665 ->
> 23:26:09.665 -> Display SPI frequency = 40.00
> 23:26:09.711 -> Touch SPI frequency = 2.50
> 23:26:09.711 -> [/code]
> 
> In this its showing to connect TFT_DC=GPIO 0,TFT_RST = GPIO 2 but in header it is mentioned TFT_DC=GPIO 2,TFT_RST = GPIO 4,The esp32 devkit which I am using has only 32 pins and there is no access to GPIO 0,So how can I assign TFT_DC to other GPIO.
> If I change in User_Setup.h and run Read_User_Setup Still its printing TFT_DC=GPIO 0

This is the issue I am facing sir, can you please tell me how to assign TFT_DC=GPIO 0 to other GPIO

Bodmer, 2021-03-28:
Post your entire User_Setup.h file as a .zip file (Just drag zip file to the window here).

bobburudinesh, 2021-03-29:
[User_Setup.zip](https://github.com/Bodmer/TFT_eSPI/files/6219192/User_Setup.zip)

Bodmer, 2021-03-29:
The User_Setup file looks good. The DC and RST signals are allocated to GPIO 2 and 4.

Since this does not agree with the Read_User_Setup output a different header file must be selected.

Post your User_Setup_Select.h file as a zip. This can also be found inside the TFT_eSPI library folder.

bobburudinesh, 2021-03-29:
[Uploading User_Setup_Select.zip…]()

bobburudinesh, 2021-03-29:
[User_Setup_Select.zip](https://github.com/Bodmer/TFT_eSPI/files/6220969/User_Setup_Select.zip)

Bodmer, 2021-03-29:
Comment out line 24 in User_Setup_Select.h
Take note of comment on line 20
Pins are then defined in User_Setup.h
Run Read_User_Setup sketch to verify the correct pins are being used.

bobburudinesh, 2021-03-29:
> Comment out line 24 in User_Setup_Select.h
> Take note of comment on line 20
> Pins are then defined in User_Setup.h
> Run Read_User_Setup sketch to verify the correct pins are being used.

Sir, I followed all this and I am still facing the problem.

bobburudinesh, 2021-03-29:
It worked Sir, Thankyou So much for the response, Now I Dumped  Keypad_240x320 its just displaying button but Touch is not responding

bobburudinesh, 2021-03-29:
It worked Sir, Thankyou So much for the response, Now I Dumped Keypad_240x320 its just displaying buttons but Touch is not responding.
[image]

Bodmer, 2021-03-29:
If the values reported by Test_Touch_Controller change significantly when you touch the screen then it is working.

I assume you have wired up the touch controller pins... list all connections you have made from the ESP32 to the screen. T_DO -> TFT_MISO
T_DIN -> TFT_MOSI
T_SCK -> TFT_SCK
T_CS -> TOUCH_CS

The keypad example should run through a calibration test where you are asked to touch the screen in the requested positions.

bobburudinesh, 2021-03-30:
> If the values reported by Test_Touch_Controller change significantly when you touch the screen then it is working.
>
> I assume you have wired up the touch controller pins... list all connections you [...]

It didn't run through a calibration test for me, I waited for 2 minutes its only the screen with buttons displayed on it touch is not responding

Bodmer, 2021-03-30:
Based on an earlier post it looks like you have T_DO and T_DIN connected incorrectly.

bobburudinesh, 2021-03-31:
Sir do I have any function to change brightness of display and I want to display some ICONS(like poweron, poweroff, settings Icon) without using SD card.

bobburudinesh, 2021-04-02:
Sir, How do I control backlight intensity of ILI9341 display using esp32?
I tried using PWM Signal  by setting frequency as 5KHz,resolution as 8 bit, I was getting some thing like this
[image]

Bodmer, 2021-04-03:
Brightness control is not a function built into the library. It has to be implemented in you sketch. Start with simple on/off control.

bobburudinesh, 2021-04-03:
I am able to achieve on/off by sending high and low signal on that GPIO but I want to control it at different steps using PWM signal. When I control using PWM I display is getting pixelated as shown in picture.
[image]

Bodmer, 2021-04-03:
Make sure you are toggling the right pin.

Does the backlight LED brightness change?

If you control the ESP32 onboard LED instead of the backlight, does it work as expected?

The problem may be due to signal coupling as a large current is being switched on/off frequently. Add more 0V (GND) wires and ideally shorten all the wires to reduce the inductive coupling.

bobburudinesh, 2021-04-04:
yeah when I control on board LED brightness is changing according to PWM Signal

Bodmer, 2021-04-04:
Have you changed the wiring as suggested?

bobburudinesh, 2021-04-04:
yes sir

bobburudinesh, 2021-04-07:
Sir I want to interface 2.8" ILI9341 Modules(SPI) with esp32 both displays will be placed at different places but both will be displaying same information and they should respond accordingly if touch on any one of the display.
For now I Shorted VCC,GND,CS,RESET,DC,MOSI,SCK,LED,MISO of both Displays and I am able to display same content on both displays, But I want to include touch feature also of both Displays how can I do it Sir?

Bodmer, 2021-04-07:
Yes, but you have to manage the two chip selects from your sketch. The approach is to set the pin in the setup file to -1

In your sketch when you want to check the touch controller, drive the correct chip select low with digitalWrite (see blink example, so not forget the mode command to set pins to output) , get the touch data then drive the correct chip select high. Do not draw graphics on the TFT with either touch chip select low. Only drive one chip select low at a time.

bobburudinesh, 2021-04-07:
OK sir, there are 2 chip selects right one is (CS,T_CS) which one are you mentioning about Sir, Can you Please send the connections or connection diagram.

Bodmer, 2021-04-07:
The CS line is for the TFT. The T_CS line is for the touch controller.

You need to connect display 1 T_CS to a GPIO of your choice and control it from your sketch.
You need to connect display 2 T_CS to a different GPIO of your choice and control it from your sketch.

bobburudinesh, 2021-06-28:
Sir I tried this but its not working can you suggest me other method or send example code.
Thanks in advance.