Bodmer/TFT_eSPI #671: [ili9341]: switch off ili9341 display

hasenradball, 2020-06-21:
Hi,

is it possible to switch off the display?
For example over night to switch off the display via an function similar turn off backlight of an LCD?

Bodmer, 2020-06-24:
There is a command to put the display into low power mode but there is not feature in the library to do this. You can send commands to the display from the sketch. This should work:

tft.writecommand(0x10); // Sleep
delay(5); // Delay for shutdown time before another command can be sent

tft.writecommand(0x11); // Wake display
delay(120); // Delay for pwer supplies to stabilise

This command puts the TFT ship into a lower power mode, the TFT graphics should stil be stored in the screen RAM. The backlight is a set of white LEDs and so that supply pin will also need to be controlled to turn off the backlighting.

hasenradball, 2020-06-27:
Thanks for your reply I will test it and give you a feed back.
If it works is it possible to add in the library?

hasenradball, 2020-06-29:
Hi,

I tested the functions.
1.) Sleep switches of the data of the screen but let the Background LED on, is there an possibity to turn off the Background LED too?

2.) wake displays the data on the screen again, ok so.

So if I had a comand to switch off/on the Background LED it would be great ;-)

I found other things.
With the ili9341 and the ESP8266 spi Freq of 40 MHz don't work for me. I need 27 MHz

Bodmer, 2020-06-29:
Only a few displays have the required transistor switch. If your does then it can be controlled by your sketch by using digitalWrite. As on/off control is so simple to do in a sketch I do not plan to add a backlight control function.

closed 2020-06-29 by Bodmer

M4rc3lv, 2023-08-21:
The ILI9341 displays have a BL connection which is normally wired to 3,3V Vcc. You can wire that BL connection to a GPIO pin of your micro controller instead and then use that pin to switch the backlight on or off. When switched off the display is completely blacked out.