Bodmer/TFT_eSPI #672: Display drivers for 4" 800x480 OTM8009A / NT35510 displays in parallel 8-bit bus mode / WIP / need help

Mike120Git, 2020-06-24:
[New display variant] / [Question] / [Potential enhancement]
Hi!
I am working on creating display drivers for 4" 800x480 OTM8009A / NT35510 based displays in parallel 8-bit bus mode.

Good news is that both of these displays do work correctly with 8-bit bus - I have verified it with ESP32 (and parts of code taken from your library).

After a lot of trials and errors I discovered that these displays just expect 16-bit commands and 16-bit data, 
sent as two bytes (tft_Write_16).
On the hardware side - LCD need to be switched to 8-bit bus mode by desoldering R3 resistor and soldering it as R2.

I have successfully created appropriate config files for these displays (_Defines.h, _Init.h, _Rotation.h).

**There is just one problem: these displays need driver specific 16-bit commands to set row / column address.**
The generic code to set TFT_CASET / TFT_PASET in methods setWindow, drawPixel, readAddrWindow need to be overriden for them.

And I don't see any way to inject display-specific code into the methods above from a display driver, so I need your help... :-)
If you could just provide some hook point to override current hardcoded TFT_CASET / TFT_PASET code from a display driver code, please?

I would be very happy, if you could help me with integration of these displays with your library.
I have seen many questions about these displays on forums, but it looks like that nobody managed to drive them by 8-bit bus successfully before...
The dispalys themselves are really good quality, with wide viewing angles and pretty high resolution - they are undoubtedly worth supporting.

Greetings,
Mike

--------------------------------------------------------------------------------
More info about these displays:
--------------------------------------------------------------------------------

3.97inch 16BIT Module OTM8009A SKU:MRB3971
http://www.lcdwiki.com/3.97inch_16BIT_Module_OTM8009A_SKU:MRB3971

3.97inch 16BIT Module NT35510 SKU:MRB3973
http://www.lcdwiki.com/3.97inch_16BIT_Module_NT35510_SKU:MRB3973

Resolution:	800*480
Module Interface:	16Bit parallel interface (by default, can be switched to 8-bit) 

The module hardware supports 8-bit and 16-bit parallel port data bus mode switching, as follows:

- Solder R3 and R4 with 0Ω resistor or short circuit directly, and disconnect R2 and R6: select 16-bit data bus mode (default), use DB0-DB15 data pin

- Solder R2 and R4 with 0Ω resistor or short circuit directly, and disconnect R3 and R6: select 8-bit data bus mode, use DB0-D7 data pin

--------------------------------------------------------------------------------
OTM8009A_Init.h with defined 16-bit commands:
--------------------------------------------------------------------------------

```
// Configure OTM8009A display

#define Byte8H(ByteH) ((uint8_t)(((uint16_t)(ByteH)&0xFF00)>>8))
#define Byte8L(ByteL) ((uint8_t)( (uint16_t)(ByteL)&0x00FF))

#define writecommand16(cmd) \
	writecommand(Byte8H(cmd)); writecommand(Byte8L(cmd))

#define writedata16(data) \
	writedata(Byte8H(data)); writedata(Byte8L(data))

//3.97inch OTM8009 Init 20190116
	/* Enter CMD2 */
	writecommand16(MCS_CMD2_ENA1);
	writedata16(0x80);
	writecommand16(MCS_CMD2_ENA1 + 1);
	writedata16(0x09);
	writecommand16(MCS_CMD2_ENA1 + 2);
	writedata16(0x01);
	
//... more init code
```
--------------------------------------------------------------------------------
Display specific TFT_CASET / TFT_PASET code:
--------------------------------------------------------------------------------

```
  // // No need to send x if it has not changed (speeds things up)
  // if (addr_col != (x0<<16 | x1)) {
  //   DC_C; tft_Write_8(TFT_CASET);
  //   DC_D; tft_Write_32C(x0, x1);
  //   addr_col = (x0<<16 | x1);
  // }

  // // No need to send y if it has not changed (speeds things up)
  // if (addr_row != (y0<<16 | y1)) {
  //   DC_C; tft_Write_8(TFT_PASET);
  //   DC_D; tft_Write_32C(y0, y1);
  //   addr_row = (y0<<16 | y1);
  // }

  DC_C; tft_Write_16(0x2A00); 
  DC_D; tft_Write_16(Byte8H(x0));
  DC_C; tft_Write_16(0x2A01); 
  DC_D; tft_Write_16(Byte8L(x0));
  DC_C; tft_Write_16(0x2A02); 
  DC_D; tft_Write_16(Byte8H(x1));
  DC_C; tft_Write_16(0x2A03); 
  DC_D; tft_Write_16(Byte8L(x1));

  DC_C; tft_Write_16(0x2B00); 
  DC_D; tft_Write_16(Byte8H(y0));
  DC_C; tft_Write_16(0x2B01); 
  DC_D; tft_Write_16(Byte8L(y0));
  DC_C; tft_Write_16(0x2B02); 
  DC_D; tft_Write_16(Byte8H(y1));
  DC_C; tft_Write_16(0x2B03); 
  DC_D; tft_Write_16(Byte8L(y1));

    
  DC_C; tft_Write_16(0x2C00);
  DC_D;

```

Bodmer, 2020-06-25:
Thanks for the information.

The tft_Write_8(C) macro is only used by the library to send commands, so for the Raspberry Pi (SPI) display which needed 16 bit commands I just redefined the 8 bit macro to send 16 bits.  However I can see that different commands are needed for each x,y parameter so that method will not work for you.

If you wish to create a push request or disclose all the changes needed than I will add this display. Ideally the tft read code will also need to be adapted and tested.

Mike120Git, 2020-06-26:
I will create push request soon, just need some time to learn GitHub project contribution basics first... :-)
I have an idea - to define macros for setting TFT_PASET and TFT_PASET and use them in the methods, for example:

```
#ifndef TFT_CASET_CMD
	#define TFT_CASET_CMD(x0, x1) \
		DC_C; tft_Write_8(TFT_CASET); \
		DC_D; tft_Write_32C(x0, x1)
#endif

#ifndef TFT_PASET_CMD
	#define TFT_PASET_CMD(y0, y1) \
		DC_C; tft_Write_8(TFT_PASET); \
		DC_D; tft_Write_32C(y0, y1)
#endif
```

That would give an opportunity to define some driver specific implementation of TFT_CASET_CMD / TFT_PASET_CMD in the LCD _Init.h file.
If no definition of TFT_CASET_CMD / TFT_PASET_CMD is provided by the driver, than current default implementation would be used.
What do you think about it?

Bodmer, 2020-06-28:
I think the macro approach is the way to go.  I suggest the following route to adding this display.

If you can provide the following as part of stage 1 changes:

- A replacement setWindow function that just supports the display.
- A list of I/O macros, maybe already in the above posts?
- A set of files to put in the TFT_Drivers folder
- Info on any other changes needed.

With that info I can consider how best to adapt the library with minimally intrusive changes to the core code.

Once that is tested and running in the library then stage 2 changes would be, if read is supported by the display:
- A replacement readAddrWindow function
- Adaptions to read CGRAM functions (readPixel readRect etc)](url)

Bodmer, 2020-06-28:
It is not clear if both TFT types will work with the same changes, so I think two #defines will be needed:

#define OTM8009A_DRIVER
and
#define NT35510_DRIVER

Are you able to test both or just one of these displays?

Mike120Git, 2020-07-12:
I have forked the project, my copy is avaliable here: [https://github.com/Mike120Git/TFT_eSPI](https://github.com/Mike120Git/TFT_eSPI)

- TFT_Drivers files for both of the displays are ready, two new display #defines are needed
- I/O macros are defined in _Defines.h files
- Here is a modified the setWindow function that uses macros:
```
void TFT_eSPI::setWindow(int32_t x0, int32_t y0, int32_t x1, int32_t y1)
{
  //begin_tft_write(); // Must be called before setWindow

#ifdef CGRAM_OFFSET
  x0+=colstart;
  x1+=colstart;
  y0+=rowstart;
  y1+=rowstart;
#endif

#ifdef MULTI_TFT_SUPPORT
  // No optimisation to permit multiple screens
  TFT_CASET_CMD(x0, x1);
  TFT_PASET_CMD(y0, y1);
#else
  // No need to send x if it has not changed (speeds things up)
  if (addr_col != (x0<<16 | x1)) {
    TFT_CASET_CMD(x0, x1);
    addr_col = (x0<<16 | x1);
  }

  // No need to send y if it has not changed (speeds things up)
  if (addr_row != (y0<<16 | y1)) {
    TFT_PASET_CMD(y0, y1);
    addr_row = (y0<<16 | y1);
  }
#endif

  TFT_RAMWR_CMD;
  DC_D;

  //end_tft_write(); // Must be called after setWindow
}

```
I have made a few absolute minimal modifications to the TFT_eSPI.cpp file to make it work with the new displays.
Here is the diff of my changes:
[https://github.com/Mike120Git/TFT_eSPI/commit/8bce569788b4782142aa2fbf3a64dd98bcf1aca0#diff-a30868faa3e646e6ae51eab6ef6450f3](https://github.com/Mike120Git/TFT_eSPI/commit/8bce569788b4782142aa2fbf3a64dd98bcf1aca0#diff-a30868faa3e646e6ae51eab6ef6450f3)

I own both variants of the displays (OTM8009A and NT35510) and I will be able to use them for testing.
I have bought both of them here: https://www.aliexpress.com/item/32807629460.html 
Currently NT35510 variant is in stock.

There are some issues I can think of:
- since the commands for the new displays have lengthh of 16-bits this code will not work properly with them:

```
#ifdef TFT_RST
  if (TFT_RST >= 0) {
    digitalWrite(TFT_RST, HIGH);
    delay(5);
    digitalWrite(TFT_RST, LOW);
    delay(20);
    digitalWrite(TFT_RST, HIGH);
  }
  else writecommand(TFT_SWRST); // Software reset - this will not work
#else
  writecommand(TFT_SWRST); // Software reset - this will not work
#endif
```

...

```
#ifdef TFT_INVERSION_ON
  writecommand(TFT_INVON);
#endif

#ifdef TFT_INVERSION_OFF
  writecommand(TFT_INVOFF);
#endif
```
...
The readcommand methotds won't work also I suppose.

I have tested some basic drawing functions and JPG drawing with  tft.pushImage - looks good.
Image rotation also works.

I hope my fork will be useful as source of stage 1 changes.
If you need any more info, I will do my best to help, just give me some time.

Bodmer, 2020-07-12:
Thanks, I am busy with other paid for projects at the moment so it will be 3+ weeks before I can look at this in detail but it looks very promising. Reading from the display is a "nice to have" feature but is not critical as not all displays can be read anyway (eg Raspberry Pi based ones).

Guerillah, 2020-12-05:
Nice Mike. If you don't mind, can you run TFT_graphicstest_one_lib and report back the results? I'd certainly appreciate it. Thanks!

closed 2020-12-16 by Bodmer