Bodmer/TFT_eSPI #673: Dynamic Text pixelated in 320 x 240 TFT_Meters

tinwinaung, 2020-06-25:
I am using ESP32 and ILI9341.
When I run Example >320 x 240 >TFT_Meters, I got pixelated text for dynamic text labels.  I got the same issue in other examples when I use background parameter in setTextColor method (A0 A1 ... labels in the below picture require to remove background parameter in method to be able to view clearly. I try to use Sprite in the same line in the code but also got the same issue).

[image]

Bodmer, 2020-06-25:
I am guessing you have hand wired the display as 8 bit parallel. It is likely the problem is that the wires are too long and/or there is not a good GND connection from the display to the ESP32. The logic rise and fall times are fast on the ESP32 so short connections are necessary particularly for GND. Try shorter multiple wires for the GND.

It is best to use a UNO style display with an UNO style ESP32 to avoid these issues, those boards provide a good ground plane and low impedance paths for the high edge speeds to propagate without edge ringing..

tinwinaung, 2020-06-25:
You are right 👍. 
I am using cable and 8 bit parallel (the worst is that my display is 16 bit 👎 and using PSEUDO 8 BIT wiring).
I will need to test diff configuration.

By the way, I changed below code (in current HW config) and **dynamic text** looks **good** _(there have ghost in picture because of camera shutter speed)._

Basically adding  size to 2 and font to 1 give me the result some how.
```
void plotPointer(void){
// ..... codes
tft.setTextColor(BM_GREEN,BM_BLACK);
tft.setTextSize(2);
tft.drawRightString(buf, (i * 40 + 36 - 5), 187 - 27 + 155 - 18,1);
// ..... codes
}
```

But I think I just lucky with this values.

[image]

tinwinaung, 2020-06-25:
Potential hardware configuration issue

closed 2020-06-25 by tinwinaung