Bodmer/TFT_eSPI #675: 3.5 Inch MHS Support - MHS3528

dirtbikr, 2020-06-27:
Hello,

Thank you for the awesome library! 

I am trying to get the MHS 3.5 inch Raspberry Pi displays [(http://www.lcdwiki.com/MHS-3.5inch_RPi_Display)](url) working at 80MHz from an ESP32. They run fine at 40MHz, but I would like to use the full capacity of the ESP32's SPI bus since this display claims to support it.

To make this display function at 40MHz, all that was required was to select the RPI_ILI9486_DRIVER and set the SPI_FREQUENCY to 40000000. I figured since the display theoretically supports up to 125MHz that setting the SPI_FREQUENCY to 80000000 would work, but I end up with a white screen. 

All pin configurations are default in User_Setup for an ESP32 dev kit in SPI mode. I have gone through and commented out everything in the DC and CS pin drive code except for the default setup with TFT_DC and TFT_CS > 0, which works fine for this display (below code blocks are the only lines left in the pin drive code sections of TFT_eSPI_ESP32.h).

`#define DC_C GPIO.out_w1tc = (1 << TFT_DC)//;GPIO.out_w1tc = (1 << TFT_DC)`
`define DC_D GPIO.out_w1ts = (1 << TFT_DC)//;GPIO.out_w1ts = (1 << TFT_DC)`

`define CS_L GPIO.out_w1tc = (1 << TFT_CS);GPIO.out_w1tc = (1 << TFT_CS)`
`define CS_H GPIO.out_w1ts = (1 << TFT_CS);//GPIO.out_w1ts = (1 << TFT_CS)`

Switching the 8 bit write code from `#define tft_Write_8(C) TFT_WRITE_BITS((C)<<8, 16)` to `#define tft_Write_8(C) TFT_WRITE_BITS(C, 8)` stops the display from working at any frequency using the "TFT_graphicstest_one_lib" example. 

Do you have any recommendations to try to make this screen work at 80MHz? I can have one shipped to you if you would like to try it out yourself. 

Thanks!

dirtbikr, 2020-06-27:
Playing with this some more, I was able to use the above settings and removed the check for `&& !defined (RPI_DISPLAY_TYPE)` from the DMA initialization code (line 46 of TFT_eSPI_ESP32.h) and can upload the "SpriteRotatingCube" example sketch at 80MHz and it works (with some artifacting) at a hair over 300fps. Sometimes it will run up to ten seconds with no artifacts on the screen. Disabling DMA (commenting out the `#define USE_DMA_TO_TFT` in the "SpriteRotatingCube" example sketch on line 29 causes a momentary flash of the cube on boot, then a white screen and nothing after that (serial monitor reports around 220fps).

I was also able to upload my own sketch which uses LVGL at 80MHz, and it is distorted / artifacty but touch functions and you can kind of make out what is on the screen. I can also set the TFT_eSPI_ESP32.h file to absolute defaults and upload the LVGL sketch at 80MHz with the same result (distorted). Calling `tft.initDMA();` in my LVGL sketch causes everything to work as expected, but the menu tabs at have distortion around them and the scroll menus have a little bit as well. 

I'm thinking that enabling DMA is causing the screen to behave better, but there's something I am missing to fully support this screen at 80MHz.

Bodmer, 2020-06-28:
I have the MHS4.0 screen with ST7796 and it works fine with 80MHz and Setup 27. It is important to use the SPI ports native pins when operating at 80MHz, e.g. for the typical VSPI port:

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18

Unfortunately it will be difficult for me to help you once you have made any changes to the library code, so I recommend that you install a clean copy of the library as I think some of the changes you have made will make any timing issues worse.

Other suggestions:

- It using wires, keep them short (<100mm)
- Connect at least 2 GND wires from the ESP32 to the TFT
- Make sure the power supply is adequate as the power consumption increases with speed

dirtbikr, 2020-07-06:
Sorry for the delay in responding and thank you for your input!

I grabbed a fresh copy of the latest library, used setup 27 (just changed the driver to the ILI9486 driver) and attempted to run the demo sketches. Everything works fine at 40MHz, but at 80MHz you can see a shimmer of hope that it will work then it turns into a distorted screen with no hope of recovery. 

The wires I am using are about 60mm, I connected all the ground pins and am using a bench power supply set to 5v. I can see when running at 40MHz, the power consumption is ~190mA when drawing and ~170mA when idle. Bumping up to 80MHz increases power consumption during drawing by ~20mA. 

I have noticed plenty of strange behaviors before hooking up all the grounds, so that is certainly a good call. With the library in its current state, I was absolutely unable to get any meaningful output at 80MHz on this display. With the modifications I noted above from a previous version of the library, I was able to get a decent output (at least visible and interactive) at 80MHz. I'm sure it's possible (and I'm not well versed enough), but I'm going to run at 40MHz for now so I can get on with the project.

If you're interested in supporting this screen as a user setup I would be happy to pay to get a screen to you (Amazon gift card?). If you have other things to do, that's totally fine as well! Thank you so much for putting the library together.

closed 2020-07-06 by dirtbikr