Bodmer/TFT_eSPI #676: Is it a bug?

cgrrty, 2020-06-28:
Hi
I did a test，I don’t know if anyone else asked the same problem when using stm32f103c8t6
```
#include <TFT_eSPI.h> // Hardware-specific library

TFT_eSPI tft = TFT_eSPI(); // Invoke custom library

void setup(void) {
tft.init();
tft.setRotation(3);
tft.fillRect(0,0,1,1,TFT_WHITE);    
tft.fillScreen(TFT_PINK);
delay(100);
}
void loop() {
}
```
`tft.fillRect(0,0,1,1,TFT_WHITE);    `
When this statement is added, the `fillScreen `function is normal，
and this sentence is removed, the function is not normal.
Very confused, I don’t know why this happens.
[#652](https://github.com/Bodmer/TFT_eSPI/issues/652)

Bodmer, 2020-06-28:
Interesting. This may be due to a recent optimisation of the setWindow function.

Can you add:
```
#define MULTI_TFT_SUPPORT
```
to your setup file, recompile and upload to see if that cures the problem.

cgrrty, 2020-06-29:
i add` #define MULTI_TFT_SUPPORT`
Test code
```
#include <SPI.h>

#include <TFT_eSPI.h>       // Hardware-specific library

TFT_eSPI tft = TFT_eSPI();  // Invoke custom library

void setup(void) {
  tft.init();
  tft.setRotation(3);
  delay(100);
}

void loop() {
  tft.fillScreen(TFT_RED);
  delay(1000);   
  tft.fillScreen(TFT_BLUE);
  delay(1000);
  tft.fillScreen(TFT_GREEN);
  delay(1000);
  tft.fillRect(0,0,1,1,TFT_BLACK);
  delay(2000);  
}
```
When` tft.fillScreen(TFT_RED)` is executed for the first time,the fillScreen function is not normal.
After that, all fillScreen operations will become normal.

Bodmer, 2020-07-06:
I do not have that display so am unable to help further. I suggest you continue with using your work-around solution.

closed 2020-07-06 by Bodmer