Bodmer/TFT_eSPI #677: ST7789 display get flipped (rotate 180 degree) automatically, and then doesn't recovered till restart

mightyDipak, 2020-06-30:
Hello,
I am facing this strange issue, after display on about 4-5 hours(random hours, no pattern), display get rotated automatically. I tried to use function setRotation() in loop also but when this happend nothing works, display shows all data but rotated.
Please suggest best solution for it.
[image]

Bodmer, 2020-06-30:
Intermittent problems like this are tricky to identify the root cause. It looks like you are running a sketch written by yourself so some diagnostic experiments need to be run to see if this is:
1. A problem with the hardware reiability (power glitches, electrical noise spikes (aka EMC, possibly due to use of long wire or inadequate 0V/GND connections), impact of other faulty hardware connected to your system, SPI clock rate too high etc)
2. An environmental issue (high temperatue, vibration triggering a loose connection/solder joint fault)
3. A problem with the TFT_eSPI or indeed another support library WiFi/web, sensor, relay drive) you are running!
4. A fault in your own sketch causing the processor to crash (e.g. running out of memory, memory block fragmentation, pointer running off the end of an array for example).

So some experiments are needed to get a better understandind of the fault signature and narrow down the list of potential causes by running some test scenarios. To start with I would run, for many hours, an unmodified copy of a TFT_eSPI library example that does continuous screen updates to see if you get the same fault. Then consider which of the possible causes above might be the problem. Next make your sketch as simple as possible whilst still demonstratin the fault. In your sketch post regular messages to the serial monitor window or flash a LED so you can see if the processor has crashed. Post a copy of that sketch so that I can take a look.

Without a much clearer understanding of your setup and the conditions under which the fault shows itself I will not be able to help much more.

mightChamp, 2020-07-01:
Thanks for response, Is there any parameter I can read from ST7789, so that I know it has filliped or misbehaving, so I can reinitialize it again.

Bodmer, 2020-07-01:
Reduce the clock rate to 40MHz, ie 40000000
Bear in mind that the specs for these displays is around 25MHz.

mightChamp, 2020-07-02:
ok thanks I will Try this

mightChamp, 2020-07-03:
> ok thanks I will Try this

closed 2020-07-03 by mightChamp