Bodmer/TFT_eSPI #678: Question using padding

hasenradball, 2020-06-30:
Hi,

I would like to know how does it work when using a spezified padding?

Is the area first cleared using the width of padding, and then draw the string for example.

Bodmer, 2020-06-30:
The way it works is to draw the new text then clear any remaining area.

The text (font) rendering method depends on the font format used. The Smooth and Adafruit style fonts flicker because the text area is cleared first, this might lead to the assumption that the whole padding width is cleared first. This allows these fonts to be rendered with background without the sketch having to determine the bounding box and clear it within the sketch. The built in fonts (2,4,7,8) use a better approach where the whole character with background is rendered in one pass, this is possible because the font character bounding boxes never overlap by design (ie no italic characters).

hasenradball, 2020-06-30:
I have made a pic you see in the upper time.
To prevent flickering I do sererate showing second from the time and day.
[image]
But I could make it like this
Write the day and time from right to left.
And write the seconds from the colon from left to right.

Right?

Instead of calculate the day and time width...

Bodmer, 2020-06-30:
Yes, that would work.

closed 2020-06-30 by Bodmer

hasenradball, 2020-07-01:
Do I that by drawRightString() and drawString()?

hasenradball, 2020-07-01:
How do I manage to write:
1.) from left to write and add Padding at right (default)
2.) from right to left and add padding to the left?

Question is the padding automatically added at the left if TextDatum is TR?

Bodmer, 2020-07-01:
Normally strings are printed relative to the top left corner but this can be changed with the setTextDatum() function. The library has #defines for:
TL_DATUM = Top left
TC_DATUM = Top centre
TR_DATUM = Top right
ML_DATUM = Middle left
MC_DATUM = Middle centre
MR_DATUM = Middle right
BL_DATUM = Bottom left
BC_DATUM = Bottom centre
BR_DATUM = Bottom right

So use for example setTextDatum(TR_DATUM), set the padding width then draw the string. Padding will blank any area not covered by the string. There are examples that use padding and datum functions if you need more guidance.

hasenradball, 2020-07-02:
Hi Bodmer,

great works fine.
One question I had left.

Why do you think is it not possible to use 40 MHz on the ESP8266? Maybee it the tft ist not possible to use it?

And How could I use the SPI Pins of the flash?

Bodmer, 2020-07-02:
All my ILI9341 displays work at 40MHz reliably and start failing at about 80MHz (particularly when they warm up due to the high data rate and higher ambient temperatures). 40MHz is however "overclocking" the displays as typically the data sheets quote circa 25MHz.
There do appear to be ILI9341 "clone" chips on some displays sold as being ILI9341, I only have seen reported on the web but maybe they need a lower clock rate. When overclocking the timing of other signals such as DC and CS becomes important and I has to put in extra delays on these signals for the ILI9486 displays for example. The ILI9341 and ST7796 seems to be the displays that can be overclocked significantly.
The ESP32 data sheet also states timing parameters are not met unless the SPI native pins are used for each port (i.e these signals shoud not be routed via the IO matrix to other pin as that matrix introduces differential phase delays.
Other factors affecting clock rate are supply voltage and stability, length of wires, parasitic capacitance from other SPI devices on the bus and poor 0V/GND connections.
You may find if you tried another display from another vendor it will work reliably at 40MHz, some claim reliable operation at 80MHz but I suspect the tests used do not perform high rate screen updates for extended periods as this raises the temperature of the chip and it is then more likely to produce errors.

hasenradball, 2020-07-03:
Where do you buy your displays?

Bodmer, 2020-07-03:
For displays where I am likely to buy another I get mine from https://www.buydisplay.com/ as they have a consistent supply of each type and promise continued supply of the same module in future. They are slightly more expensive though complared to eBay or Aliexpress.
If you order from them you get exactly what is advertised.

The displays from Aliexpress are probably the cheapest but you may not be able to buy the same display at a future date because the vendors sell many products that I suspect are purchased in low cost oportunistic batches and this means continuity of supply is less likely. I have always had reliable service from Aliexpress vendors only in 2 cases have I received a cracked display and then I got a no quibble refund. You do need to read the advert carefully to make sure you are getting the display you want as some vendors do not seem to know what they are selling and the adverts are vague. Sometimes the sketch examples do not work with the display they are selling and the pictures are misleading. So I choose the vendor carefully and look for the quoted sales quabities and genuine feedback. I have spent severa hundred dollars on Aliexpress and have no significant complaints.