Bodmer/TFT_eSPI #679: Weather Station & 2 ILI9341 display

Arzhmoren, 2020-07-01:
Hi, i like your work and your weather station example.
Do you think it's possible to split the informations on 2 same ILI9341 display??
If it's possible do you have an exemple of how ??

Thanks for everything you do.

hasenradball, 2020-07-02:
The same question I had also in mind.
How is it possible to use 2 displays with your library?

Bodmer, 2020-07-02:
Two displays can be used by managing the chip selects in the sketch. See #663 

It would be possible to split the information but this is not something I have tried. I did buy a "gesture" sensor so a wave of the hand could be used to change the screen but I never got around to implementing this.

Assuming you are referring to the OpoenWeather library... All information is extracted and positioned in the TFT_eSPI_OpenWeather.ino sketch so examination of that file and experimentation will be required.

closed 2020-07-02 by Bodmer

Arzhmoren, 2020-07-02:
Thanks  for your response I going further on it

⁣Télécharger BlueMail pour Android ​