Bodmer/TFT_eSPI #680: ILI9341 screen scrolling

jarekdudzinski, 2020-07-02:
Not an issue, rather question - probably I'm doing something wrong... 

I'm trying to get vertical scrolling working. Found 'Terminal' example - but code from there doesn't work with rotated screen ( tft.setRotation(2); in my case). Don't have a clue how to fix it - checked functions from this example (which are using direct ILI9341 commands, not library functions), downloaded ILI9341 datasheet with commands... But can't figure this out :D

Any ideas how to fix scrolling with rotated screen? Also - how to change direction of scrolling (top-to-bottom, bottom-to-top)?

Bodmer, 2020-07-04:
I had the same question when I started to understand the feature. The scrolling is a hardware feature built into the display. It will only scroll up/down in the portrait orientation (rotations 0 and 2). Because it is a hardware function, in landscape orientation you can only get left/right hardware scrolling. See the [ILI9341 data sheet](https://www.crystalfontz.com/controllers/Ilitek/ILI9341/) on page 205+206 for more details.

jarekdudzinski, 2020-07-10:
@Bodmer Yes, I understand that. My question is, if this can be supported by the library functions (of course within hardware scrolling limitations).
In the "Terminal" example, which is using screen rotation 0, it's implemented by direct commands to the device, not library functions. I've tried to understand what those are doing (with help of ILI9341 datasheet) and how to reverse direction of the scrolling - but this exceeds my knowledge and/or abilities ;) (I'm an photographer / architect...)
Also searched google - no luck.

Bodmer, 2020-07-10:
OK, try the attached. Does it help?
[HW_Scroll.zip](https://github.com/Bodmer/TFT_eSPI/files/4904502/HW_Scroll.zip)

jarekdudzinski, 2020-07-10:
You are my hero! ;D 
This is exactly what I needed :) And it works correctly in any orientation, even in horizontal ones (with horiz scrolling), BOT_FIXED_AREA and TOP_FIXED_AREA works. Thanks :)

Bodmer, 2020-07-10:
Just bear in mind the coordinate frame moves with the scroll.

Hope your project is a success.

closed 2020-07-10 by Bodmer

VICLER, 2023-11-19:
> You are my hero! ;D This is exactly what I needed :) And it works correctly in any orientation, even in horizontal ones (with horiz scrolling), BOT_FIXED_AREA and TOP_FIXED_AREA works. Thanks :)

Hello!
How did you achieved vertical scrolling in landscape orientation (setRotation(1)) ? I want to have a vertical text scrolling in landscape view like it was done in Terminal example, but the scrolling direction does not change if I change setRotation to 1 in HW_Scroll sketch from @Bodmer.
I am really missing the function for scrolling text verticaly in landscape view. It works perfectly in LovianGFX, but I wanted to try smooth fonts, which lgx does not support.

VICLER, 2023-11-19:
> OK, try the attached. Does it help? [HW_Scroll.zip](https://github.com/Bodmer/TFT_eSPI/files/4904502/HW_Scroll.zip)

Hello! First, thank you very much for your library!
I find it very powerfull, but missing the function to scroll the text vertically in landscape orientation. Like your Terminal example, but with screen rotated to 90deg. I was using lovianGFX library and there is the feature for doing this, but I wanted to try smooth fonts, which lgfx does not support. I will be very happy if it is possible or if it will be added in feature releases.