Bodmer/TFT_eSPI #683: Multi Screen with Touch

SunboX, 2020-07-06:
Hi,

I have two identical displays (2.8" TFT) that are working fine using custom CS. Now I also want to use both Touch controller of these displays + I'm using a GPIO Expander (PCF8574) for controlling the CS signal.

My test code looks like this:
```c++
void loop()
{
  uint16_t tX = 0, tY = 0; // To store the touch coordinates
  boolean pressed = false;

  pcf8574.digitalWrite(firstTouchCS, LOW);
  pressed = tft.getTouch(&tX, &tY);
  pcf8574.digitalWrite(firstTouchCS, HIGH);
  if (pressed)
  {
    pcf8574.digitalWrite(firstScreenCS, LOW);
    tft.fillScreen(TFT_RED);
    pcf8574.digitalWrite(firstScreenCS, HIGH);
  }

  pcf8574.digitalWrite(secondTouchCS, LOW);
  pressed = tft.getTouch(&tX, &tY);
  pcf8574.digitalWrite(secondTouchCS, HIGH);
  if (pressed)
  {
    pcf8574.digitalWrite(secondScreenCS, LOW);
    tft.fillScreen(TFT_RED);
    pcf8574.digitalWrite(secondScreenCS, HIGH);
  }
}
```

Now I'm getting this error message at compile time:
`error: 'class TFT_eSPI' has no member named 'getTouch'`

I can not configure the CS for the Touch Controller, because of two displays + GPIO expander.

I already tied to set the config like this, but it did not solve the issue:
`#define TOUCH_CS -1`

Is there any way to get it to work using my setup?

SunboX, 2020-07-06:
ahh, maybe I can define `T_CS_L` and `T_CS_H` like this?

```c++
int currentTouchEnabled = 0;

void setCurrentTouch(int number) {
    currentTouchEnabled  = number;
}

void touchCSLow() {
    switch () {
        case 0:
            pcf8574.digitalWrite(firstTouchCS, LOW);
            break;
        case 1:
            pcf8574.digitalWrite(secondTouchCS, LOW);
            break;
    }
}

void touchCSHight() {
        switch () {
        case 0:
            pcf8574.digitalWrite(firstTouchCS, HIGH);
            break;
        case 1:
            pcf8574.digitalWrite(secondTouchCS, HIGH);
            break;
    }
}

#define T_CS_L touchCSLow()
#define T_CS_H touchCSHight()

void loop()
{
  uint16_t tX = 0, tY = 0; // To store the touch coordinates
  boolean pressed = false;

  setCurrentTouch(0);
  pressed = tft.getTouch(&tX, &tY);
  if (pressed)
  {
    pcf8574.digitalWrite(firstScreenCS, LOW);
    tft.fillScreen(TFT_RED);
    pcf8574.digitalWrite(firstScreenCS, HIGH);
  }

  setCurrentTouch(1);
  pressed = tft.getTouch(&tX, &tY);
  if (pressed)
  {
    pcf8574.digitalWrite(secondScreenCS, LOW);
    tft.fillScreen(TFT_RED);
    pcf8574.digitalWrite(secondScreenCS, HIGH);
  }
}
```

SunboX, 2020-07-06:
Sadly, it does not work, because I can not re-define `T_CS_L` and `T_CS_H`. Is there any other way to achive this?

Bodmer, 2020-07-07:
I see no reason why adding:
```
#define TOUCH_CS -1
```
to your setup file does not work for you.

Have you tried a simplified setup with one screen and the sketch directly controlling the touch cs line?

SunboX, 2020-07-07:
Hi, thanks for your quick response! Not yet, but I will try it as next. Will let you know if it works.

SunboX, 2020-07-07:
Btw., there's a small typo here:
https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Touch.cpp#L20

But it seems not to be the issue.

Bodmer, 2020-07-07:
It's not a typo. It is to make sure the TFT is not selected, it is there in case startWrite() had been called in a sketch without endWrite().
Which processor are you using?

SunboX, 2020-07-07:
Are you sure? Shouldn't it be named `T_CS_H` instead of `CS_H`?

Bodmer, 2020-07-07:
Yes, I am sure. The aim is to ensure the TFT chip select is high before selecting the touch controller.

Which processor are you using?

SunboX, 2020-07-08:
Ah, ok. Now I got it. Sorry for that! :smile:

I'm using a AZ Delivery ESP32 NodeMCU v4

Yesterday I tried a simple setup with my test code, only the ESP32 and one Display without doing CS by my own ... and it worked really great and fast! So the issue is not related to the ESP32 or the Display or your great Lib when used as it meant to be used.

What I want to try next to reach our goal:

1) doing CS by my own
2) doing CS using the GPIO Expander (PCF8574)
3) add the second Display

I keep you informed when I found out what I did wrong here. Thanks for your support!

Bodmer, 2020-07-08:
That is a good plan! Starting with a simplified system and adding complexity in small steps with tests at each design stage is a good development practice and is easier to debug.

closed 2020-07-16 by Bodmer