Bodmer/TFT_eSPI #684: <Adafruit_ADS1015.h> dont works

shalom198, 2020-07-07:
hello , 
i have aproblem with the TFT_eSPI , 
when i used it with the ads , the ads stop works , 
my ads configure on D1,D2 , 
when i used it i dont use the TFT , and when i use the TFT i dont use the ADS 
it change during the run of the program 
the tft configuration 
#define TFT_CS   PIN_D2  // Chip select control pin D8
#define TFT_DC   PIN_D1  // Data Command control pin
#define TFT_RST  PIN_D0  // Reset pin (could connect to NodeMCU RST, see next line)
#define TOUCH_CS PIN_D4     // Chip select pin (T_CS) of touch screen

thanks

Bodmer, 2020-07-07:
You can't share the TFT DC and CS pins with an I2C bus. Allocate different pins. This is not an issue I can solve in the library.

closed 2020-07-07 by Bodmer

shalom198, 2020-07-08:
My project is a bit complex, and I use all the pins therefore
I need to have the ability while running the software to disable / enable the TFT setting
While running the software switch between TFT and ADS
How to do this?
Is it possible to turn off the TFT interface to the controller without disabling the LCD setting, since there are commands facing the LCD while running the program [and I do not need to see data on the TFT at the same time]
thanks alot

Bodmer, 2020-07-08:
Disabling the TFT is not the problem. The problem is that the I2C signalling requirements are incompatible with the TFT drive signals. Note that when driving the TFT the CS and DC lines toggle at high speed and this is not compatible with the expected I2C signalling protocol.

My advice is to look at your design and try to simplify it so you can achieve some of the main project goals first, the only other solutions I can think of add even more complexity to your design which will means more problems to solve. 

Options:
1. Look at sharing some other signals other than the TFT ones
2. Use an I/O expander (e.g. PCF8574 [see here](https://github.com/Bodmer/TFT_eSPI/issues/683)). The I/O extender is itself and I2C device but perhaps you can move some of the low speed digital I/O to the expander outputs to free up some pins.
3. Choose another processor board with more free pins.
4. Add a second processor with a communication link to the main processor, maybe via I2C or a serial link.

This is the limit of the help I can give as this is not an issue with the TFT_eSPI library.

shalom198, 2020-07-08:
If I include  the library but do not run it , i mean that i will not
execute yet TFT_eSPI Lcd = TFT_eSPI();
Will there still be a conflict ?

‫בתאריך יום ד׳, 8 ביולי 2020 ב-22:22 מאת ‪Bodmer‬‏ <‪
notifications@github.com‬‏>:‬

shalom198, 2020-07-10:
hy 
can i use the pin A0 for CS or DC  ??

Bodmer, 2020-07-10:
To answer that I need to know which processor board you are using.

Bodmer, 2020-07-10:
An option if you have an ESP8266 NodeMCU board is to share the FLASH SPI bus with the TFT, this is called overlap mode. See Setup9_ST7735_Overlap.h as an example.

shalom198, 2020-07-10:
i'm using mini D1

‫בתאריך יום ו׳, 10 ביולי 2020 ב-10:24 מאת ‪Bodmer‬‏ <‪
notifications@github.com‬‏>:‬

Bodmer, 2020-07-10:
The documentation for that board will probable tell you that A0 is input only but you need to check that.

shalom198, 2020-07-14:
hello again
how i'm using the overlay mode ?
just configure in the user_setup.h  & wiring the pcb ??
or that i need to wrote somthingin the program ?

‫בתאריך יום ו׳, 10 ביולי 2020 ב-11:17 מאת ‪Bodmer‬‏ <‪
notifications@github.com‬‏>:‬

Bodmer, 2020-07-14:
Configure in the user_setup.h, you must use the pins listed.
Wire display to the correct pins.
Nothing changes in your sketch.

shalom198, 2020-07-14:
thanks

‫בתאריך יום ג׳, 14 ביולי 2020 ב-21:15 מאת ‪Bodmer‬‏ <‪
notifications@github.com‬‏>:‬

shalom198, 2020-07-14:
what about the tuch CS , is it stay alon ?

‫בתאריך יום ג׳, 14 ביולי 2020 ב-22:09 מאת שלום יוסף <‪shalom198@gmail.com
‬‏>:‬

Bodmer, 2020-07-14:
I do not understand your question.

The bad news: you cannot use a SPI touch screen if you are using overlap mode as there can only be one device on the SPI bus in overlap mode.

shalom198, 2020-07-15:
ooh , know i understand , why isnt working , i have with it ethernet card
too on the spi
so this not the solution for me
i need to go back , and separate them🙁

‫בתאריך יום ד׳, 15 ביולי 2020 ב-0:25 מאת ‪Bodmer‬‏ <‪
notifications@github.com‬‏>:‬

shalom198, 2020-07-15:
how can i resum/cancel the activity of the DC or CS of the TFT with in the
program
i need to control it within the program running, so the user can select
between show the screen or use other device ?

‫בתאריך יום ד׳, 15 ביולי 2020 ב-6:56 מאת שלום יוסף <‪shalom198@gmail.com
‬‏>:‬

Bodmer, 2020-07-15:
You will need to add external logic gates such as AND, OR, INVERT gates to route and disable the signals to different peripherals.