Bodmer/TFT_eSPI #685: TFT_eSPI Version 2.2.6 sprite/PSRAM bug(s)?

WA2FZW, 2020-07-07:
After installing version 2.2.6 of the TFT_eSPI library last night ( I 
was using version 2.2.0) the program I am working on started crashing
when trying to create a rather small sprite (10x7). Version 2.2.5 of
the library also causes the crash, however I didn't do a lot of testing
with that version.

The crash seems to be happening when the library tries to allocate
memory for the sprite and if I understand the library code correctly
it looks like it is trying to allocate that memory from the PSRAM.

I was originally compiling using the Arduino IDE Version 1.8.10
The ESP32 is the TTGO T8 V1.7 (WROVER) board
The display is an ILI9341 using the HSPI bus

Updating the IDE to Version 1.8.13 made no difference.

Changing the board type to "ESP32 Dev Module" in the IDE (no
PSRAM) fixes the crash problem, however rotated sprites aren't 
displaying in the proper colors. I can't tell is this is also
happening with the board type set to WROVER as it crashes well
before displaying anything other than the program's splash
screen.

This is happening both on my device and another member of the
developement team's device.

Below is the traceback, the register dump and the function in
which the crash occurs. Again this all works fine with version
2.2.0 of the library.

Decoding stack results:

0x400deb94: TFT_eSprite::callocSprite(short, short, unsigned char) at F:\Jim\Documents\Arduino\libraries\TFT_eSPI\Extensions/Sprite.cpp line 147

0x400decb5: TFT_eSprite::createSprite(short, short, unsigned char) at F:\Jim\Documents\Arduino\libraries\TFT_eSPI\Extensions/Sprite.cpp line 77

0x400d5f67: Marker::Init(TFT_eSPI*, unsigned char) at C:\Users\Jim\AppData\Local\Temp\arduino_build_482931\sketch\Marker.cpp line 36

0x400d3cc6: setup() at F:\Dropbox\Documents\Ham_radio\TinySA\JMP_TinySA_V2.9/JMP_TinySA_V2.9.ino line 574

0x400f148b: loopTask(void*) at C:\Users\Jim\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 14

0x4008cf25: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

This came up on the serial port:  

17:20:45.488 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
17:20:45.488 -> Core 1 register dump:
17:20:45.488 -> PC      : 0x400deb94  PS      : 0x00060430  A0      : 0x800decb8  A1      : 0x3ffb1f00  
17:20:45.535 -> A2      : 0x00000000  A3      : 0x00000001  A4      : 0x0000000a  A5      : 0x00000001  
17:20:45.535 -> A6      : 0x00000010  A7      : 0x00000008  A8      : 0x00000007  A9      : 0x3ffb1ee0  
17:20:45.535 -> A10     : 0x00000001  A11     : 0x3f409e82  A12     : 0x00000002  A13     : 0x0000ff00  
17:20:45.535 -> A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
17:20:45.535 -> EXCVADDR: 0x00000008  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Here's the function causing the crash:

void Marker::Init  ( TFT_eSPI* tft, uint8_t marker )
{
/*
 *  Create our "sprite" and set it up:
 */

    _tft = tft;
    _sprite.createSprite  ( MARKER_SPRITE_WIDTH, MARKER_SPRITE_HEIGHT );  // Set the size
    _sprite.setRotation   ( 0 );                                          // Portrait mode
    _sprite.setColorDepth ( 16 );                                         // Using 16 bit (RGB565) colors

/*
 *  Set some default conditions for the time being:
 */

    _index   = marker - 1;    // Convert marker number to an index
    _enabled = false;         // Turn it off for now
    _status  = 0;             // White, disabled & invisible
    _x = 0;                   // No real position yet
    _y = 0;
    _frequency = 0;           // Don't know the frequency
    _mode = MKR_PEAK;         // Assume peak frequency mode

/*
 *  Set the pivot point for the marker - this is the point used when pushing
 */

    _sprite.setPivot ( X_MARKER_OFFSET, Y_MARKER_OFFSET );
}

Bodmer, 2020-07-08:
I usually use the "ESP32 Dev Module" board option and it allows control of whether PSRAM is enabled.

The "ESP32 Wrover Module" board option enables PSRAM by default. If your board does not have functional PSRAM then it would crash.

I have just released 2.2.14 please try that.

Issue #536 may be of interest.

To disable/enable use of PSRAM you can now use:
```
sprite.setAttribute(PSRAM_ENABLE, false); // true to enable
```
Where "sprite" is the Sprite class instance.

WA2FZW, 2020-07-08:
Hi Bodmer!
 
Thanks for the quick response, but alas, version 2.2.14 doesn’t fix either of the problems I reported. Again, version 2.2.0 of the library works.
 
With the board type set to “WROVER”, the program still crashes when trying to create the sprite. Setting the board type to “ESP32 Dev Module” fixes the crash problem as expected, but the colors of rotated sprites are still wrong (non-rotated ones seem ok for the few colors I’m using). Note, I didn’t test this with the version 2.2.0 library, only 2.2.5 and up.
 
      Specified  Displayed
 
      WHITE      WHITE
      RED        BLUE
      BLUE       GREEN
      GREEN      RED
      ORANGE     PURPLISH
      YELLOW     MAGENTA (or close)
 
Looks like the RGB’s might be mixed up somewhere.
 
The ability to disable the PSRAM use would be ok, but if I can’t create the sprite in the first place, how would I set the PSRAM disable attribute? That would have to be done prior to trying to create it. 
 
Perhaps you need a symbol in the user setup file to disable the PSRAM use altogether, or figure out why it’s crashing when using the PSRAM. By the way, it doesn’t appear to be a problem with the PSRAM on my board as the problem was actually discovered by someone else.
 
I have another project that requires the WROVER’s PSRAM capability (it builds the entire screen image and sends the entire image at once). That program isn’t using sprites (yet). It also uses the ILI9341 display and seems to be working ok.
 
Thanks for the link to the other article; it explains some things we knew about.
 
John
WA2FZW

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Bodmer, 2020-07-08:
I have found a TTGO T8 V1.7 board in my box of bits! I have compiled and run the **"Sprite_draw"** example using both the "ESP32 Dev Module" and "ESP32 Wrover Module" board options without any problem. The FLASH program size for the ESP32 Wrover module option is ~50kbytes bigger as expected because PSRAM is enabled by default. Try the same sketch.

Also tried the "ESP32 Dev Module" board option with the menu PSRAM option turned on, code grows ~50kbytes and it still compiles and runs.

My setup:
Arduino IDE 1.8.11 (unlikely to cause the issue)
ESP32 board manager package 1.0.4
TFT_eSPI version 2.2.14

So my guess is that your setup is different or there is a bug in your sketch. To eliminate the latter can you provide a copy of a simplified sketch that exhibits the problem on your setup?

Does your sketch use SPIFFS: I ask because the later board packages are incompatible with the old SPIFFS images that may be present on your board. Solution is to upload all files SPIFFS files again. Bear in mind that error reports are not always truthful about where the problem is, especially if there is an out of memory/heap or fragmented memory problem when executing a sketch.

Other points:
setColorDepth should be used before creating the sprite, otherwise the sprite will be destroyed and recreated again with the new specified depth, sprites by default are 16 bits when created
setRotation ( 0 ) has no effect on 16 bit sprites, it is only relevant to 1bpp sprites

WA2FZW, 2020-07-08:
My setup:
Arduino IDE 1.8.13 (was using 1.8.10 when the problem first occurred)
ESP32 board manager package 1.0.4
TFT_eSPI version 2.2.14

The real program does use SPIFFS for a web interface; I reloaded them with the version 2.2.14 library in use.

The attached program works using either "ESP32 Dev Module" or "ESP32 Wrover Module" when using version 2.2.0 of the library.

It works when using "ESP32 Dev Module" with version 2.2.14 of the library, but crashes when the board type is set to "ESP32 Wrover Module"  with version 2.2.14 of the library. I looked through the "Sprite.cpp" files for the 2.2.0 and 2.2.14 libraries and not that I fully understand either, but did notice that there are a lot of differences.

There are a couple of minor differences between this program and the real one to eliminate references to things that only exist in the real program. In the "Marker.cpp" file I commented out the "setRotation" and "setColorDepth" calls as you indicated these are defaults.

[Marker_Test.zip](https://github.com/Bodmer/TFT_eSPI/files/4893573/Marker_Test.zip)

toetoast, 2020-07-08:
Hi there, 

Looking at the Guru error I see the words "LoadProhibited". In my experience I get this error when trying to access memory locations that are *outside* what is acceptable or what has been previously defined by a MALLOC etc. If a memory *write* is attempted in one of these said locations then I definitely get this Guru error :)

I've just created the following Sprite code enhancement that should put your on the right path - https://github.com/Bodmer/TFT_eSPI/issues/688

Some other points to consider:
1) Are you using large sprites like 240 x 320? The reason I say this is that on ESP32 devices the amount of on-board RAM that can be allocated in one (contiguous block) is around the 150KB mark. Such large sprites (16bpp say) try to use way more than this, but will fail when trying to use on-board RAM.
2) Are you using DMA to transfer the Sprite data to the TFT in some way? There is a limit of around 76KB(?) that can be sent at once.

Cheers
Tony

Bodmer, 2020-07-09:
Thanks for the example.

The problem is at line 91+92 of Marker.h
```
	TFT_eSPI*	_tft;
	TFT_eSprite _sprite = TFT_eSprite ( _tft );	// Sprite for the marker
```
An instance of the Sprite class is being defined with a pointer to a tft instance, but that instance has not yet been defined so it will be a bad pointer. When a PSRAM sprite is created the callocSprite function in the Sprite class uses that duff tft instance pointer, but when it is in RAM it does not. Earlier versions of the TFT_eSPI library did not use the tft instance pointer so if it was invalid that would not have shown up at that time. This is a nice example of a latent bug.

You need to pass a valid Sprite instance pointer (i.e. img) to the Marker class. Just as you passed the tft instance pointer. e.g.:
```
void	 Init ( TFT_eSPI* tft, TFT_eSprite* sprite, uint8_t index );			// Psuudo constructor #1
```

Bodmer, 2020-07-09:
I have made a change and when the tft pointer is valid the example runs fine. You could also only create the sprite instance after a valid tft pointer has been passed to the Marker class.

WA2FZW, 2020-07-09:
So far so good!!!

My real program no longer crashes! Passing the address of the sprites to the Marker objects fixed that. It now works with either ESP32 board type and also works with or without Tony's mod (which should probably be included permanently) in the library and with the mod, it works with the PSRAM enabled or disabled.

Now the only problem is the color confusion mentioned in the original post. In the picture below, the markers should be (from left to right) red, blue and green, but are blue, green and red. The marker sprites are actually rotated and pushed into another sprite used to paint the graph. That sprite is only 2 pixels wide and the height of the graph. The colors put directly into that sprite are correct.

[image]

WA2FZW, 2020-07-09:
I added a "SwapBytes" function to swap the color bytes when setting the pixels in the marker sprite and that fixed the color problem, so it looks like somewhere in the library's pushRotated logic, there is a swap that shouldn't be there or vice-versa.

Swapping the bytes made the colors correct.

Bodmer, 2020-07-09:
I think the colour problem is due to changes resulting from issue #606. Essentially the the Sprites pixel colors are stored as big endian (so DMA or block transfer to the screen does not need to swap bytes in real time) whereas the processor is actually little endian. Unfortunately there is a long standing inconsistency in the way the endianess swaps are handled by the library in various scenarios, so if I fix this it will break some user sketches and one or more examples... I will think on this further. As you have a solution I will put this update on the back burner for now as it will need to push the library to 3.x.x and there are other possible changes in the pipeline. I see Tony has raised a separate issue so that will be handled in due course.

WA2FZW, 2020-07-09:
I'm OK with keeping my fix in the program, but remember, the colors (rotated sprites) were correct in version 2..2.0 and broken as of version 2.2.6 (I didn't check 2.2.5), so some people's programs may already be broken (although I seem to be the only one who pointed this out).

As for Tony's suggestion, I think it is a good idea. You pointed out that disabling the PSRAM can be done on a sprite by sprite basis, but I think it would also be able to do it globally.

As far as I'm concerned, this issue can be closed.

Thanks for the help (as usual)!!!

closed 2020-07-09 by Bodmer