Bodmer/TFT_eSPI #686: Possible to disable Datum smarts?

red-fox-star, 2020-07-08:
There's a lot of time savings built into the datum logic, and it's really helpful in a lot of situations. However, when I want really precise control of strings it can break because of the datum logic automatically adjusting position in order to make a string fit. I'd like to able to place a character a smidge off screen, for example.

Thanks!

Bodmer, 2020-07-08:
I think you should be able to place text off screen in the Y direction but maybe not in the x direction. Changes should be simple but it is a while since I looked at that code. Can you provide a simple complete simple sketch to be used as a test?

red-fox-star, 2020-07-13:
After some debugging, I don't know if this will be a valuable addition or not. There's a bit more to the story than I initially thought. Here's where I got led astray.

I am combining the pong example with the digital clock example, and moving the startWrite/endWrite calls up to a top level to try to increase performance. However the `setAddrWindow` function also calls startWrite. So I was getting a start/start/end/end nesting, which seems to have caused all sorts of problems with text alignment. I subbed the `setWindow` function with altered coordinates and text alignment is working _much better_ than it was. I don't understand the semantics of setAddrWindow because I expected it to need to be un-set, but perhaps this pong is a strange subset of the problem setAddrWindow is solving -- it only uses pushColor in one place and the rest of the code ignores the address window?

I modified the display logic of the digital clock example in this way -- I'm calculating the font metrics manually so I can center the text regardless of which font is used. It's pretty good, but not perfect.

```c++
#define LILYGO_TWATCH_2020_V1

#include <TTGO.h>
#include <WiFi.h>

int16_t hours_font = 8;
int16_t minutes_font = 8;
int16_t seconds_font = 4;

int16_t old_hour = 24;
int16_t old_minute = 61;
int16_t old_second = 61;

int16_t x_middle;
int16_t y_middle;

int16_t hours_half_height;

int16_t hours_x;
int16_t hours_y;

int16_t minutes_x;
int16_t minutes_y;

int16_t seconds_x;
int16_t seconds_y;

TTGOClass * watch;
TFT_eSPI * screen;

void setup(void) {
  watch = TTGOClass::getWatch();
  watch->begin();
  watch->rtc->check();
  watch->bl->adjust(150);

  screen = watch->eTFT;
  screen->fillScreen(TFT_BLACK);
  screen->setTextColor(TFT_WHITE, TFT_BLACK);

  x_middle = screen->width() / 2;
  y_middle = screen->height() / 2;
  hours_half_height = screen->fontHeight(hours_font) / 2;
  hours_x = x_middle - screen->textWidth("00", hours_font);
  hours_y = y_middle - hours_half_height;

  minutes_x = x_middle;
  minutes_y = hours_y;

  seconds_x = x_middle - screen->textWidth("0", seconds_font);
  seconds_y = y_middle + hours_half_height + 10;
}

void loop() {
  int16_t position_x;
  RTC_Date time = watch->rtc->getDateTime();

  watch->eTFT->startWrite();
  // hour
  if (old_hour != time.hour) {
    old_hour = time.hour;
    position_x = hours_x;
    screen->setTextPadding(screen->textWidth("0", hours_font));
    if (time.hour < 10) position_x += screen->drawChar('0', position_x, hours_y, hours_font);
    position_x += screen->drawNumber(time.hour, position_x, hours_y, hours_font);
  }

  // minute
  if (old_minute != time.minute) {
    old_minute = time.minute;
    position_x = minutes_x;
    screen->setTextPadding(screen->textWidth("0", minutes_font));
    if (time.minute < 10) position_x += screen->drawChar('0', position_x, minutes_y, minutes_font);
    position_x += screen->drawNumber(time.minute, position_x, minutes_y, minutes_font);
  }

  // seconds
  if (old_second != time.second) {
    old_second = time.second;
    position_x = seconds_x;
    screen->setTextPadding(screen->textWidth("0", seconds_font));
    if (time.second < 10) position_x += screen->drawChar('0', position_x, seconds_y, seconds_font);
    position_x += screen->drawNumber(time.second, position_x, seconds_y, seconds_font);
  }
  watch->eTFT->endWrite();

  vTaskDelay(250);
}
```

Bodmer, 2020-07-17:
I do not use or provide support for the TTGO library. Raise an issue with TTGO.

closed 2020-07-17 by Bodmer