Bodmer/TFT_eSPI #687: 'TFT_CS' was not declared in this scope

kitecraft, 2020-07-08:
Hi.  I just pulled the current version but I get an error now.

<--begin snippet-->
In file included from C:\Workspace\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:20:0:

C:\Workspace\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c: In member function 'bool TFT_eSPI::initDMA()':

C:\Workspace\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:685:21: error: 'TFT_CS' was not declared in this scope

     .spics_io_num = TFT_CS,
<--end snippet-->

I'm using multiple displays and am managing the cs pins myself.  So, no TFT_CS pin is defined.  
While using '#define TFT_CS   -1' does seem to work so far, I don't know what this might break.  
Also, I have some demo's and videos on using multiple displays where I explicitly state to not define the CS pin and I'd hate to be made a liar. 

Thank you.

Bodmer, 2020-07-09:
The particular case where this is an issue is using and ESP32 & using multiple displays & not defining TFT_CS.

I don't attempt to support the use of multiple displays and, as the library evolves, it is inevitable that tutorials for open software based projects will become out dated. I recently had this problem for a tutorial for a Raspberry Pi project due to Raspian OS evolution, so you have my sympathy. The simple solution is to quote the exact library version number used for the tutorial, then readers can't complain.

I will see if there is a simple update that keeps your tutorial valid.

Bodmer, 2020-07-09:
Library has been updated so you can relax!

Special care will be needed if you wish to use DMA with sketch managed TFT chip selects,r, use dmaBusy() before deselecting the TFT.

closed 2020-07-09 by Bodmer