Bodmer/TFT_eSPI #688: Enhancement: How to choose where your new Sprites will reside - onboard RAM or PSRAM

toetoast, 2020-07-08:
When creating new Sprites, I've found a way to guarantee what type of memory (on-board RAM or PSRAM) the new Sprite will exist on, when working with ESP32 devices like the TinyPICO etc that have 4MB PSRAM chips attached.

Here is what is needed:

1) In the **TFT_eSprite::createSprite** method (as shown below) the following line needs to be added:

` _psram_enable = _tft->getAttribute(PSRAM_ENABLE);    // Update the new Sprite Class parameter with the latest value from the TFT Class`

```
/***************************************************************************************
** Function name:           createSprite
** Description:             Create a sprite (bitmap) of defined width and height
*************************************************************************************x*/
// cast returned value to (uint8_t*) for 8 bit or (uint16_t*) for 16 bit colours
void* TFT_eSprite::createSprite(int16_t w, int16_t h, uint8_t frames)
{

  if ( _created ) return _img8_1;

  if ( w < 1 || h < 1 ) return NULL;

  _psram_enable = _tft->getAttribute(PSRAM_ENABLE);    // Update the new Sprite Class parameter with the latest value from the TFT Class																																			 
  _iwidth  = _dwidth  = _bitwidth = w;
  _iheight = _dheight = h;
```

2) Then in your code (just before you call the method to create your new Sprite) choose to add the following lines as needed:

`tft.setAttribute(PSRAM_ENABLE, false);  // Forces all future Sprites to be created in the on-board RAM area`
`tft.setAttribute(PSRAM_ENABLE, true);  // Forces all future Sprites to be created in external PSRAM`

If you don't add the single line of code to the createSprite method then the default behaviour is to *force* the Sprite to be created in PSRAM, if that chip exists. 

This fix has been tested successfully on Library V2.2.6 and V2.2.14

WA2FZW, 2020-07-09:
Thanks Tony!

I made the mod to the library (2.2.14) and added the "tft.setAttribute(PSRAM_ENABLE, false);" statement immediately after creating the TFT object. It sure looks like it should work, but it didn't solve the crash problem either in my real program or in the test program I posted earlier. As a matter of fact, the test program now crashes when I set the board type to "ESP32 Dev Module" which worked before.

Adding a "tft.getAttribute(PSRAM_ENABLE)" call immediately after the "setAttribute" call followed by "while ( true ) {}" to keep it from getting to the crash point shows the PSRAM is indeed disabled.

The real program isn't using any extremely large sprites; the largest is roughly 32x300; The one that I'm attempting to create when the crash occurs is 7x10.

Just got a reply from Bodmer which may solve the problem. Will take a few minutes to test.

WA2FZW, 2020-07-09:
So far so good!!!

The real program now runs for the "ESP32 Dev Module" and the "ESP32 Wrover Module". Passing the pointers to the sprites into the Marker objects fixed that. It also works with Tony's modification to the library (which should be made a standard capability me thinks) included or not and with the PSRAM either enabled or disabled.

Now the only problem is the color confusion mentioned in the original post. The markers on the attached picture should be (from left to right) red, blue and green, but are blue, green and red. Again as noted in the original post, the marker sprites are actually pushed into another sprite (see the "Paint" function in the example I sent.

[image]

Bodmer, 2020-07-09:
I understand your reasoning but that is not the intended way of managing the use of PSRAM.

If you wish to control the use of PSRAM separately is different sprites then use the inherited function:

spr1.setAttribute(PSRAM_ENABLE, false);
spr2.setAttribute(PSRAM_ENABLE, true);

Then you have independent control of using PSRAM in the parent TFT class with:
tft.setAttribute(PSRAM_ENABLE, true);

It is some time since I tested this, so let me know if this does not work for you.

toetoast, 2020-07-09:
Hi, thanks for the suggestion.

I'm thinking that perhaps using the inherited function may be too late in the life of the Sprite. In other words, the RAM allocation for the Sprite has *already* been defined (RAM or PSRAM) as it was created, so setting the Attribute will have no affect on that Sprite. That is why I thought about setting the Attribute at the TFT level *first* before creating any new Sprites.

The other approach I was thinking about was having an optional parameter on the Sprite Creation method, so that the user can force the RAM allocation to be what they want when asking for the Sprite to be created, which may be a little neater. What do you think?

The reasoning for me choosing the RAM type for Sprites is that:
1) larger Sprites (240 x 320 say) won't fit in ESP32 onboard RAM, but will easilly fit in PSRAM
2) As you probably know, PSRAM seems to be a certain percentage slower, so moving data around in larger Sprites will show this speed difference more.

Cheers
Tony

Bodmer, 2020-07-09:
To create a sprite first set any parameters or attributes (e.g colour depth, PSRAM switch) then create it. The library then knows what to do.

If you wish to move the sprite from PSRAM to RAM or vice versa then use deleteSprite, then set up the new attributes (e.g colour depth, PSRAM switch) and use createSprite again.

Be aware that creating + deleting + recreating sprites (in particular large ones) is likely to cause memory fragmentation unless managed very carefully with a knowledge of the evolving memory allocation, without proper memory management consideration the sketch will then eventually crash (in hours or many days time...). The simplest management method is to allocate then in the setup() function in the sketch and leave them alone!

At the moment I don't see a need to add any new functionality, unless I have misunderstood what you wish to achieve.

toetoast, 2020-07-09:
Yes, I agree totally with what you say here and I'm following the best practice methods you mentioned :)

I'm not doing anything that special really, just wanting to create the Sprite in the right RAM chip for my needs on a Sketch-by-Sketch basis.

What happened was that I was scratching my head for hours trying to understand why a Sprite would not be allocated to the type of RAM that I had specified at the start of the Sketch (by doing so in the TFT Class beforehand). After using a lot of Trace code I found that the issue would **only** be fixed when inserting this line of code in the _spriteCreate_ method as mentioned in the original comment:

`_psram_enable = _tft->getAttribute(PSRAM_ENABLE);        // Update the new Sprite Class parameter with the latest value from the TFT Class`

The crux of the issue is shown by the comment on the above line. I found that the Sprite get's defined with an **old value** of __psram_enable_ and therefore never creates the Sprite in the RAM that was intended (it defaults to PSRAM).

I'm thinking that this may be the first time that this bug has been found. The reason I say this is that I'm using an ESP32 TinyPICO device that has the ability to enable/disable it's under slung PSRAM chip from the Arduino IDE at compile time. This is how I was able to take my time to debug what was going on.

Here is a small section of the Sketch that is about to load a .BMP file (ImageFile) from an SD card to a large Sprite. You will see that before the Sprite is created I'm setting the desired destination RAM for it in it's Parent TFT Class. Without the addition of the single line of code mentioned above, the outcome is that the Sprite is created in PSRAM (if present).

Hope this helps.
Cheers
Tony

```
    ImageInfo = GetImageInfo(ImageFile);
    
    if ((ImageInfo.width == 0) || (ImageInfo.height == 0))
    {
        DEBUGLN("Unable to access the image file - system halted!");
        while (1) yield(); // System halted!
    }
    if ((ImageInfo.width > 240) || (ImageInfo.height > 320))
    {
        DEBUGLN("The image size is too large to create a Sprite for - system halted!");
        while (1) yield(); // System halted!
    }

    tft.setAttribute(PSRAM_ENABLE, false);                          // Ensure that the Sprite goes to the on-board RAM.
    if(!spr11.createSprite(ImageInfo.width, ImageInfo.height))      // This Sprite covers the whole TFT screen and will be used for backround images.
    {
        DEBUGLN("Not enough RAM to create the Sprite - system halted!"); 
        while (1) yield(); // System halted!
    }

    spr11.setTextDatum(TL_DATUM);                               // Top Left datum
    if(!ImageToSprite(ImageFile,&spr11))      // This Sprite covers the whole TFT screen and will be used for backround images.
    {
        DEBUGLN("Unable to load the image file - system halted!"); 
        while (1) yield(); // System halted!
    }
```

Bodmer, 2020-07-10:
I see no evidence of a bug in the TFT_eSPI library, just a misunderstanding of how to select the memory type for a sprite. 

The problem (aka bug) is this line in your sketch example above:
```
    tft.setAttribute(PSRAM_ENABLE, false);                          // Ensure that the Sprite goes to the on-board RAM.
```
This line will not (as the comment implies) ensure the sprite goes to onboard RAM if PSRAM is present. It should read:
```
    spr11.setAttribute(PSRAM_ENABLE, false);                          // Ensure that the Sprite goes to the on-board RAM.
```
and all will be well.

To clarify, the use of the tft.setAttribute(PSRAM_ENABLE, false); is to manage where tft class data such as smooth font metrics is stored.

Linking the tft class flag to the sprite class is not desirable. Incidentally, if it were desirable then the simple solution is to use this format in the Sprite class:
```
  _tft->_psram_enable
```
instead of:
```
  this->_psram_enable
```

closed 2020-07-10 by Bodmer

toetoast, 2020-07-10:
I got it now thanks. The thing that was sending me along the wrong path was that I could not find this method in the **TFT_eSprite** Class file **Sprite.cpp**:

`setAttribute`

 so I assumed it was not present :)

All is well now that I have adjusted my code as per your last comment.

Would I be right in assuming that the best approach is to set **all** the Sprite parameters after it's Class has been defined and then create the Sprite as in this example:

```
TFT_eSprite spr = TFT_eSprite(&tft);             // Create new Sprite Class Instance

// Set the Sprite parameters now that it's Class has been initialised.
spr.setColorDepth(16);
spr.setTextDatum(TL_DATUM);
spr.setAttribute(PSRAM_ENABLE, false);           // Ensure that the Sprite goes to the on-board RAM.

// Now to actually create the Sprite and allocate the RAM for it.
spr.createSprite(100, 100)
```
Many thanks for your guidance.
Tony

Bodmer, 2020-07-10:
Yes, create the sprite class instance after the tft class instance so the tft class instance pointer is valid. Creating the sprite class does very little except set up some default variables (16 bit color etc).
Then set up all the parameters for the sprite (color depth etc).
Then create the sprite.

The sprite class inherits functions from the parent tft class (drawCircle for example), over-rides the virtual ones (drawPixel for example so then the inherited drawCircle works in a sprite) and has sprite specific functions.

Given the absence of any user manual for the library (so far...) you have done well!