Bodmer/TFT_eSPI #689: Touch Interface not working with External SD Drive

Pizza-Monster, 2020-07-09:
Hi,

I have been stuck on problem since I introduced an external micro SD adaptor/drive to my design.  My design has the following components:
1. KeeYees ESP32 ESP-32S Development Board
2. HiLetgo ILI9341 2.8" SPI TFT LCD Display Touch Panel 
3. HiLetgo Micro SD TF Card Adaptor Reader Module 6Pin SPI Interface Driver Module with chip Level Conversion

I purchased all components on Amazon.

My Devices are connected to the ESP32 as follows:

TFT Display:
TFT_CS	       GPIO_15
TFT_RST	       GPIO_04
TFT_DC	        GPIO_02
TFT_MOSI 	GPIO_23
TFT_SCK	GPIO_18
TFT_MISO 	Not Connected

Touch:
TOUCH_SCK	        GPIO_18
TOUCH_CS		GPIO_21
TOUCH_MOSI	GPIO_23
TOUCH_MISO	GPIO_19

External Micro SD Adaptor:
SD_MISO	GPIO_19
SD_MOSI	GPIO_23
SD_SCK	        GPIO_18
SD_CS	        GPIO_05

I am using the external micro SD adaptor because I cannot physically fit the LI9341 2.8" SPI TFT LCD Display into the enclosure using the internal SD drive with an SD card.

Below is list of scenarios of what works and what doesn’t work:
1. If I use the internal SD drive on the  LI9341 2.8" SPI TFT LCD Display, everything works at 3.3V and 5V, that is the display, SD drive and touch all work. But like I said, due to physical limitations. I would like to use the external micro SD adaptor. But using this configuration, I at least know that my wiring and software all seem to work together. I used the ‘On_off_Button’ example to test the touch interface, and it works fine.

2. If I use 5V, the external SD drive and the display work fine but the touch doesn’t work properly. When I use the ‘On_off_Button’ example, the program races through the calibration process without me ever touching the display, and then jumps to the On/Off button display, without working. 

3. If I connect 5V to the external SD adaptor and 3.3 volts to the Display/Touch, I get the same results as I did in scenario 2, i.e. The display and the external SD drive work, but not the touch. The touch interface is just not happy when the external SD drive is in the circuit.

4. When using 3.3V in the entire circuit, the SD drive does not seem to work, which makes sense since I believe it is designed to be connected to a 5V power source.

Any help will be greatly appreciated.

Thanks

Bodmer, 2020-07-09:
Thanks for the complete description of your problem. I rarely get such a good one, so you deserve a more comprehensive answer ;-)

Re. your scenarios, so

1. That is good news and as you say shows the configuration and wiring is good.

2. This suggests a problem with the connections to the touch controller. You have probably checked the wiring a dozen times but I would check again. Run the diagnostic Test_Touch_Controller sketch, output is to the serial monitor only. In the sketch at the start of setup use digital writ to set the SD card chip select high so the SD card does not produce a SPI bus conflict.  Check screen presses change the output values in the serial monitor window. Report findings.

3. To be expected, given scenario 2 behavior.

4. This suggests the 3.3V line (is it from the ESP32 board regulator?) cannot deliver the current needed or the power wires are too long. On some ESP32 boards the 3.3V regulator is already running near full capacity just powering the ESP32. I would ignore this for not and focus on getting scenario 2 working.

Other possible reasons for scenario 2 failing:
a) Signal wires too long, the ESP32 outputs high speed edges to the signals that can cause signal "ringing" due to wire inductance.
b) Poor GND/0V connections between boards, add extra wires to reduce inductance.
c) Typically the display SD interface has series resistors in the signal lines. Probably to either reduce ringing/reflections or probably protect the SD card I/O signals from folk who try to drive it with 5V logic levels! Since the SD card interface works only the ringing suppression is a possible candidate for the behavior.

Pizza-Monster, 2020-07-10:
Hello, Thanks for the quick response and suggestions! I tried making wiring as short as possible (still using jumper wires, but all devices are now plugged directly into the bread board. Below are are my results of running the Test_Touch_Controller sketch, with the following line added to the top of the setup() function:
digitalWrite(5, HIGH);

Output at 5V, with SD Adaptor in Circuit. Output updates constantly, with no change in value - even when I touch the display:

18:59:09.085 -> x: 8191     y: 8191     z: 4095 
18:59:09.337 -> x: 8191     y: 8191     z: 4095 
18:59:09.580 -> x: 8191     y: 8191     z: 4095 
18:59:09.823 -> x: 8191     y: 8191     z: 4095 
18:59:10.070 -> x: 8191     y: 8191     z: 4095 
18:59:10.313 -> x: 8191     y: 8191     z: 4095 
18:59:10.594 -> x: 8191     y: 8191     z: 4095 

Output at 5V, with SD Adaptor removed from circuit. Output does respond to swiping finger across display:

19:12:28.104 -> x: 322     y: 3375     z: 8 
19:12:28.355 -> x: 0     y: 3806     z: 19 
19:12:28.599 -> x: 338     y: 3411     z: 8 
19:12:28.839 -> x: 84     y: 3535     z: 44 
19:12:29.091 -> x: 336     y: 3290     z: 14 
19:12:29.334 -> x: 1992     y: 3694     z: 1203 // Finger swipe starts here
19:12:29.591 -> x: 1991     y: 2703     z: 1959 
19:12:29.840 -> x: 1972     y: 3350     z: 1509 
19:12:30.094 -> x: 1883     y: 2879     z: 1898 
19:12:30.348 -> x: 1983     y: 3003     z: 1894 
19:12:30.600 -> x: 0     y: 3647     z: 39 
19:12:30.847 -> x: 11     y: 3904     z: 21 
19:12:31.089 -> x: 0     y: 3743     z: 27 

Again, I really appreciate your help - Thanks.

Bodmer, 2020-07-10:
OK, so this indicates the SD card adapter is somehow affecting the SPI bus, preventing read and/or write communication  with the Touch controller.
Try reducing the touch controller SPI clock rate to 1MHz in the setup file.
What happens if the SD adapter card is plugged in but no SD card is inserted into the adapter?
Run the Read_User_ Setup diagnostic sketch and post the output from the serial monitor window.
Post a close-up picture of your breadboard setep.

Bodmer, 2020-07-10:
Check also you have a good 0V/GND connection to the SD adapter and replace these jumper wires. It looks like the SD card does not have a good GND connection and is using the CS line as a power return.

Pizza-Monster, 2020-07-10:
Test case 1: No SD card in SD adaptor, 5V, using Test_Touch_Controller sketch in unmodified form:

13:14:17.342 -> x: 0     y: 0     z: 4095 
13:14:17.584 -> x: 0     y: 0     z: 4095 
13:14:17.861 -> x: 0     y: 0     z: 4095 
13:14:18.115 -> x: 0     y: 0     z: 4095 
13:14:18.331 -> x: 0     y: 0     z: 4095 
13:14:18.609 -> x: 0     y: 0     z: 4095 

Still doesn’t work, but result is different than when I have the SD card inserted (as shown below):

13:18:12.965 -> x: 8191     y: 8191     z: 4095 
13:18:13.218 -> x: 8191     y: 8191     z: 4095 
13:18:13.461 -> x: 8191     y: 8191     z: 4095 
13:18:13.734 -> x: 8191     y: 8191     z: 4095

——————————————————————————————————————————————————————————————————

Test case 2:  SD card in SD adaptor, 5V, using Test_Touch_Controller sketch in unmodified form, setting Touch Controller SPI clock to 1 MHz:

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
//#define SPI_TOUCH_FREQUENCY  2500000
#define SPI_TOUCH_FREQUENCY  1000000

13:31:00.619 -> x: 8191     y: 8191     z: 4095 
13:31:00.868 -> x: 8191     y: 8191     z: 4095 
13:31:01.116 -> x: 8191     y: 8191     z: 4095 
13:31:01.364 -> x: 8191     y: 8191     z: 4095 
13:31:01.610 -> x: 8191     y: 8191     z: 4095 

——————————————————————————————————————————————————————————————————

Test Case 3: Disconnect MISO from SD Adaptor (all other pins connected)

I tried adding another ground connection to the SD Adaptor and this did not help. You mentioned that you thought that the SD Adaptor might be using the SD CS as a ground - so I disconnected it, but the problem persisted. So, I started removing other connections to the SD Adaptor. Problem persisted until I removed the MISO connection to the SD Adaptor. I restored all other connections to the SD Adaptor, leaving MISO unconnected, and the Touch interface appears to work (obviously the SD adaptor won’t work, but issue now seems to be narrowed to MISO connection):

4:02:45.649 -> x: 0     y: 3739     z: 15 
14:02:45.892 -> x: 0     y: 3718     z: 17 
14:02:46.141 -> x: 0     y: 3727     z: 39 
14:02:46.391 -> x: 1654     y: 1948     z: 1850 
14:02:46.642 -> x: 1230     y: 2585     z: 1743 
14:02:46.884 -> x: 1074     y: 2353     z: 1592 
14:02:47.137 -> x: 1786     y: 2631     z: 1803 
14:02:47.381 -> x: 149     y: 3537     z: 34 
14:02:47.626 -> x: 301     y: 3439     z: 8 

I will continue investigating. Again. thanks for the suggestions and help :)

Bodmer, 2020-07-10:
That is a good bit of investigative work.

It looks like the MISO output does not tri-state when the SD chip select is high. Look at the SD card adapter and see if it has a buffer or a 5V compatible level translator chip which is placing a load on the MISO line.

Ideally post a picture of both sides of the SD card adapter, or provide a link to the vendors page.

The XPT2046 chips have a fairly weak output driver so is may not be able to overcome any line load placed by the SD adapter on the MISO line.

Check your SD card still works in a PC.

The touch controller SPI clock speed has no effect so put it back to 2500000.

Bodmer, 2020-07-10:
OK, you appear to have [one of these](https://www.ebay.co.uk/i/112043107525?chn=ps&norover=1&mkevt=1&mkrid=710-134428-41853-0&mkcid=2&itemid=112043107525&targetid=908661474856&device=c&mktype=pla&googleloc=9045614&poi=&campaignid=10195651586&mkgroupid=107296210212&rlsatarget=pla-908661474856&abcId=1145987&merchantid=115120049&gclid=EAIaIQobChMIi-b309XD6gIVAuDtCh2GXQ5HEAQYASABEgJFqPD_BwE). This will not work as it has an output level translator that can not be tri-stated.  You need [one like this](https://uk.banggood.com/SD-Card-Module-Slot-Socket-Reader-Mp3-player-p-74105.html?gmcCountry=GB&currency=GBP&createTmp=1&utm_source=googleshopping&utm_medium=cpc_bgcs&utm_content=lijing&utm_campaign=ssc-gbg-all-newcustom-0822&gclid=EAIaIQobChMI1K2S0tbD6gIVRevtCh28NAH-EAQYASABEgIrA_D_BwE&cur_warehouse=CN).

Pizza-Monster, 2020-07-10:
Great! Coincidentally, I just received such an SD Card Reader Module from Amazon today (I ordered it incase the micro SD adaptor didn't work). I will try this out tonight or tomorrow. Thanks!

Pizza-Monster, 2020-07-11:
Wow! It is feast or famine! I actually now have 2 different working solutions!

Solution 1:
I used your suggestion and replaced the Micro SD Adaptor with SD Card Reader Module 3.3V to 5V Card Adapter and this works! Touch and SD play nicely together!
( https://www.amazon.com/gp/product/B06XH3FCN1/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1 )

Solution 2 :
I kept thinking about your statement: The XPT2046 chips have a fairly weak output driver so is may not be able to overcome any line load placed by the SD adapter on the MISO line. I am very new to electronics (software is my background), but I wondered what would happen if I connected the MISO from the ESP32 to the Touch, and then the MISO from the Touch to a resistor and then connect the resistor to the MISO on the Micro SD Adaptor. Would that sufficiently weaken SD MISO signal? It turns out that it does work when I use 1K resistor. Touch and micro SD play nicely together! I don’t know if what I did is good practice or if it will  introduce instability elsewhere in the circuit, but it appears to work. I also don’t know if the 1K value is the optimum resistance for this purpose. What I like about this solution is that I can use the micro SD card form factor.
( https://www.amazon.com/gp/product/B07BJ2P6X6/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1 )

I want to thank you very much for helping me with this problem, especially since it was a problem with my circuit design and not the library. I also want to say that I really like the TFT_eSPI library - it has so much functionality and covers such a wide range of devices. Thank you!

Bodmer, 2020-07-11:
Great! I like good news. The 1k series resistor is fine, am output can easily drive ~3.3mA to overcome the SD buffer output.

closed 2020-07-11 by Bodmer