Bodmer/TFT_eSPI #691: Modifying Font16.c

leadenmoth, 2020-07-11:
Hi! Apologies for raising an issue - this is more of a usage question. I'd like to add more characters to Font16.c - it seems to be easy enough structurally, with each character as a separate array, each value looks to be an 8-pixel line, makes total sense (unlike gfx fonts, for example). But how did you generate that font file in the first place? Things like Adafruit's fontconverter or Mikroelektronika's GLCD Font Creator seem to be following completely different structures.

Context: TFT_eSPI is used by CircuitMess Ringo DIY phone, they are using Font16.c to display SMS messages and I'd like to expand language support there. Switching it to one of the GFX fonts opens a completely different can of worms (finding a way to convert multiple non-consecutive character subsets into one font file, for example)

Bodmer, 2020-07-12:
I use a piece of software called "Cosmisvoid's Font Editor" (I think it was called called CFE_64) which is free and contains some fonts. The fonts were created 6+ years ago but I think the software can be found on Crystalfontz website and was I believe updated in 2018.

xfdr0805, 2020-07-13:
> I use a piece of software called "Cosmisvoid's Font Editor" (I think it was called called CFE_64) which is free and contains some fonts. The fonts were created 6+ years ago but I think the software can be found on Crystalfontz website and was I believe updated in 2018.

Is there a tool to generate this rle font? thanks

Bodmer, 2020-07-13:
Font16.c is not an RLE font. So the raw bitmaps are used from the font generator.

leadenmoth, 2020-07-14:
@Bodmer thank you. I won't be able to use the app directly (it has a hardcoded limit with highest character code being 511), but it might help

closed 2020-07-14 by leadenmoth

Bodmer, 2020-07-14:
Note that each row in the bitmap is "N" bytes where N is typically 1, For wide characters like "W" a row in the bitmap will be 2 If a bitmap image is 9 bits wide then 7 bits in that row are wasted, but there are relatively few characters that need 2 bytes in a row.

Post again if you need more help.