Bodmer/TFT_eSPI #693: TFT_eSprite::printToSprite cuts of string when using space char.

dr-gino, 2020-07-13:
Situation is as follows:

I have used Processing IDE to create a VLW font that includes the space character code point (0x20).
To mitigate flickering while drawing smooth fonts I use ```printToSprite``` however when the string I'm trying to print contains space characters the string gets cut off on the screen.
When I comment out the ```if (code == 0x20)``` block in ```drawGlyph``` in Sprite.cpp it prints the string correctly.

It seems that the ```printToSprite``` function determines the required width for the sprite by querying the width for every character in the string, which is fine. My space character has a width of 4, but the ```drawGlyph``` function seems to want to use a shortcut when printing space characters, it adds ```gFont.spaceWidth``` to ```cursor_x``` whenever the ```code``` argument is 0x20. ```gFont.spaceWidth``` is never actually properly set to the actual width of the space char in the font but is inferred. ```spaceWidth```ends up being 7 when debugging so the total length if the string will then start to exceed to width of the sprite reserved initially.

Am I not supposed to add the space code point to my font?

Bodmer, 2020-07-14:
Do not include space. [See here](https://github.com/Bodmer/TFT_eSPI/blob/master/Tools/Create_Smooth_Font/Create_font/Create_font.pde#L58-L59).

Bodmer, 2020-07-14:
If that does not fix the problem, please post a full example sketch that illustrates the problem.

dr-gino, 2020-07-14:
> 
> 
> Do not include space. [See here](https://github.com/Bodmer/TFT_eSPI/blob/master/Tools/Create_Smooth_Font/Create_font/Create_font.pde#L58-L59).

Thanks, didn't catch that. Works fine if I leave the space out of the font.

closed 2020-07-14 by dr-gino