Bodmer/TFT_eSPI #694: I changed nothing but arduino ide doesn't compile the library

TLS1000, 2020-07-14:
Hi, after using the library two years or so -with great content-, my projects don't compile anymore.
I recently upgraded the IDE, and after that the programs still compiled successfully.

However since yesterday it returns like very basic errors on the .ccp file while I didn't touch that file.
I'm using IDE version 1.8.12, and espi version 1.4.16. 

this is de compiler return:
`U:\NewPrimary\My Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:5427:0: error: unterminated #ifdef

 #ifdef ESP32_PARALLEL

 ^

U:\NewPrimary\My Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:4615:0: error: unterminated #ifdef

 #ifdef LOAD_GFXFF

 ^

U:\NewPrimary\My Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp: In member function 'int16_t TFT_eSPI::drawString(const char*, int32_t, int32_t, uint8_t)':

U:\NewPrimary\My Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:4686:32: error: qualified-id in declaration before '(' token

 inline void TFT_eSPI::spi_begin(void){

                                ^

U:\NewPrimary\My Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:5426:1: error: expected '}' at end of input

 {

 ^

U:\NewPrimary\My Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:5426:1: error: expected '}' at end of input

U:\NewPrimary\My Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:5426:1: error: expected '}' at end of input`

Bodmer, 2020-07-14:
Upgrade to the latest released TFT_eSPI version 2.2.14 using the IDE menu or download the latest 2.2.15 direct from Github.

If that does not fix it then report back.

Make note of your settings or backup the setup files first if you have updated them.

TLS1000, 2020-07-15:
ok, I was a bit reluctant to upgrade the library because of the altered setup files I created like two or three years ago. But luckely I followed your advice (readme file) making my own setup folder at the time. the update performed without drama, everything seems to work.

thank you for this great library, it's one of the cornerstones in most of my projects.

would be nice to have features:
- menu tree navigation support
- screen scrolling
- all the fuzz about the fonts have passed my programming capabilities.

closed 2020-07-15 by TLS1000

Bodmer, 2020-07-15:
OK, it sounds like an incompatibility with old setup files was introduced at some point.

The Arduino[ LVGL library](https://github.com/lvgl/lv_arduino) is the way to go for a GUI, there is [very good documentation here](https://lvgl.io/), but it is a library where more advanced programming knowledge is needed or a lot of patience.

I may add some GUI features to a new library at some point but currently I am engaged in other projects.

Thanks for the appreciative comments.