Bodmer/TFT_eSPI #696: the CS of the TFT

shalom198, 2020-07-15:
hello 
is the CS of the tft can works with D0 pin ? 
is the DC can works with D0 pin ?

Bodmer, 2020-07-15:
That depends on which processor board you are using.

shalom198, 2020-07-16:
wemose mini D1

Bodmer, 2020-07-16:
You can use D0 however make on NodeMCU boards that is GPIO 16. Some WEMOS boards have a different mapping. Use:
```
#define TFT_CS D0
```
rather than PIN_D0

shalom198, 2020-07-16:
i use mini D1
 #define TFT_CS D0 -> that exactly what i did , and works fine
tell me, i had 2 pins D8 & D0 for the DC & CS , Does it matter, who to
connect to whom ?
I use D8 for another purpose, to run a relay from time to time, of course
when i use the relay, the software is in a state that does not use the TFT
if i use the TFT , the relay not in use !

‫בתאריך יום ה׳, 16 ביולי 2020 ב-19:59 מאת ‪Bodmer‬‏ <‪
notifications@github.com‬‏>:‬

shalom198, 2020-07-16:
Another question, if possible please
Is it possible to connect TX and RX to DC and CS

‫בתאריך יום ה׳, 16 ביולי 2020 ב-21:23 מאת שלום יוסף <‪shalom198@gmail.com
‬‏>:‬

Bodmer, 2020-07-16:
DC and CS should not be shared with any other device unless that device can be disabled by an independent means that will then prevent the toggling of DC and CS that occurs during acces of the TFT from affecting that device. I suggest you do not use TX and RX as you then lose debug capability via serial messages.

closed 2020-07-16 by Bodmer

shalom198, 2020-07-17:
thanks alot