Bodmer/TFT_eSPI #697: Filled rectangle sprite possible bug - when start x-coord is odd number

patfelst, 2020-07-18:
Hi there, I'm using a ST7735 160x80 pixels with an ESP32 with PSRAM (TinyPico). I'm using a sprite to prevent flickering on a rectangular bargraph. The bargraph has a "zero" point at 40 pixels from the left. Positive data makes the bargraph increase to the right, data negative values make it "increase" to the left from the "zero" point.

I'm getting what looks like a bug for the negative data values, I think it's because the starting x-coord of the rectangle changes almost every new data value (sample), whereas the positive samples always start at x=40.

I've narrowed the "bug" down so that if the starting x-coord of the rectangle is even, the rectangle draws ok. If I change the x-coord by even 1-pixel, or basically **x-coord is any odd value, then the rectangle is drawn broken**. It draws an unfilled rectangle at what appearas to be the correct position, but the filled in part is off to the right. It is independent of the width of the rectangle. It happens whether I use a fillRoundRect or fillRect. Unfilled rectangles (drawRect) are ok. **If I don't use sprites, it works ok**.

Hoping you can help please, and thanks so much for this library!

I've hard coded the coordinates and width for the purposes of this issue. Also attached some photos:
```c++
TFT_eSPI tft = TFT_eSPI();  // Invoke library, pins defined in 'src/Pats_Setup_ST7735.h'
TFT_eSprite sprite_bst_numeric = TFT_eSprite(&tft);  // Declare Sprite object "spr" with pointer to "tft" object
TFT_eSprite sprite_bargraph = TFT_eSprite(&tft);  // Declare Sprite object "spr" with pointer to "tft" object

#define BARGRAPH_HEIGHT 32

void st7735_160x80_init() {
    
  tft.init();
  tft.setRotation(1);
  tft.setTextSize(1);

  sprite_bargraph.setColorDepth(4); // 16 colours
  sprite_bargraph.setAttribute(PSRAM_ENABLE, true);
  sprite_bargraph.createSprite(tft.width(), BARGRAPH_HEIGHT);

  // Draw a faint outline that the bargraph sits inside of
  sprite_bargraph.drawRoundRect(0, 0, tft.width(), BARGRAPH_HEIGHT, 4, _4BIT_DARKGREY);

}

void test_rect_bug() {
  sprite_bargraph.fillRoundRect(23, 1, 30, 30, 4, _4BIT_PURPLE);
  sprite_bargraph.fillRect(111, 1, 15, 30, _4BIT_YELLOW);
  sprite_bargraph.pushSprite(0, 0);
}
```
Both rectangles ok with even x-coords:
[image]

Filled rectangle x-coord changed by 1-pixel (23), odd number, broken:
[image]

Close up of the above case:
[image]

And finally, the unfilled yellow rectangle, change x-coord from 110 to 111:
[image]

Bodmer, 2020-07-19:
Hi,
Thanks for reporting this. It is indeed a bug.

I have updated the code but am unable to test it today due to other commitments.  Can you download the updated Sprite.cpp file (or complete library from Github) and test please.

closed 2020-07-19 by Bodmer

patfelst, 2020-07-19:
I've just tested it thoroughly, it's fixed now. Thanks for doing so quickly!

patfelst, 2020-07-19:
I just realised I only tested .fillRoundRect, I won't be able to test .fillRect for a couple of days. Unless it's the same / inherited code.

Bodmer, 2020-07-20:
fillRoundRect uses fillRect so the problem would have shown up.
Thanks.

commit c124688a (2020-07-19): Fix #697

Untested but looks correct!