Bodmer/TFT_eSPI #698: ringmeter on ESP32 which connect to 4" ST7796 via SPI

lysiong, 2020-07-19:
Hi,
I'm using ESP32 on ST7796(SPI mode) which display specification can be found on below link
[http://www.lcdwiki.com/4.0inch_SPI_Module_ST7796](url)

I been working out on ringmeter for display my car info but being stuck with screen refresh & decimal issue.

below is my sample code (my actual draft actually have 11 ringmeter but I just simplify it to 4 meters for easy discussion)
[ESP32_ST7796_meter_ring.txt](https://github.com/Bodmer/TFT_eSPI/files/4943899/ESP32_ST7796_meter_ring.txt)

Problem 1:
when the data change to other value on display, it seem like there is a residual pixel remain on screen, refer to below picture
[image]
how do I clear the residual pixel everytime when there is a change of data?

Problem 2:
how do I individually set decimal point for each meter?
for example, 
HV Fan Speed only required number without decimal point.  
AVG Cell V required 2 decimal point.
HV Batt V required 1 decimal point
HV SOC % same requirement as HV Fan Speed which without decimal point.

Thanks in advance.

Bodmer, 2020-07-20:
The residual pixels occur because the string width is different for numbers with different digit counts. So "1" will bot obliterate "999" and will look like "919". The string needs to be padded out to always by the same width. One way to do this is to add spacesbefore and after the 1 so the string becomes "_1_" where _ marks the space in this example. However this approach does not work wel for poroportions fonts because characters can vary in width, so "i" in narrower than "W", also the space character may be arrower than a "W".

To avoid this problem there is a function called setTextPadding, which defined the width to be blanked. It appears you have removed this function from the example.

To control whether you want a decimal point or not you can either create new functions, that uses drawNumber and one that uses drawFloat. Or you could pass a new parameter to the function via to define how many decimal places to use.

lysiong, 2020-07-20:
Thanks for reply, I will try find back the setTextPadding function from example. 

For the decimal matter, can u give example how to do it? I not very sure where to insert the command into current draft.

Bodmer, 2020-07-24:
Have a look at the [function prototypes here]( https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h#L478-L485) then you can avoid dtostrf().

closed 2020-07-24 by Bodmer