Bodmer/TFT_eSPI #699: White screen with ST7735 and STM32H743

bacteriu, 2020-07-20:
Hello, I have used TFT_eSPI with various displays in ESP32 without any problem so this library I love it, but now I would like to use it on a board with an STM32H743VIT6 and I only get a blank display, looking at the pins with the oscilloscope I see that SCLK and MOSI is missing, the others are fine but I don't know why no signal appears in MOSI and SCLK. Is there any way to get the display to work on this processor even without DMA?

Bodmer, 2020-07-22:
The STM32H743VIT6 has a different peripheral interface API at the low level so you probably need to use the standard Arduino board package interface calls. To do this remove the #define STM32 from your setup file.

bacteriu, 2020-07-23:
Hello, again, when I delete the #define STM32 the program hangs when executing tft.init(), it never returns from this function, so that does not happen if the #define STM32 and the #define NUCLEO_144_TFT are present if I delete any of them tft.init () never returns.

Bodmer, 2020-07-24:
Is the STM32H743VIT6 supported by the Arduino environment?

Have you used the SPI port 1 pins?

Bodmer, 2020-07-24:
I have tried a Nucleo STM32H743ZI2 and it works OK. I suspect the Arduino IDE does not support the STM32H743VIT6. Please check.

bacteriu, 2020-07-28:
Yes, I am using SPI1.
I will do a little tracing to check which function it hangs and I will comment later where, since it could indeed be in STM32duino core.
Thank you very much for your quick responses.

bacteriu, 2020-08-01:
Sorry to take so long.
I think the definitions of the TFT_MOSI, TFT_SCLK and TFT_MISO pins defined in User_Setup.h do not reach the spi_init () function in the file spi_com.c of STM32Duino.
Is it possible that I am missing some definition of these pins somewhere?
Are there any TFT_eSPI files besides User_Setup.h where those pins must be defined for the STM32Duino SPI to use?

Bodmer, 2020-08-01:
The STM32 pins are defined in User_Setup.h or in your custom file, [see here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L241-L263) for examples.

There is a[ diagnostic sketch](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Test%20and%20diagnostics/Read_User_Setup) you can run on the board, this does not display anything on the TFT but generates a serial output report visible via the serial monitor window.

bacteriu, 2020-08-02:
When tft.init () is executed, it hangs, except
if in the User_Setup.h I have defined STM32 and Nucleo_144, then the result of Read_User_Setup.ino is the following:

[code]
TFT_eSPI ver = 2.2.14
Processor    = STM32
Transactions = Yes
Interface    = SPI
Display driver = 7735
Display width  = 128
Display height = 160

MOSI    = GPIO 77
SCK     = GPIO 55

>>>>> Note: STM32 pin references above D15 may not reflect board markings <<<<<
TFT_CS   = D16
TFT_DC   = D37
TFT_RST  = D36

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.00
[/code]

and in my User_Setup.h I have defined the pins in the following way:

// ######       EDIT THE PINs BELOW TO SUIT YOUR STM32 SPI TFT SETUP        ######

// The TFT can be connected to SPI port 1 or 2
#define TFT_SPI_PORT 1 // SPI port 1 maximum clock rate is 55MHz
#define TFT_SCLK	PA5
#define TFT_MOSI	PA7
#define TFT_RST		PC4  		// Reset pin (could connect to NodeMCU RST, see next line)
#define TFT_DC		PC5  		// Data Command control pin
#define TFT_CS		PB0  		// Chip select control pin PB_0

bacteriu, 2020-08-05:
Ok, if before the tft.init () I use the functions
   SPI.setMISO (TFT_MISO);
   SPI.setMOSI (TFT_MOSI);
   SPI.setSCLK (TFT_SCLK);
the display works and almost all the examples work correctly, although, accustomed to ESP32 where it was not necessary to initialize those pins in the main program if they were previously defined in the User_Setup.h, I do not understand well because now I have to initialize them before executing tft .init ().

Overcome this problem now I find another one in the function testHaD () of the example TFT_graphicstest_PDQ3, in that function testHaD () only one iteration of the loop "for (int i = 0; i <0x10; i ++)" is executed, the second iteration occurs a hang on the tft.pushColor (curcolor, cnt); and it seems to be related to the tft.endWrite () after the "while (cmp <& HaD_128x160 [sizeof (HaD_128x160)])" because if I remove it everything works correctly.

Thank you very much for your help.

Bodmer, 2020-08-11:
If the STM32H743 was supported by the Arduino environment then the changes would not be necessary.

Glad to hear you have a work-around solution.

closed 2020-08-11 by Bodmer