Bodmer/TFT_eSPI #700: Add more virtual modifiers to existing functions in TFT_eSPI to allow for polymorphism.

dr-gino, 2020-07-22:
The TFT_SPIFFS_BMP sample has a nice drawBmp function that draws directly to the screen. I wanted to add an extra argument to the drawBmp function to allow drawing on either the screen or a sprite. 
e.g.:
```void drawBmp(TFT_eSPI &surf, const char *filename, int16_t x, int16_t y)```

My intention was to be able to pass either an instance of TFT_eSPI or TFT_eSprite to it. However the functions used (```getSwapBytes```, ```setSwapBytes```, ```pushImage```) don't have the ```virtual``` modifier set in ```TFT_eSPI``` that will allow for polymorphism to work. So when passing in a sprite it always draws on the screen instead.

Adding the ```virtual``` keyword to those function in the header file made it work. May I suggest making all functions that are overridden by TFT_eSprite, ```virtual``` in ```TFT_eSprite```? This allows for writing functions that can take either the screen or a sprite as the surface to draw on.

Bodmer, 2020-07-24:
OK I can do that!

To show library users the possibilities, can you first create a well commented example demo sketch. Also it would be useful to list what other functions that you think could be usefully made virtual. Bear in mind this is an Arduino library and the more esoteric capabilities in C++ are likely to remain a mystery to most users so examples are the key to such features being used.

dr-gino, 2020-07-26:
I'm not sure how much these changes warrant a new example since I noticed that in the ```TFT_eSPI``` class that a couple of member functions such as ```drawPixel(), drawChar(), height(), width()``` etc. were already marked virtual, it only makes sense to extend it to other functions. Here's a list of functions in the TFT_eSPI that should be marked as virtual as well in the TFT_eSPI.h file:
```
setSwapBytes
getSwapBytes
pushImage (int32_t, int32_t, int32_t, int32_t, const uint16_t *)
pushImage (int32_t, int32_t, int32_t, int32_t, uint16_t *)
readPixel
setPivot
getPivotX
getPivotY
setRotation
getRotation
```

I slightly modified the `drawBMP()` function from the TFT_SPIFFS_BMP example to accept a surf argument. This allows you to use that same function for drawing directly on the screen but also pass in a TFT_eSprite. Much nicer than duplicating the function for both cases.

```
void drawBmp(TFT_eSPI &surf, const char *filename, int16_t x, int16_t y) {

  if ((x >= surf.width()) || (y >= surf.height())) return;

  fs::File bmpFS;

  // Open requested file on SD card
  bmpFS = SPIFFS.open(filename, "r");

  if (!bmpFS)
  {
    Serial.print("File not found");
    return;
  }

  uint32_t seekOffset;
  uint16_t w, h, row, col;
  uint8_t  r, g, b;

  uint32_t startTime = millis();

  if (read16(bmpFS) == 0x4D42)
  {
    read32(bmpFS);
    read32(bmpFS);
    seekOffset = read32(bmpFS);
    read32(bmpFS);
    w = read32(bmpFS);
    h = read32(bmpFS);

    if ((read16(bmpFS) == 1) && (read16(bmpFS) == 24) && (read32(bmpFS) == 0))
    {
      y += h - 1;

      bool oldSwapBytes = surf.getSwapBytes();
      surf.setSwapBytes(true);
      bmpFS.seek(seekOffset);

      uint16_t padding = (4 - ((w * 3) & 3)) & 3;
      uint8_t lineBuffer[w * 3 + padding];

      for (row = 0; row < h; row++) {
        
        bmpFS.read(lineBuffer, sizeof(lineBuffer));
        uint8_t*  bptr = lineBuffer;
        uint16_t* tptr = (uint16_t*)lineBuffer;
        // Convert 24 to 16 bit colours
        for (uint16_t col = 0; col < w; col++)
        {
          b = *bptr++;
          g = *bptr++;
          r = *bptr++;
          *tptr++ = ((r & 0xF8) << 8) | ((g & 0xFC) << 3) | (b >> 3);
        }

        // Push the pixel row to screen, pushImage will crop the line if needed
        // y is decremented as the BMP image is drawn bottom up
        surf.pushImage(x, y--, w, 1, (uint16_t*)lineBuffer);
      }
      surf.setSwapBytes(oldSwapBytes);
      Serial.print("Loaded in "); Serial.print(millis() - startTime);
      Serial.println(" ms");
    }
    else Serial.println("BMP format not recognized.");
  }
  bmpFS.close();
}
```

I hope this works for you, I'm pretty new to git and Github, I guess normally one would submit a pull request for these kind of modifications? I've never done that before but if it's more convenient for you I'll look into it. Let me know.

dr-gino, 2020-09-24:
I made a pull request a while ago, did you get a chance to check it out yet?

Bodmer, 2020-11-28:
This increases code size and I suspect the majority of Arduino users will not benefit. The changes are quick to do by the users who wish to use polymorphism.

closed 2020-11-28 by Bodmer