Bodmer/TFT_eSPI #705: ILI9488 & ESP32

smiffycom, 2020-07-24:
Anyone else having problems with ILI9488 displays on ESP32 using TFT_eSPI? Initially, I couldn't get the touch demo to run (or any of the 480x320 demos to run). So I downloaded the latest version of TFT_eSPI, but now all I get is a white screen, serial debug shows the code is running but the screen just stays white - stumped... All the settings in user.settup are the same

Bodmer, 2020-07-24:
Check the setup with the Read_User_Setup diagnostic sketch.

Do NOT connect MISO from the ILI9488 to the ESP32 as that pin does not tristate.  If you wish to read the display ass a 1K series resistor in the MISO signal line to the TFT, connect the touch controller SDO (aka MISO) direct to the ESP32, not through that resistor).

smiffycom, 2020-07-24:
This is the result of running the diag sketch. The actual user.setup.h had changed my TFT reset pin to -1 (still got white screen), I changed it back to 33 (still white screen)
All the other pins are correct.
[code]
TFT_eSPI ver = 2.2.14
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9488
Display width  = 320
Display height = 480

MOSI    = GPIO 23
MISO    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 14
TFT_DC   = GPIO 27
TOUCH_CS = GPIO 25

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Smooth font enabled

Display
SPI frequency = 27.00
Touch SPI frequency   = 2.50
[/code]

Bodmer, 2020-07-24:
TFT_RST should show up in the diagnostic output. Make sure it is in setup and run Read_User_Setup again and post output.

Provide a weblink to the display vendors page.
Post a picture of your display+wiring + ESP32 setup.
Post a picture of the back of the board.

TheDefpom, 2020-07-25:
Check the CS pin definition, its default is 15 in the setup info, BUT the actual pin for VSPI CS is 5 on the ESP32... unless of course you are actually using pin 14 as shown in your diag. I had a white screen too until I noticed that difference.

smiffycom, 2020-07-25:
I've had limited success, I found where the rogue TFT_RST = -1 comes from, it comes from User_Setups/Setup21_ILI9488.h (it was defaulting to -1)
I set it 33 and now it sort of works, interesting side effect with the new version - any of the icons I use (pushimage) are now inverted.

I bought the display in February this year, but the vendor no longer lists the item.
1 picture shows the inverted icon (well, the colours are wrong).

I hope you appreciate I dismantled my weather station for the pix - LOL
But I do appreciate your help...

Many regards,

smiffy

TFT_RST  = GPIO 33

Bodmer, 2020-07-26:
I do not see any pictures.  The colour errors suggest the display is not an ILI9488, post a picture of the back of the display so I can compare to the one I have. Color errors can be corrected but it depends what they are (wrong color depth setting in TFT, byte swap, red/green swap).

smiffycom, 2020-07-26:
I did attach photos to the last post, hmmmm.
The colours were fine with the older version of TFT_eSPI.
I'll attach the photos (again) to this E-mail... If they don't arrive, give me a direct E-mail address.

Regards,

smiffy

Bodmer, 2020-07-28:
Just drag the images to the Github write pane, or paste a weblink. I don't provide support via email.

smiffycom, 2020-07-28:
Here ya go :)
[diagnostics.txt](https://github.com/Bodmer/TFT_eSPI/files/4985440/diagnostics.txt)
[image]
[image]
[image]
[image]

Bodmer, 2020-07-30:
The icon has byte reversal. You need to include the line tft.setSwapBytes(true); before the icon is rendered or in setup. If that does not work the try using tft.setSwapBytes(false) ;

smiffycom, 2020-07-30:
Excellent Sir, I included tft.setSwapBytes(true); in setup and all is well. Thank you for all your help.

Regards,

smiffy

closed 2020-07-30 by Bodmer