Bodmer/TFT_eSPI #706: Enhancement: Adding new "bitmap" method to the Sprite Class and "copySprite" to the TFT Class

toetoast, 2020-07-25:
I have 2 enhancements here for consideration that I've been using for quite a while now.

The 1st enhancement adds a new method to the TFT_eSprite Class called "bitmap" shown below.
The purpose of this is to return the memory location of the bitmap data contained within the Sprite.
I understand that this memory location is returned as a pointer upon Sprite creation, but I feel it's very handy to use a method like this when you have many Sprites defined in your system.

This has been designed for Sprites using 16bit pixels only, but could be expanded to work with 1, 4 or 8 bits per pixel.
However, at present I'm not familier with how this is done.
```
/***************************************************************************************
** Function name:           bitmap
** Description:             Returns the location of the Sprite bitmap in memory
***************************************************************************************/
uint16_t* TFT_eSprite::bitmap(void)
{
  if (!_created) return 0;

  if (getColorDepth() != 16) return 0;	

  return _img;  
}
```
The 2nd enhancement (shown below) adds a new method to the TFT_eSPI Class called "copySprite". I'm presenting it here for the following reasons:

- It's an example on how the above Sprite "bitmap" method can be used :)
- Sometimes it's useful to be able to make an exact copy of a Sprite. Maybe the user wants to create the copy as a backup, eg the orginal Sprite is going to be altered (before being displayed) and then it needs to be reverted back or some other operation performed. Again, this one only supports 16bits per pixel at the moment.

My Sketches use "copySprite" as a function defined within the Sketch, but I would like to add it directly as a new method to the TFT_eSPI Class. I've tried to do this but get stuck when trying to reference the source and destination Sprite instances **from within** the TFT_eSPI Class.
_Question:_ Can anyone help please with suggestions on how to convert this function to a TFT_eSPI Class method ? Once I know how to do this I have some more useful functions that I'd like to add to the TFT_eSPI Class.

Thanks
Tony

```
void copySprite(TFT_eSprite *_sprSrc, TFT_eSprite *_sprDest)
{
    uint16_t sprSrcW;
    uint16_t sprSrcH;
    uint16_t sprDestW;
    uint16_t sprDestH;
    uint16_t *sprSrcBitmap;
    uint16_t *sprDestBitmap;

    if (!_sprSrc->created() || !_sprDest->created())
    {
        Serial.println(F("Undefined Sprite"));
        return;
    }

    if ((_sprSrc->getColorDepth() != 16) || (_sprDest->getColorDepth() != 16))
    {
        Serial.println(F("Sprite is not 16bpp"));
        return;    
    }

    sprSrcW = _sprSrc->width();
    sprSrcH = _sprSrc->height();
    sprDestW = _sprDest->width();
    sprDestH = _sprDest->height();   
    sprSrcBitmap = _sprSrc->bitmap();
    sprDestBitmap = _sprDest->bitmap();

    if ((sprSrcW == sprDestW) && (sprSrcH == sprDestH))     // Only perform the copy operation if the Sprites are the same size.
    {
        memcpy(sprDestBitmap, sprSrcBitmap, (sprSrcW * sprSrcH * 2));  // Need to double the copy size since memcpy deals with Bytes not Integers. 
    }    
}
```

Bodmer, 2020-07-25:
The capability to copy a Sprite into another sprite already exists. A sprite is simply an image stored in RAM so you can use code like this:
```
// Create the instances and global pointers
TFT_eSprite spr1 = TFT_eSprite(&tft);
TFT_eSprite spr2 = TFT_eSprite(&tft);
uint16_t* sp1rPtr;
uint16_t* spr2Ptr;

// Create the sprites and save the pointers
  spr1.setColorDepth(COLOR_DEPTH);
  spr1Ptr = (uint16_t*)spr1.createSprite(50, 50);
  spr1.fillSprite(TFT_BLACK);
  spr2.setColorDepth(COLOR_DEPTH);
  spr2Pt2 = (uint16_t*)spr2.createSprite(25,25);
  spr1.fillSprite(TFT_WHITE);

// Copy sprite 1 into sprite 2 at position 5,5
  spr2.pushImage(5, 5, spr1.width(), spr1.height(), spr1Ptr);

//Push sprite 2  to TFT screen
  spr2.pushSprite(0, 0);
```

At the moment this works if sprite 1 and sprite 2 have the same color bit depth of 1 or 16 bits. Bit depths of 4 and 8 will be supported, I have an update pending to replace the existing untested code. The sprites can be of different sizes (but not color depths at the moment, but supporting differences is planned) and the copy position can be set anywhere in the destination sprite. Eventually the process will have a simpler function call (passing an instance instance pointer instead of a memory pointer e.g. spr2.pushSprite(0, 0, &spr1)), similar to the one as you suggest as then different color depths can be handled, but the basic functionality you require for 16 bit sprite copying exists already.

I will add this getPointer function in the next release:

```
/***************************************************************************************
** Function name:           getPointer
** Description:             Returns pointer to start of sprite memory area
*************************************************************************************x*/
void* TFT_eSprite::getPointer(void)
{
  if (!_created) return nullptr;
  return _img8;
}

```

then you can use:
```
sprPtr1 = (uint16_t*) spr1.getPointer(); // Cast to 16 bit pointer
```

Be aware that you can get a nullptr returned so your sketch should check for this.

toetoast, 2020-07-27:
That's great you have the 'getPointer' function in the pipline for a future release - I'll start using that instead of mine, since it gives the ability for it to be used on more than 16bpp Sprites.

Just one more question please. How would I go about having a new Method in the TFT_eSPI Class access Sprite info using it's instance pointer? As an example, this is what I'm trying to achieve in my Sketch using an example method:

TFT.performSomeActionOnSprite(&sprSrc)

I'm wanting this TFT Method to have the ability to process/act-upon the Sprite that was passed to it. I've spent some time trying to figure out the Method declaration in the code (and subsequent header too) with no such luck :(

Cheers
Tony

Bodmer, 2020-07-28:
The TFT_eSPI class is a parent of the Sprite class, so this inheritance means TFT_eSPI class has no direct access to the Sprite class functions or public varaibles and will it not recognise a sprite class instance or sprite class instance pointer.

The correct place for the new function would be in the sprite class itself. [This function](https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Sprite.h#L98) can be used as an example, it uses an instance pointer to another sprite, thus one sprite can access the functions in another sprite class instance. A sprite class can also access the tft class functions as demonstrated in [this function](https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Sprite.cpp#L666).

closed 2020-07-28 by Bodmer

toetoast, 2020-07-28:
That's great, I'm learning lots thanks!