Bodmer/TFT_eSPI #708: 5% speed improvement with 16bpp full screen sprite transfers

pjmi1, 2020-07-26:
I have been using a full screen sprite as a frame buffer, which has a background bitmap copied onto it, and get flicker-free, tear-free 50fps display rates (on a TTGO T-Display - ESP32 - ST7789 16bpp) - thanks to your library.

As a challenge, I managed to get a measured overall 5% speed improvement to this frame rate by optimising pushImage() and thought you might like to use it in the library.
The method is just to avoid conditionals (and multiplies) in the inner loop -  could also be done to other intensive loops.
```
void  TFT_eSprite::pushImage(int32_t x, int32_t y, int32_t w, int32_t h, uint16_t *data)
{
...
if (_bpp == 16) // Plot a 16 bpp image into a 16 bpp Sprite
  {
#if PJMI_CODE
	if (!_iswapBytes)
	{
		uint16_t * pdest = &_img[xs + (ys * _iwidth)];	// start addr in dest
		uint16_t * psrc = &data[xo + (yo * w)];	// start addr in source
		int32_t yc;
		for (yc = hs; yc > 0; yc--)
		{
			int32_t xc;
			for (xc = ws; xc > 0; xc--)
			{
				uint16_t color =  *psrc++;
				*pdest++ = (color<<8 | color>>8);
			}
			psrc += (-ws + w);
			pdest += (-ws + _iwidth);
		}
	}

	else
	{	// _swapbytes set
		uint16_t * pdest = &_img[xs + (ys * _iwidth)];	// start addr in dest
		uint16_t * psrc = &data[xo + (yo * w)];	// start addr in source
		int yc;
		for (yc = hs; yc > 0; yc--)
		{
			int xc;
			for (xc = ws; xc > 0; xc--)
			{
				*pdest++ = *psrc++;
			}
			psrc += (-ws + w);
			pdest += (-ws + _iwidth);
		}
	}
#else	// original code:
	for (uint32_t yp = yo; yp < yo + hs; yp++)
	{
		x = xs;
		for (uint32_t xp = xo; xp < xo + ws; xp++)
		{
		  uint16_t color =  data[xp + yp * w];
		  if(!_iswapBytes) color = color<<8 | color>>8;
		  _img[x + ys * _iwidth] = color;
		  x++;
		}
		ys++;
	}
#endif
  }
```

Bodmer, 2020-07-27:
I ran a test of the code by copying a 16 bit 80*80 sprite into another one 1000 times to see the raw performance gain within the function. e.g. this code snippet:
```
    uint32_t dt = millis();
    uint16_t n = 1000;
    while (n--) {
    // Copy sprite 1 into sprite 2
    spr2.pushImage(0, 0, spr1.width(), spr1.height(), spr1Ptr);
    }
    Serial.println(millis()-dt);
```
The following times are the results for an ESP32 in milliseconds:
```
No byte swap:    Original code = 588ms, suggested code = 393ms
With byte swap:  Original code = 600ms, suggested code = 410ms
```
So that is a useful performance gain especially for larger sprites.

The "if" will be the main culprit is slowing it down, the multiplies probably don't have much impact because they compute to a fixed value within the loop and the compiler recognises this and optimises it to a once per loop multiply.

Further optimisation is possible using memcpy as that is moves memory much faster, using 32bit words where possible.

This replacement code uses pre-calculated 8 bit pointers and memcpy:

```
  if (_bpp == 16) // Plot a 16 bpp image into a 16 bpp Sprite
  {
    uint8_t *ptro = (uint8_t *)data + ((xo + yo * w) << 1);
    uint8_t *ptrs = (uint8_t *)_img + ((xs + ys * _iwidth) << 1);

    if(_iswapBytes)
    {
      while (hs--)
      {
        // Fast copy with a 1 byte shift
        memcpy(ptrs+1, ptro, (ws<<1) - 1); 
        // Now correct just the even numbered bytes
        for (int32_t xp = 0; xp < (ws<<1); xp+=2)
        {
          ptrs[xp] = ptro[xp+1];;
        }
        ptro += w<<1;
        ptrs += _iwidth<<1;
      }
    }
    else
    {
      while (hs--)
      {
        memcpy(ptrs, ptro, ws<<1);
        ptro += w << 1;
        ptrs += _iwidth << 1;
      }
    }
  }
```
Results for this are good:

```
No byte swap:      48 ms
With byte swap: 373 ms
```
Of course the dramatic 12x performance performance improvement using memcpy won't translate to a big improvement in your actual project due to the relatively small time spent copying just one memory block compared to 1000, however it may give a few percent further improvemnt. I will update the library in the next week or so.

pjmi1, 2020-07-27:
That's good - why didn't I think of doing that - thanks!
Though, in your test code snippet haven't you left the sense of _iswapBytes inverted compared with original code, which does the swap if _iswapBytes is false. Perhaps the variable should be called _notswapBytes?

Testing your new code in my program - it's like you said regarding overall speed improvement.
Per plot loop,

```
With byte swapping: 
Original code: 20ms
My suggested code: 19ms
Your new code: 19ms
```

```
With no byte swapping:
Original code: 20ms
My suggested code: 19ms
Your new code: 17ms - a decent improvement - we've got from 50fps to 58fps.
```
I shall have to change my bitmap endianness to make use of this huge improvement. Maybe your library should have an endian matching function which pre-swaps the byte order - run once at initiation for RAM bitmaps?

Cheers!

Bodmer, 2020-07-27:
Re. byteswap, yes, there is an inconsistency in the library on Github within the pushRotated function that was corrected by an incosistency in the pushImage! My copy has the pushRotated function corrected but it is under test and has not been pushed to github yet. So, you should use the !_iswapBytes as you suggest for the time being if you are using pushRotated to copy a rotated sprite into another sprite.

16 bit sprite colours are always stored in RAM with swapped bytes compared to the normal ESP32 processor endianess. This is deliberate as it allows direct DMA of Sprites to the SPI port (for ESP32 and STM32) without further byte manipulation, so then the transmitted byte order is compatible with the TFT displays. Images from other sources can have a different endianess hence the setSwapBytes command options for both the TFT and Sprite classes. If you are using the TJPG_Decoder library then this also has its own comand to output swapped bytes for 16 bit colours as this is faster to do during conversion of the the jpeg 24 bit colour to 16 bit..

Bodmer, 2020-07-27:
You may find you can improve performance by using DMA or using both the ESP32 processor cores. The Arduno sketch runs on top of an RTOS so a task can be asigned to thhe 2nd processor. Performance gains depend on whether your sketch can do something useful with the 2nd core while the sketch updates the screen. A dual core [example is here](https://github.com/Bodmer/TJpg_Decoder/tree/master/examples/ESP32_Dual_Core_Flash_Jpg), in that sketch the jpeg decoding is done by the other processor core and the screen updated by the other. This dual core approach needs an understanding of how semaphores and yield() must be managed by each core to avoid crashes (note: global varaibles are avilable to both cores).

pjmi1, 2020-07-28:
Thanks - some really useful things for me to explore to go faster still, though, when I last had a look at ESP32 DMA I concluded that for RAM to RAM transfers it's not so useful - maxing out at 80MHz + overheads?

Bodmer, 2020-08-05:
Library updated, report any bugs in a new issue. Thanks.

closed 2020-08-05 by Bodmer