Bodmer/TFT_eSPI #709: Could not run ST7735S with ESP8266-12E

ago-cs, 2020-07-28:
Hello guys,

I spend about a week trying to determine the root of the problem and during this time I grew in my understanding of settings from a full 0 up to a really good understanding of what each line in the configuration is written for. But my ST7735S shines with background light ONLY if I link it to 5v and almost invisible (detected at night) if I link it to 3.3v.
The display itself is http://ali.onl/1DQf. On the reverse side of the PCB-panel, there is a special note: "_Driver IC:ST7735S_".

I have two the same brand new displays of this type, several NodeMCU Lolin V3 boards. I checked the wires. Played around with several configuration parameters that I didn't understand well like ST7735_INITB/ST7735_GREENTAB3/ST7735_REDTAB. etc.

I saw tens of videos and read a lot of articles but I didn't saw exactly my LCD there. 

**Hardware Configuration**
My screen's cover film has a green tab, but I checked all the lines.
I am using the most default settings:
```
TFT > Lolin NodeMCU V3
--- > -----------------
GND > GND
VDD > +5V/+3.3V
SCL > D5 (GPIO14)
SDA > D7 (GPIO13)
RST > D4 (GPIO2) but there I experimented with +3.3V with no results`
DC  > D3 (GPIO0)
CS  > D8 (GPIO15)
BLK > D2 (GPIO4) but I tried to link it to 3.3V/5V and even not plug at all
```

**Software Configuration**
I am working in VS Code / PlatformIO plug-in
I am directly updated <project-path>/.pio/libdeps/esp12e/TFT_eSPI/User_Setup.h
Here is what I have uncommented there:
```
#define ST7735_DRIVER  
#define TFT_WIDTH  128
#define TFT_HEIGHT 160
#define ST7735_REDTAB // I tried all 8 options but 
                      // according to internet
                      // red PCB with ST7735S 
                      // is mostly working with it.

#define TFT_MOSI PIN_D7
#define TFT_SCLK PIN_D5
#define TFT_CS   PIN_D8
#define TFT_DC   PIN_D3
#define TFT_RST  PIN_D4 // Also tried with PIN_D0

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF
#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000
```

Main.cpp
```
#include <TFT_eSPI.h>
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI(); 
#define TFT_GREY 0x5AEB

void setup()
{
  tft.init();
  tft.setRotation(1);
  tft.drawRect(50,50,70,70,TFT_GREEN);
}

void loop()
{
  tft.fillScreen(TFT_GREY);
  tft.setCursor(0, 0, 2);
  tft.setTextColor(TFT_WHITE,TFT_BLACK);
  tft.setTextSize(2);
  tft.println("Hello World!");
  while(1) yield(); // We must yield() to stop a watchdog timeout.
}
```

Probably it is something hardware but I already do not know what to check I missed? Any ideas, thank you in advance.
I also checked other libraries with the same result, your one looks the most perspective taking into account that I plan to work with e-ink as well.

Thank you again for your time!

ago-cs, 2020-07-28:
Forgot the details: model of 1.8 inch / 128x160

Bodmer, 2020-07-28:
Run the "Read_User_Setup" diagnostic sketch to make sure the compiler is picking up the right pins and settings etc.

There may be a configuration issue with PlatformIO, I do not use it so am unable to advise so trying the Arduino environment is an option.

Bodmer, 2020-07-28:
Run[ this sketch](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/320%20x%20240/Read_ID_bitbash/Read_ID_bitbash.ino) as well and see if it can read from the display.

ago-cs, 2020-07-28:
Thank you!

Looks like I am a happy holder of **ST7735S**

The sketch initially didn't work for me with my previous settings - silence in serial - but it works with the following ones:
```
#define TFT_MOSI  D7
#define TFT_SCK D5
#define TFT_CS D1
#define TFT_DC  D2
#define TFT_RESET D0
```

**Output**
```
TFT driver register values:
===========================
Register 0x01: 0x00
Register 0x04: 0x7C89F0
Register 0x09: 0x610000
Register 0x0A: 0x08
Register 0x0B: 0x00
Register 0x0C: 0x06
Register 0x0D: 0x00
Register 0x0E: 0x00
Register 0x0F: 0x00
Register 0x2E: 0xC6725C
Register 0x3F: 0xFFFFFFFF
Register 0xDA: 0x7C
Register 0xDB: 0x89
Register 0xDC: 0xF0
===========================

Looks like driver chip is: ST7735S (empirical value)
```

But the corresponding changes in the configuration User_Setup.h didn't help.
Also checked in Arduino IDE (as I understand I had to update the User_Setup.h in package zip stored in c:\Users\<profile>\AppData\Local\Arduino15\staging\libraries\TFT_eSPI-2.2.14.zip\TFT_eSPI-2.2.14\)

**Final configuration from Arduino IDE**
```
#define ST7735_DRIVER

#define TFT_WIDTH  128
#define TFT_HEIGHT 160

// #define ST7735_INITB
// #define ST7735_GREENTAB
#define ST7735_GREENTAB2
// #define ST7735_GREENTAB3
// #define ST7735_GREENTAB128    // For 128 x 128 display
// #define ST7735_GREENTAB160x80 // For 160 x 80 display (BGR, inverted, 26 offset)
// #define ST7735_REDTAB
// #define ST7735_BLACKTAB
// #define ST7735_REDTAB160x80   // For 160 x 80 display with 24 pixel offset

#define TFT_MOSI PIN_D7
#define TFT_SCLK PIN_D5
#define TFT_CS   PIN_D1
#define TFT_DC   PIN_D2
#define TFT_RST  PIN_D0

//#define TFT_BL PIN_D8 // also checked with uncommented due to not sure

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF
#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000
```

Sketch I tested it with is from TFT_eSPI demo **TFT_Print_Test**.

ago-cs, 2020-07-28:
**It's a big step forward!!!**
I could see at least something via Adafruit example while brutforcing pin combinations. I saw their demo graphicstest.ino using the following config:
```
  #define TFT_CS         5  // WAS 4
  #define TFT_RST        16                                            
  #define TFT_DC         4  // WAS 5
```
and
```
tft.initR(INITR_BLACKTAB);      // Init ST7735S chip, black tab
```
I totally unplugged the BLK PIN due to not sure where it should be and as a result, I received a very dim image.

I did the corresponding changes in your config:
```
#define ST7735_BLACKTAB

#define TFT_CS   D1
#define TFT_DC   D2
#define TFT_RST  D0
```
But after the blinking on load, there is a totally black screen. The code of Main.cpp with your sketch still the same:
```
#include <TFT_eSPI.h>
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI(); 
#define TFT_GREY 0x5AEB

void setup()
{
  tft.init();
  tft.setRotation(1);
  tft.drawRect(50,50,70,70,TFT_GREEN);
}

void loop()
{
  tft.fillScreen(TFT_GREY);
  tft.setCursor(0, 0, 2);
  tft.setTextColor(TFT_WHITE,TFT_BLACK);
  tft.setTextSize(2);
  tft.println("Hello World!");
  while(1) yield(); // We must yield() to stop a watchdog timeout.
}
```

Hope this research will help to do the next step forward together, thank you in advance!

ago-cs, 2020-07-28:
And probably you also could help with BLK pin... Where could plug it in to get a brighter picture?

Bodmer, 2020-07-30:
If [this is a true picture](https://ae01.alicdn.com/kf/HTB13DuNbBWD3KVjSZFsq6AqkpXav/3-3V-1-44-1-8-inch-TFT-LCD-Display-Module-128-128-128-160-65K.jpg_50x50.jpg) of the back of the board then it has no 5V to 3.3V regulator fitted so must only be powered by 3.3V otherwise you risk damaging your connected ESP8266 GPIO pins.  The display will tolerate 4.8V absolute maximum according to the data sheet, but the problem is it will output higher logic levels will are incompatible with the ESP8266 and will force current into the GPIO pin when high.

The board has a transistor Q1 and this is biased ON, by 1 resistor from VCC so you can leave tha BLK pin unconnected. You should be able to see the backlight LEDs glowing at the edges of the screen. If you connect BLK to GND (0V) then the backlight LEDs should switch off.

Bodmer, 2020-07-30:
The other possibility is that the power supply for the ESP8266 and display is inadequate. Make sure the VCC and GND connection to the ESP8266 is good. These displays can run dimly just powered from the logic lines (yes some folk forget to connect the power or GND!)

Bodmer, 2020-10-03:
Assumed solved by user.

closed 2020-10-03 by Bodmer

wisnshaftler, 2021-11-21:
i also have this issue