Bodmer/TFT_eSPI #710: RPi MPI3501 display with NUCLEO-F091RC (STM32)

BhendiGOD1995, 2020-07-28:
Hi!

Thanks for the great library! I have been trying to test [this display](http://www.lcdwiki.com/3.5inch_RPi_Display) with NUCLEO-F091RC, an STM32 dev board, but for some reason the screen remains white when I try to run any of the examples in /examples/480x320. 

The display works with the RPi (I have tested this and I can confirm). So there is no question of having a faulty display. 

The User_Setup file that I am using is /User_Setups/Setup33_RPi_ILI9486_STM32.h. The contents are as follows
`#define STM32
#define RPI_DISPLAY_TYPE
#define ILI9486_DRIVER

// Can use Ardiuno pin references, arbitrary allocation, TFT_eSPI controls chip select
#define TFT_CS   D5 // Chip select control pin to TFT CS
#define TFT_DC   D6 // Data Command control pin to TFT DC (may be labelled RS = Register Select)
#define TFT_RST  D7 // Reset pin to TFT RST (or RESET)

// Chip select for XPT2046 touch controller
#define TOUCH_CS D4

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// STM32 support for smooth fonts via program memory (FLASH) arrays
#define SMOOTH_FONT

#define SPI_FREQUENCY  20000000   // 27MHz SPI clock

#define SPI_TOUCH_FREQUENCY  2500000 // Must be very slow

#define SUPPORT_TRANSACTIONS`

Using the same setup file, I ran Test_Touch_Controller, and it works. I could see the ADC values varying as I applied the stylus on the screen.

I am not sure where the issue is. Can anyone help me with getting the display to work with NUCLEO-F091RC?

Thanks a lot!

BhendiGOD1995, 2020-07-29:
Hi!

So after trying to debug the issue for about 3 days, I finally got it to work with NUCLEO-F091RC. As it turns out, for me the issue was with the values being sent in ILI9486_Init.h, for initialisation. I shamelessly copied the command and data values from https://github.com/sonicpp/ili9486/blob/master/stm32f4/main.c and it worked. I think I need to go through ILI9486 datasheet to understand it better.

Bodmer, 2020-07-30:
Can you post your new ILI9486_Init.h ?
As far as I can see there is no substantial difference in the command sequence in the main.c file you linked to. Some commands are unnecessary and set up the default reset values (B0 command register is 0 after reset antway). Other commands are repeated but this should not be important or necessary. The delays are slightly longer but the ones in the TFT_eSPI init file comply with the data sheet.

BhendiGOD1995, 2020-07-30:
[ILI9486_Init.txt](https://github.com/Bodmer/TFT_eSPI/files/4998754/ILI9486_Init.txt)
Hi!
Attaching the modified ILI9486_Init.h file. Please note that I have used the Arduino digitalWrite functions to reset the display before writing anything to the SPI bus. Also, from the main.c file I have linked in the previous comment, I used a function the author there had writte to issue delays, based on no-op. I defined that function in TFT_eSPI.cpp before the definition for TFT_eSPI::init, because that is where the init file for specified displays are loaded.

Bodmer, 2020-07-30:
I see you have made quite a few changes, I suspect only one may necessary and it is not obvious what the key change is. For example there is an extra reset added but that does not include the delays required by the data sheet so it probably has no effect. The board is reset here by the TFT_eSPI code with the required delays.  I do not see any NOP delays, only a millisecond delay which is required after some commands (e.g. sleep out) and again TFT_eSPI includes these delays. Unless the key change your setup required for your setup is isolated it would be guesswork what change is needed in the TFT_eSPI code. Can you isolate that key difference?

BhendiGOD1995, 2020-09-02:
Hi!

First of all I apologize for the late response. Got busy with some other things and could not attend to this. I am really sorry for this delay.

I performed a few tests, detailed out as follows:
1. I deleted the installed library, and the local repo. Then I cloned the repo again. Made the changes with respect to user setup, so that Setup 33 (/User_Setups/Setup33_RPi_ILI9486_STM32.h) is in use. Then I ran one of the examples. It did not work
2. Then I added the piece of code that I had added before, as per the file I uploaded in my last comment, in ILI9486_Init.h. I commented out the code that was already there (the original initialization code). Then I ran the same example, and it worked.
3. Then I commented the code I had added, and uncommented the code original initialization code. Ran the same example. This time it worked (????).
4. I repeated the points 1 to 3 a couple more times, in the same sequence. The results were same.
5. While repeating the points 1 to 3 for the fourth time, I realized that while uncommenting the original initialization code (as mentioned in point 3), I was also uncommenting [these lines](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9486_Init.h#L11-L12), which are commented in the original initialization code. These lines correspond to a SW reset.
6. Deleted the installed library, and re-installed it again. Uncommented [these lines](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9486_Init.h#L11-L12), and it worked in the first attempt.

So, in short, adding a SW reset fixed the issue in the original initialization code. But it does not make much sense. Because in the initialization code I had replaced it with, there is no standalone SW reset being done. So I performed one more test
7. I commented out the SW reset again, and added a sequence for NOP, like this: 
` writecommand(0x0);
  writedata(0x0);
` 
before [these lines](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9486_Init.h#L11-L15). This time also it worked.

closed 2020-10-03 by Bodmer