Bodmer/TFT_eSPI #711: [Help Needed] Using manual Chip Select

SunboX, 2020-07-28:
Hi Bodmer,

I've an issue using manual CS. The issue looks like this:
https://www.transfernow.net/YCVly9072020

I'm using a display with touch. Both display driver and touch controller are running over the same SPI. I want to do the CS manually, because I have to use a GPIO Expander later on. But the issue happens already using only a ESP32. (simplest setup possible) Using your library with the exact same setup, but without manual CS, all works fine.

I'm using TFT_eSPI's built-in ILI9341 driver together such a "generic" 2.4" touch display:
[image]

I have three of them from different sources and for different prices. Also one from a German seller for "a lot of" money. They behave all the same.

Attached you will find my sample code:
[DisplayTest.zip](https://github.com/Bodmer/TFT_eSPI/files/4988391/DisplayTest.zip)

The Hardware setup looks like this:
[image]

Do you have any clue why it does not work or how I can get it to work? I have no clue whats going wrong here.

Bodmer, 2020-07-29:
Look at the Arduino Blink example. You will see you need to configure the GPIO pins for output using pinMode. Also you must initialise all chip select pins to the high state at the beginning of setup, otherwise you do not know what state they may be in.

SunboX, 2020-07-29:
Ahh, I'm an idiot. 😅 You are right, I will try it. Thanks for your support!

Bodmer, 2020-07-29:
No problem. Hope it works for you. Only other point is the touch screen should be calibrated by running the calibration sketch with the rotation set in that sketch to the rotation you will use in your final project. Let me know how it goes.

SunboX, 2020-07-29:
It works now, great! Now I can go to the next step and add the GPIO Expander. After that, I will add the second screen. And if this works, I will try to get GUISlice running with it.

closed 2020-07-29 by SunboX