Bodmer/TFT_eSPI #715: New Feature: Power save mode (sleep) for supported TFT displays

toetoast, 2020-08-02:
Hi Bodmer,

I was needing a function to put my displays to **sleep**, but then I noticed the command details were covered in a previously closed Issue. So that I could use these commands in my Sketches, I have created the following TFT_eSPI Method: 
```
/***************************************************************************************
** Function name:           sleep
** Description:             Allows the display to be put to sleep then awake again
***************************************************************************************/
void TFT_eSPI::sleep(bool value)
{
    if (value)
    {
        writecommand(0x10);         // Send command to put the display to sleep.
        delay(5);                   // Delay for shutdown time before another command can be sent.
    }
    else
    {
        writecommand(0x11);         // Send command to wakeup the display.
        delay(120);                 // Wait for the display power supplies to stabilise.
    }
}
```
This works really well on the 2 types of ILI9341 based displays that I have.

- 240x320 Color TFT Resistive Touchscreen supplied by PJRC:
[https://www.pjrc.com/store/display_ili9341_touch.html](https://www.pjrc.com/store/display_ili9341_touch.html)
- 240x320 Color TFT Capacitive Touchscreen supplied by Adafruit:
[https://www.adafruit.com/product/2090](https://www.adafruit.com/product/2090)

Using the following code snippit, I determined that the current consumption of the Resistive Touchscreen TFT dropped by about 13mA when in sleep mode (as shown in the photos below), and then the current drawn dropped to _almost zero_ once the backlight was turned off (more on this in a future post).

```
do
{
   tft.sleep(true);    
   delay(5000);    
   tft.sleep(false);
   delay(5000);
} while (1);
```

Do I need to consider different command codes to put other **supported displays** in this library to sleep, or are the above command codes universal?

Just thought I'd share this here for anyone looking to achieve a similar feature in their Sketches to save battery power.

Regards
Tony

TFT screen that is "awake" (on the left):
[image]

TFT screen when in "sleep" mode (on the left):
[image]

Bodmer, 2020-08-02:
Hi,
This has been raised a few times before. Some displays might require full initialisation after coming out of sleep and others require different delays. TFT_eSPI allows init()/begin() to be called multiple times in a sketch to facilitate the "display on" recovery when a display power switch is used. Some displays have inaccessible LED power, all these things need to be considered carefully and explained to users.  The parameters need to be determined and will likely be display type specific.

 I will think further how this might be added, the results are interesting.

toetoast, 2020-08-03:
Understood. It's a shame that there is no set standard across displays for this.

I also observed the following during my ILI9341 screen testing:

1. the Capacitive Touch screen from Adafruit consumed only 5mA less when in sleep mode (78mA to 73mA), with the backlight on.
2. both the Resistive Touch and Capacitive Touch screens performed with no issues when in the sleep mode (meaning the Touch interface was still working correctly, which is cool). This allowed me to implement a "go to sleep" mode when the screen has not been touched for a predetermined time, then the screen wakes up again when touched :)  I did use a backlight control feature here and the current drawn by both displays was negligable when asleep/backlight off, which is great.

I'll be hooking up the following displays this week, so I can check how the "sleep" functions performs on them (no backlight control here though from what I've seen):

- ST7735S based 128x128 1.44" TFT LCD Color
- ST7789VW based 240x240 1.54" TFT LCD Color

Thanks
Tony

Bodmer, 2020-08-03:
A quick look at some data sheets indicates the displays change to the default registers after the sleep out comand.

This should work with all displays but is untested:

```
/***************************************************************************************
** Function name:           sleep
** Description:             Allows the display to be put to sleep then awake again
***************************************************************************************/
void TFT_eSPI::sleep(bool value)
{
    if (value)
    {
        writecommand(0x10);         // Send command to put the display to sleep.
        delay(150);                         // Delay for shutdown time before another command can be sent.
    }
    else
    {
        init();                                 // This send the wake up command and initialises the display
    }
}
```

toetoast, 2020-08-04:
That's great thanks. I'll test on the displays I have and get back to you.

toetoast, 2020-08-06:
Hi, the new "sleep" method is working well so far on 2 x ILI9341 controller based displays as shown here: [https://www.instagram.com/p/CDibQ1LHZdP/](https://www.instagram.com/p/CDibQ1LHZdP/)

There is a white flash as it comes out from sleep. I suspect I can solve that by enabling the backlight **after** the screen has been initialised from sleep.

I'll continue testing on other controller types.
Tony

toetoast, 2020-08-11:
Hi Bodmer, 

Here are the results of my further testing on 3 screens with differing controllers.
### ST7735S based 128x128 1.44" display
The display enters sleep and leaves with no issues. 

### ST7789VW based 240x240 1.54" display
When the display is powered up the colours are inverted. Then when coming out of sleep the colours are correct from that point onwards.
Have a look at this video which expains it. Each time the screen goes blank it’s being forced into “sleep” mode for 1 second.
[https://www.instagram.com/p/CDvdmpLlj2H/](https://www.instagram.com/p/CDvdmpLlj2H/)

Here is my display setup code. If I insert the **tft.init();** line then the above inverted colour issue goes away!
```
tft.begin();
tft.setRotation(0);
tft.invertDisplay(false);
tft.setSwapBytes(false);
tft.fillScreen(TFT_BLACK); 
tft.setTextWrap(false, false);

tft.init();
```

### ILI9341 based 240x320 2.8" display
The display enters sleep with no issues. However, there is a brief white 'flash' when coming out of sleep mode.
I needed to add the **delay(50)**; as shown below to fix that issue.
```
void TFT_eSPI::sleep(bool value)
{
    if (value)
    {
        writecommand(0x10);   // Send command to put the display to sleep.
        delay(150);           // Delay for shutdown time before another command can be sent.
    }
    else
    {
        init();               // This sends the wake up command and initialises the display
	delay(50);            // Extra delay to stop a "white flash" while the TFT is initialising.
    }
}
```

On all displays, the internal screen data buffer remains intact and subsequent commands to update it work correctly. For each display test the Sketch is the same. Only the necessary alterations to the **Setup_User.h** is made.

Tony

Bodmer, 2020-08-12:
My tests indicate the ILI9241 CGRAM gets wiped so hence the flash. Adding a delay did not solve this for me.  An ST7796 behaved better and the CGRAM was maintained.

In view of the incosistent behaviours I think it is best to leave this function to be within a users sketch, otherwise I will end up having issues raised.

Closing for now but will keep this as a potential enhnacement.

The incosistent behaviour of the ST7789V will be investigated.

closed 2020-08-12 by Bodmer

Axelman8, 2021-11-03:
Hey there @toetoast . Sorry to break in on this closed thread.  I am interested how you got the ST7735 to go to sleep
 https://github.com/Bodmer/TFT_eSPI/issues/715#issuecomment-671885854  . 

I tried to add this option to my project with 15 TFT screens but i cannot get it to work.  Any help is appreciated.  thank you
Axelman8

toetoast, 2021-11-03:
Hi there, I wolud say the answer would be to use the following:

writecommand(0x10);   // Send command to put the display to sleep.

Which comes from the datasheet for the display controller.

Regards
Tony

micha6554, 2022-06-23:
thanks, works for PCF8574!  Thanks

yeodeanyeo, 2022-07-11:
Hey @toetoast , do you have codes for the sleep function? 
I am working on a school project and ive got the interface all done. 
Now i just need a sleep mode if not screen will be on forever. 

If youre able to share it do let me know. Thanks!

Andy-droid, 2023-07-05:
hi has this been implemented in to the library?
or how do I use it?

toetoast, 2023-07-06:
Hey, I would say it's not been implemented.

The way to use it is shown in the comment dated Aug 11, 2020

Andy-droid, 2023-07-06:
Thanks! I have limited knowledge of code.
I'm not sure where you are adding that code. Does it go in the library, or does it go in the sketch?
Can you give me a step by step, of the files to edit.