Bodmer/TFT_eSPI #716: How to compress the font file

Lucy2003, 2020-08-03:
I created a font file using Processing,but the font file is too big to fix in SPIFFS(1.6MB),so I want to compress it.

In ReadMe.md,
`It would be possible to compress the vlw font files but the rendering performance to a TFT is still good when storing the font file(s) in SPIFFS.`

Could you please tell me how?Thanks.

--Lucy2003

Bodmer, 2020-08-03:
It would be possible... but that would need new compression/decompression code to be written. The library only supports the vlw format for smooth fonts.
At the moment you have 2 options:
1. Trim down the character set to just the ones you actually need, the Processing sketch allows you to do this.
2. Store the fonts on SD card - see ESP32 example

Lucy2003, 2020-08-04:
I know.Is there any way to build and use font in the program,Just like FreeFont?

Thanks.

Lucy2003, 2020-08-05:
Solved.I used U8g2_for_TFT_eSPI so I could make bin font by myself.It's smaller(about 300KB)!

Thank you.

closed 2020-08-05 by Lucy2003

Bodmer, 2020-08-05:
A good solution if you do not want the anti-aliasing (smoothing). The vlw smooth font format encodes each pixel as 8 bits rather than 1 bit so you would expect about x8 increase in file size.