Bodmer/TFT_eSPI #719: Support for LILYGO TTGO T8 ESP32-S2 with ST77789

vanvuongngo, 2020-08-04:
Change request ... please add a setup for this board LILYGO TTGO T8 ESP32-S2 with ST77789.
see http://www.lilygo.cn/claprod_view.aspx?TypeId=21&Id=1322&FId=t28:21:28

Bodmer, 2020-08-05:
Support for the ESP32S2/3 is planned however the Arduino board package with support for these processors is still incomplete and has not been released yet.  TTGO boards with the dual core ESP32 are commonly available and are supported by the library.

closed 2020-08-05 by Bodmer

Bodmer, 2021-11-18:
The library now supports the ESP32 S2, this has been tested with the Arduino ESP32 board package 2.0.1.

A new setup file example has been added:
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup70_ESP32_S2_ILI9341.h