Bodmer/TFT_eSPI #722: Having problem getting touch to work with STM32F103C8T6 Blue Pill

sm5oem, 2020-08-08:
Having problem getting touch to work with TFT-eSPI. Hoping problem is obvious, but I'm stuck. Here's what I have; STM32F103C8T6 and a 3.5 TFT LCD ILI9488 display. I installed a fresh, clean copy of the TFT-eSPI. Everything is working fine except for the touch.
Physical connections are:

STM32F103C8T6     ili9488 (3.5 TFT SPI)
n/c                 T_IRQ
PA6                 T_DO
PA7                 T_DIN
PA1                 T_CS
PA5                 T_CLK
PA6                 SODI(MISO)
3V3                 LED
PA5                 SCK
PA7                 SDI(MOSI)
PB0                 D/C
PB1                 RESET
PA4                 CS
GND                 GND
VCC                 VCC 3.3V 

///////Setup for STM32F103 (e.g. Blue Pill) and ili9488 display//////

// reminder: Blue Pill SPI pins are SCK=A5, MISO = A6, MOSI=A7
// Can use Ardiuno pin references, arbitrary allocation, TFT_eSPI controls chip select
#define TFT_CS   PA4 // Chip select control pin to TFT CS
#define TFT_DC   PB0 // Data Command control pin to TFT DC (may be labelled RS = Register Select)
#define TFT_RST  PB1 // Reset pin to TFT RST (or RESET)

#define TOUCH_CS PA1 // Connect to T_CS (Touch controller chip select) if XPT2046 is connected to SPI bus

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
// STM32 support for smooth fonts via program memory (FLASH) arrays
#define SMOOTH_FONT
// Assuming the processor clock is 72MHz:
#define SPI_FREQUENCY  36000000   // 36MHz SPI clock
//#define SPI_FREQUENCY  18000000   // 18MHz SPI clock
#define SPI_READ_FREQUENCY  12000000 // Reads need a slower SPI clock
#define SPI_TOUCH_FREQUENCY  2500000 // Must be very slow
Thanks
Ronny

Bodmer, 2020-08-08:
Disconnect MISO from the ILI9488. The ILI9488 chip is an oddity and the MISO output never goes tri-state.

sm5oem, 2020-08-08:
Hi do i need to disconnect it completely or only from PA6 or T_DO?
Thanks,

sm5oem, 2020-08-08:
Hi did the disconnect and this is the result now and no reaction from touch.
same with the ILI9341
// Use this calibration code in setup():
  uint16_t calData[5] = { 65535, 1, 65535, 1, 0 };
  tft.setTouch(calData);
thanks,

sm5oem, 2020-08-08:
Problem solved

STM32F103C8T6 ili9488 (3.5 TFT SPI)
n/c T_IRQ
PA6 T_DO
PA7 T_DIN
PA1 T_CS
PA5 T_CLK
Does not work here on  ili9488 (SODI MISO)
3V3 LED
PA5 SCK
PA7 SDI(MOSI)
PB0 D/C
PB1 RESET
PA4 CS
GND GND
VCC VCC 3.3V

Works fine with STM32F103C8T6 ili9488 (3.5 TFT SPI)
Thanks,
Ronny

closed 2020-08-08 by sm5oem