Bodmer/TFT_eSPI #723: Wemos D1 / ILI9341 & Thingpulse PCB

anthonyjclarke, 2020-08-10:
Appreciate some guidance, been looking at this for days and cant get to work.

I am using this PCB / wiring config : https://docs.thingpulse.com/specs/wifi-color-display-kit/#wiring and details https://docs.thingpulse.com/guides/wifi-color-display-kit/

[image]

I am able to get the Thingpulse / Weatherstation to work, even code using Adafruit GFX Drivers, but just cant get your drivers to work

Here are my defines, I think I have tried all permutations....

// Only define one driver, the other ones must be commented out
#define ILI9341_DRIVER
/ For NodeMCU - use pin numbers in the form PIN_Dx where Dx is the NodeMCU pin designation
#define TFT_CS   PIN_D1  // Chip select control pin D8
#define TFT_DC   PIN_D2  // Data Command control pin
#define TFT_RST  PIN_RST  // Reset pin (could connect to NodeMCU RST, see next line)
#define TFT_RST  -1    // Set TFT_RST to -1 if the display RESET is connected to NodeMCU RST or 3.3V

** I tried without RST

Any thoughts suggestions really appreciated!

Thanks

Bodmer, 2020-08-10:
Run the diagnostic sketch Read_User_Setup to make sure the configuration is picked up by the compiler. Post output here.

You must also make sure the T_CS line is high before calling tft.begin() if you are not using the touch driver in the TFT_eSPI library. If you will use the built in touch driver then use #define TOUCH_CS PIN_D3.

anthonyjclarke, 2020-08-11:
Here is the output,

[code]
TFT_eSPI ver = 2.2.14
Processor    = ESP8266
Frequency    = 80MHz
Voltage      = 3.27V
Transactions = No
Interface    = SPI
SPI overlap  = No

Display driver = 7789
Display width  = 135
Display height = 240

R0 x offset = 52
R0 y offset = 40
MOSI    = GPIO -119
SCK     = GPIO -118
TFT_CS   = PIN_D1
TFT_DC   = PIN_D0
TFT_RST  = PIN_D-1

TFT_BL           = PIN_D2
TFT_BACKLIGHT_ON = HIGH

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
[/code]

anthonyjclarke, 2020-08-11:
[code]
TFT_eSPI ver = 2.2.14
Processor    = ESP8266
Frequency    = 80MHz
Voltage      = 3.27V
Transactions = No
Interface    = SPI
SPI overlap  = No

Display driver = 7789
Display width  = 135
Display height = 240

R0 x offset = 52
R0 y offset = 40
MOSI    = GPIO -119
SCK     = GPIO -118
TFT_CS   = PIN_D1
TFT_DC   = PIN_D0
TFT_RST  = PIN_D-1

TFT_BL           = PIN_D2
TFT_BACKLIGHT_ON = HIGH

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
[/code]

Oh dear, Silly user error!

I realised I was not using this :

#include <User_Setup.h>           // Default setup is root library folder

It had been commented out and I have selected a user setting for another TFT I had been using previously.

I did have to add this code to enable to Display

#define TFT_LED D8
  
void setup(void) {

  pinMode(TFT_LED, OUTPUT);
  digitalWrite(TFT_LED, HIGH);    // HIGH to Turn on;

Then it worked!!

Ran Diags to confirm

[code]
TFT_eSPI ver = 2.2.14
Processor    = ESP8266
Frequency    = 80MHz
Voltage      = 3.28V
Transactions = No
Interface    = SPI
SPI overlap  = No

Display driver = 9341
Display width  = 240
Display height = 320

TFT_CS   = PIN_D1
TFT_DC   = PIN_D2

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.00
[/code]

Sorry to bother you!

Bodmer, 2020-08-11:
You have not configures the setup correctly.

Load a brand new copy of the library and just edit [these 3 lines](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L153-L155) and [this one line](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L160) to suit your setup. Do not change anything else.

Use the following alternative lines exactly as written below:
```
#define TFT_CS   PIN_D1  // Chip select control pin D8
#define TFT_DC   PIN_D2  // Data Command control pin
#define TFT_RST  -1  // Reset pin (could connect to NodeMCU RST, see next line)
#define TOUCH_CS PIN_D3     // Chip select pin (T_CS) of touch screen
```
Upload and run the Read_User_Setup again to make sure you got it right.

anthonyjclarke, 2020-08-11:
Thanks, yes all sorted as I got my User_Setup and User_Setup_Select mixed up....

closed 2020-08-11 by anthonyjclarke