Bodmer/TFT_eSPI #726: followup to issue #725

x567234, 2020-08-12:
Thank you for your prompt reply.
In another project I have a pre-defined 18-bits colours that I would like to use,to save from the double conversion losses (18 ->16 ->18)

I looked through the code of tft_eSPI but could not find what to bypass, in the code, to pass 18-bits colors directly, without going through the 16-bits to 18-bits. 
Is it possible to bypass the "16 bit colour to an 18 bit value " conversion and pass the 18-bit values directly .

Sorry for any inconvenience

Bodmer, 2020-08-12:
Changes to the library would be needed to implement 18 bit colours. The code for the conversion is in the processors folder, for example[ here](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32.h#L347-L349) for an ESP32.

Your best option is to either fork the library and make the changes or build the required functions within your sketch.

closed 2020-08-12 by Bodmer

x567234, 2020-08-13:
Thank you. That is exactly what I was looking for. Appreciate it