Bodmer/TFT_eSPI #728: ILI9481 based LCD doesn’t work in SPI mode

quangvankts, 2020-08-14:
Hi Bodmer
I bought a couple of ILI9481 displays from Aliexpress expecting they will work in SPI as advertised with ESP32
https://a.aliexpress.com/_m0wVFOx
After a few failed attemps, I contacted the seller and they sent me the LCD datasheet:
https://wetransfer.com/downloads/ec658db81ee4cd23551f30a0b3e4e76420200814065355/f26fddef412833978343beb07d7cab1d20200814065355/8efcd0

The above file translated to English:
https://www.dropbox.com/s/adaef51rbha36sl/3.5-40pin-ILI9481-IPS-190904.zh-CN.en.pdf?dl=0

.. which is different from the their Aliexpress page (SPI clock and TFT_DC swapped, IMx pins config is different). 
It didnt work neither. 
This is the Read User Setting example output 

[code]
TFT_eSPI ver = 2.2.17
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9481
Display width  = 320
Display height = 480

MOSI    = GPIO 23
MISO    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 13
TFT_DC   = GPIO 14
TFT_RST  = GPIO 12

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 1.00
[/code]

The LCD was displaying running vertical stripes when running an example with SPI set to 1Mhz. The stripes disappeared when CLK or MOSI was disconnected, and back when plugged in. So maybe some data signal was going through ?

[image]

Is there any other way to make them work with SPI (I cant run parallel I dont have enough pins left) ? 
Thank you

Bodmer, 2020-08-14:
How have you wired it up? i.e. which pin to which pin?

You do realise that there is a break in the power lines as indicated by ellispes below...
[image]

quangvankts, 2020-08-15:
This is how it is wired up:
Pin | LCD Datasheet | ESP32 Devkit C
1-4 | Touch | Not connected
5 | GND | GND
6 | VCC | 3.3V
7 | VCC | 3.3V
8 | Frame Sync / TE | Not connected
9 | SPI_CS | GPIO 13
10 | SPI_A0 | GPIO 14
11 | SPI_SCL | GPIO 18
12 | RD | 3.3V
13 | SPI_SDA | GPIO 23
14 | SPI_SDO | GPIO 19
15 | RESET | GPIO 12
16 | GND | GND
17-32 | DB0-DB15 Parallel | GND
33 | LED Anode | 3.3V
34-36 | LED Cathode | GND
37 | GND | GND
38 | IM0 | 3.3V
39 | IM1 | 3.3V
40 | IM2 | 3.3V

Yes there is a black jumper wire connecting the GND line break. 
Pin 17-32 (DIN Parallel) were partially grounded in the pic above. I grounded them all and the stripes disappeared. Now te screen is back to solid black.
I rechecked all other pins and they are correct.  

The example I ran is Color Test using PlatformIO and TFT_eSPI in libdeps 
This is my User_Setup.h
https://www.dropbox.com/s/tzh5tboydu8vxa1/User_Setup.h?dl=0

quangvankts, 2020-08-15:
Update: I wired up a ILI9341 LCD using the same SPI pins from the ESP32 and the same sketch and it works ! So I suspect it could be pin IM0-2 config ? 
[image]

Bodmer, 2020-08-15:
Grounding the paralle data pins would cause black to be loaded if the write strobe is being clocked, so that suggests the mode pins are not correctly set.

Bodmer, 2020-08-15:
IM1 and 2 should be high, and IM0 low to invoke 4 wire SPI mode.

quangvankts, 2020-08-15:
> IM1 and 2 should be high, and IM0 low to invoke 4 wire SPI mode.
I just tried and it didn't work neither.
The datasheet suggests all IMx should be high to active 4 wire SPI. Which also didn't work. Very weird :( 
[image]

Bodmer, 2020-08-15:
Aliexpress page from your link above:
[image]

ILI9481 datasheet:
[image]

Makes me wonder what else is wrong in the pin list on Aliexpress...

quangvankts, 2020-08-16:
I suspect the Aliexpress seller page's list is for ILI9341 LCD ?
The LCD datasheet they sent me over email is actually belongs to ILI9488, but it is quite similar to ILI9841 ?
I tried wiring it in parallel (MCU 8 bit) mode and it didnt work too. 
Maybe IM0-IM2 are not connected to internally ? Or they sent the wrong model ? No sign of signal passing through the LCD at all, just a black background.

quangvankts, 2020-08-16:
Update: Parallel mode is working now. Now back to SPI !
[image]

Bodmer, 2020-08-16:
Try ILI9488 driver with SPI to see it it changes the screen behaviour.

quangvankts, 2020-08-16:
> Try ILI9488 driver with SPI to see it it changes the screen behaviour.

The white vertical lines disappear with ILI9488 driver. The screen is back to solid black

quangvankts, 2020-08-18:
Update 3:The seller sent me a zip file which he said include SPI code for ILI9841 ? 

[https://wetransfer.com/downloads/c191bb087dcfbb62cbd9ceac2f5727d920200817034722/40730acc6be0bbcac56c615e4e6566d220200817034722/e27590?utm_campaign=WT_email_tracking&utm_content=general&utm_medium=download_button&utm_source=notify_recipient_email](url)

`#include "lcd.h"
#include "stdlib.h"
#include "font.h" 
#include "usart.h"
#include "delay.h"	 
   
					 
//������ɫ,������ɫ
u16 POINT_COLOR = 0x0000,BACK_COLOR = 0xFFFF;  
	 

//����LCD��Ҫ����
//Ĭ��Ϊ����
_lcd_dev lcddev;

#if LCD_FAST_IO==1 //����IO
//д�Ĵ�������
void LCD_WR_REG(u16 data)
{ 
	LCD_RS_CLR;//д��ַ  
 	LCD_CS_CLR; 
	DATAOUT(data); 
	LCD_WR_CLR; 
	LCD_WR_SET; 
 	LCD_CS_SET;   
}
#else//����IO
//д�Ĵ�������
void LCD_WR_REG(u16 data)
{ 
	LCD_RS=0;//д��ַ  
 	LCD_CS=0; 
	DATAOUT(data); 
	LCD_WR=0; 
	LCD_WR=1; 
 	LCD_CS=1;   
} 	
#endif	

//д�Ĵ���
void LCD_WriteReg(u16 LCD_Reg, u16 LCD_RegValue)
{	
	LCD_WR_REG(LCD_Reg);  
	LCD_WR_DATA(LCD_RegValue);	    		 
}	   
//���Ĵ���
u16 LCD_ReadReg(u8 LCD_Reg)
{										   
	u16 t;
	LCD_WR_REG(LCD_Reg);  //д��Ҫ���ļĴ�����  
	GPIOB->CRL=0X88888888; //PB0-7  ��������
	GPIOB->CRH=0X88888888; //PB8-15 ��������
	GPIOB->ODR=0XFFFF;    //ȫ�������
#if LCD_FAST_IO==1 //����IO
	LCD_RS_SET;
	LCD_CS_CLR;
	//��ȡ����(���Ĵ���ʱ,������Ҫ��2��)
	LCD_RD_CLR;
	delay_us(5);//FOR 8989,��ʱ5us					   
	LCD_RD_SET;
	t=DATAIN;  
	LCD_CS_SET; 
#else
	LCD_RS=1;
	LCD_CS=0;
	//��ȡ����(���Ĵ���ʱ,������Ҫ��2��)
[... 726 lines cut ...]
#define R135           0x87
#define R136           0x88
#define R137           0x89
#define R139           0x8B
#define R140           0x8C
#define R141           0x8D
#define R143           0x8F
#define R144           0x90
#define R145           0x91
#define R146           0x92
#define R147           0x93
#define R148           0x94
#define R149           0x95
#define R150           0x96
#define R151           0x97
#define R152           0x98
#define R153           0x99
#define R154           0x9A
#define R157           0x9D
#define R192           0xC0
#define R193           0xC1
#define R229           0xE5							  		 
#endif  
	 
`

quangvankts, 2020-08-27:
I will keep this in 8 bit parallel as it works best that way. Thank you for your help

closed 2020-08-27 by quangvankts

Paulie92, 2020-12-13:
@quangvankts do you have any success with the SPI? I am currently experimenting with the same display and have still no luck with the SPI. The 8 bit parallel works, but the speed is the same as in case of SPI@40mhz on ST7789 display (pushing a 16 bit 240x240 sprite takes for both displays the same time of 22ms, which is quite strange). As the speed is the same the 8 bit parallel has no benefit for me, and needs too much GPIO.

Paulie92, 2020-12-13:
I downloaded the current wersion of library and it works now! But unfortunately the maximum speed of the SPI is 15mhz, so it is also not suitable for my aplication. @Bodmer is the speed of the parallel compared to the 40mhz SPI OK? I think that I read one of your comments where you decribing the 8 bit parellel approx 1.7x faster than the SPI@40mhz. Thank you for answer in advance.

Bodmer, 2020-12-14:
Yes, the ILI9481 needed a register value change for SPI and this has been added. Some driver ICs operate at higher that data sheet speeds but the ILI9481 does not. A further slow down occurs because each pixel needs 3 bytes to be sent rather than 2.

You can run performance tests without a display connected, providing the results appear in the serial window.

SPI has the advantage of DMA but the benefit depends on the sketch processor demands. Performance also varies between processors. Typical ESP32 results for graphicstest:
```
ESP32 parallel ILI9341
Benchmark,                Time (microseconds)
Screen fill (5 times),    100717
Text,                     6594
Lines,                    39054
Horiz/Vert Lines,         9561
Rectangles (outline),     6094
Rectangles (filled),      160847
Circles (filled),         36709
Circles (outline),        21454
Triangles (outline),      9732
Triangles (filled),       80560
Rounded rects (outline),  12798
Rounded rects (filled),   263675
Total = 747795us
Total = 0.7478s

ESP32 SPI ILI9341 40MHz
Benchmark,                Time (microseconds)
Screen fill (5 times),    166771
Text,                     13485
Lines,                    78008
Horiz/Vert Lines,         14579
Rectangles (outline),     9247
Rectangles (filled),      346275
Circles (filled),         62127
Circles (outline),        47635
Triangles (outline),      18742
Triangles (filled),       120225
Rounded rects (outline),  24405
Rounded rects (filled),   381134
Total = 1282633us
Total = 1.2826s
```

ismaelit, 2022-08-15:
> I downloaded the current wersion of library and it works now! But unfortunately the maximum speed of the SPI is 15mhz, so it is also not suitable for my aplication. @Bodmer is the speed of the [...]

Hello Paulie, do you have made some advance, or give up on this display? Can you share our pin settings and relevant User_Setup settings? Thanks!